LinuxCNC/linuxcnc #1021: Occurence to EMCIO but use other ini section TOOLSENSOR

alkabal, 2021-01-02:
https://github.com/LinuxCNC/linuxcnc/blob/848ff4105a01b5eaa5c921237055f936a0f96b11/nc_files/remap_lib/python-stdglue/stdglue.py#L539

alkabal, 2021-01-02:
ALso it is normal to go directly to toolsensor rather than first go to toolchange position ?

c-morley, 2021-01-02:
linuxcnc lets you set where you change the tool already - or you can move to a position prior to calling tool change.
This is set up in the INI.
But when I changed that in the remap, I forgot to change the process after tool probe to return back to original position.
I think the best route is:  after tool change  record the current position, then tool probe, then return to that position.

c-morley, 2021-01-02:
testing this again - I see i was mistaken. emccanon.CHANGE_TOOL doesn't move to EMCIO toolchange settings.
I'm not sure i can call the function that does from python.
I'll have to recreate the actions.

I did get it to record the current position and return to it after tool probe - I haven't pushed any of this yet.

alkabal, 2021-01-02:
I understand better now ! 
i haven't tested your code, only read for comparing to remastered gladevcp remap.

i have attached the revised ocode from original vers probe manual change

[psng_manual_change.txt](https://github.com/LinuxCNC/linuxcnc/files/5759754/psng_manual_change.txt)

alkabal, 2021-01-02:
minor edit 
[psng_manual_change.txt](https://github.com/LinuxCNC/linuxcnc/files/5759762/psng_manual_change.txt)

c-morley, 2021-01-04:
Thanks for your file - I will use it for reference when i get back to this - real work calls....

alkabal, 2021-01-04:
Hi 
I suspect something is wrong in the file provided I will revisit this parts actually.

I send you other one soon 

Br

alkabal, 2021-01-05:
[psng_manual_change.txt](https://github.com/LinuxCNC/linuxcnc/files/5768014/psng_manual_change.txt)

i think this one is better i have use many time testing the stuff with or without autolenght enabled in all possible situation (i hope)

here is the corresponding python code : 

```

# One issue need to be corrected if you ask the same tool as actual tool probe start without any confirmation (patched in ocode with M0)
                 
                 
        if change:
            # if toolprepnumber = 0 we will get an error because we will not be able to get
            # any tooldescription, so we avoid that case
            if toolprepnumber == 0:
                text = "Please remove the mounted tool and unpause or cancel"
                print(text)
                message = _("Please remove the mounted tool and press OK when done")
            else:
                tooltable = self.inifile.find("EMCIO", "TOOL_TABLE")
                if not tooltable:
                    text = "**** Did not find a toolfile file in [EMCIO] TOOL_TABLE ****"
                    print(text)
                    self.gcode("(ABORT,**** %s ****)" % text)                      # Need investigation with self.command.abort()
                    return
                CONFIGPATH = os.environ["CONFIG_DIR"]
                toolfile = os.path.join(CONFIGPATH, tooltable)
                self.tooledit1.set_filename(toolfile)
                tooldescr = self.tooledit1.get_toolinfo(toolprepnumber)[16]
                text = "**** Please change to tool %f %s****" % (toolprepnumber, tooldescr)
                print(text)
                message = _(
                    "Please change to tool\n\n# {0:d}     {1}\n\n then click OK."
                ).format(toolprepnumber, tooldescr)
            result = self.warning_dialog(message, title=_("Manual Toolchange"))
            if result:
                text = "TOOLCHANGED CORRECTLY"
                print(text)
                self.halcomp["toolchange-changed"] = True
            else:
                text = "TOOLCHANGE ABORTED %f %f" % (toolnumber,self.halcomp["toolchange-prep-number"])
                print(text)
                self.command.abort()                                                  # Need investigation with self.command.abort()
                self.halcomp["toolchange-prep-number"] = toolnumber
                self.halcomp["toolchange-change"] = False  # Is there any reason to do this to input pin ?
                self.halcomp["toolchange-changed"] = True                
                message = _("**** %s ****" % text)
                self.warning_dialog(message)
        else:
            self.halcomp["toolchange-changed"] = False
```

alkabal, 2021-01-05:
Hi again

Here is my first attempt for use your full python remap, for now without logic check like do not probe tool 0...

```
#[CHANGE_POSITION]
#Z_SAFE_TRAVEL_OFFSET = -1.5
#X = 0
#Y = 0
#Z = 0

#[TOOLSENSOR]
#X = -5.00
#Y = -1
#Z = -1

#SETTER_HEIGHT = 40
#MAXPROBE =  -3
#TS_LATCH = 0.3
#REVERSE_LATCH = 2
#SEARCH_VEL = 300
#PROBE_VEL = 10

# What about checking machine is homed before allow this?
# TODO allow the autoprobe section to be a configurable option from some interface (restoring manual mode)
# TODO ver's use a configurable workheight and setterheight that can be set manually or using probe
# TODO the dialog box using hal_manualtoolchange come too early before goto change position

def tool_probe_m6(self, **words):

    # only run this if we are really moving the machine
    # skip this if running task for the screen
    if not self.task:
        yield INTERP_OK

    try:
        self.params["tool_in_spindle"] = self.current_tool
        self.params["selected_tool"] = self.selected_tool
        self.params["current_pocket"] = self.current_pocket
        self.params["selected_pocket"] = self.selected_pocket

        # cancel tool offset
        self.execute("G49")
        
        # change tool where ever we are
        # user sets toolchange position prior to toolchange
        # we will return here after

        try:
            self.selected_pocket =  int(self.params["selected_pocket"])
            emccanon.CHANGE_TOOL(self.selected_pocket)
            self.current_pocket = self.selected_pocket
            self.selected_pocket = -1
            self.selected_tool = -1
            # cause a sync()
            self.set_tool_parameters()
            self.toolchange_flag = True
        except InterpreterException as e:
            self.set_errormsg("tool_probe_m6 remap error: %s" % (e))
            yield INTERP_ERROR

        yield INTERP_EXECUTE_FINISH

        # record current position; probably should record every axis
        self.params[4990] = emccanon.GET_EXTERNAL_POSITION_X()
        self.params[4998] = emccanon.GET_EXTERNAL_POSITION_Y()
        self.params[4997] = emccanon.GET_EXTERNAL_POSITION_Z()

        try:
        	
            # first go up to toolchange position
            self.execute("G90")
            self.execute("G53 G0 Z[#<_ini[AXIS_Z]MAX_LIMIT>+#<_ini[CHANGE_POSITION]Z_SAFE_TRAVEL_OFFSET>]")
            self.execute("G53 G0 X#<_ini[CHANGE_POSITION]X> Y#<_ini[CHANGE_POSITION]Y>")
            self.execute("G53 G0 Z#<_ini[CHANGE_POSITION]Z>")
        	
        	
            # move to tool probe position (from INI)     # TRYING with TOOL_CHANGE_POSITION but TOOL_CHANGE_QUILL_UP seem to do not work
            self.execute("G90")
            self.execute("G53 G0 Z[#<_ini[AXIS_Z]MAX_LIMIT>+#<_ini[PROBE_SCREEN]Z_SAFE_TRAVEL_OFFSET>]")
            self.execute("G53 G0 X#<_ini[TOOLSENSOR]X> Y#<_ini[TOOLSENSOR]Y>")
            self.execute("G53 G0 Z#<_ini[TOOLSENSOR]Z>")

            # set incremental mode
            self.execute("G91")

            # course probe
            self.execute("G38.2 Z#<_ini[TOOLSENSOR]MAXPROBE> F#<_ini[TOOLSENSOR]SEARCH_VEL>")

            # Wait for results
            yield INTERP_EXECUTE_FINISH

            # FIXME if there is an error it never comes back
            # which leaves linuxcnc in g91 state
            if self.params[5070] == 0 or self.return_value > 0.0:
                self.execute("G90")
                self.set_errormsg("tool_probe_m6 remap error:")
                yield INTERP_ERROR

            self.execute("G38.4 Z#<_ini[TOOLSENSOR]REVERSE_LATCH> F#<_ini[TOOLSENSOR]SEARCH_VEL>")
            
            # Wait for results
            yield INTERP_EXECUTE_FINISH
            
            # FIXME if there is an error it never comes back
            # which leaves linuxcnc in g91 state
            if self.params[5070] == 0 or self.return_value > 0.0:
                self.execute("G90")
                self.set_errormsg("tool_probe_m6 remap error:")
                yield INTERP_ERROR
 
            self.execute("G1 Z#<_ini[TOOLSENSOR]TS_LATCH> F#<_ini[TOOLSENSOR]SEARCH_VEL>")  
            # Wait for final retract
            yield INTERP_EXECUTE_FINISH             

            self.execute("G38.2 Z-[#<_ini[TOOLSENSOR]TS_LATCH>*2] F#<_ini[TOOLSENSOR]PROBE_VEL>")   
            
            # Wait for final results
            yield INTERP_EXECUTE_FINISH
            
            # FIXME if there is an error it never comes back
            # which leaves linuxcnc in g91 state
            if self.params[5070] == 0 or self.return_value > 0.0:
                self.execute("G90")
                self.set_errormsg("tool_probe_m6 remap error:")
                yield INTERP_ERROR             
                
            # set back absolute state
            self.execute("G90")

            # return to recorded positon
            self.execute("G53 G0 Z[#4997]")
            yield INTERP_EXECUTE_FINISH
            self.execute("G53 G0 X[#4999] Y[#4998]")

            # adjust tool offset from calculations
            proberesult = self.params[5063]
            probeheight = self.params["_ini[TOOLSENSOR]SETTER_HEIGHT>"]  # TODO using global_var from some interface allow to set and save this value using probe or by hand
            workheight = self.params[5000]                               # Need your opinion vs ver's solution   

            adj = proberesult - probeheight
            self.execute("G10 L1 P#<selected_tool> Z{}".format(adj))
            
            adj = workheight
            self.execute("G10 L2 P0 Z{}".format(adj))

            # apply tool offset
            self.execute("G43")
 
        except InterpreterException as e:
            msg = "%d: '%s' - %s" % (e.line_number,e.line_text, e.error_message)
            print (msg)
            yield INTERP_ERROR

    except Exception as e:
        print (e)
        self.set_errormsg("tool_probe_m6 remap error: %s" % (e))
        yield INTERP_ERROR

```

alkabal, 2021-01-05:
This one need to be tested !

```
# REMAP=M6 modalgroup=10 python=tool_probe_m6
#
# auto tool probe on m6
# move to Z max - configurable offset
# move to tool change position for toolchange
# backup offset and position
# wait for acknowledge of tool change
# move to tool setter probe position
# probe tool on tool setter
# move back to tool change position
# set tool offsets
# move to Z max - configurable offset
# goto position backup
# based on Versaprobe remap
#
# param _backup_offset holds the work piece height
#
# required INI settings
# (Abs coordinates/ machine based units)
#
#[CHANGE_POSITION]
#Z_SAFE_TRAVEL_OFFSET = -0.2
#X = 0
#Y = 0
#Z = 0

#[TOOLSENSOR]
#X = -5.00
#Y = -1
#Z = -1

#SETTER_HEIGHT = 40
#MAXPROBE =  -3
#TS_LATCH = 0.3
#REVERSE_LATCH = 2
#SEARCH_VEL = 300
#PROBE_VEL = 10

# What about checking machine is homed before allow this?
# TODO ver's use a configurable setterheight that can be set manually or using probe

def tool_probe_m6(self, **words):

    # only run this if we are really moving the machine
    # skip this if running task for the screen
    if not self.task:
        yield INTERP_OK

    try:
        self.params["tool_in_spindle"] = self.current_tool
        self.params["selected_tool"] = self.selected_tool
        self.params["current_pocket"] = self.current_pocket
        self.params["selected_pocket"] = self.selected_pocket

        # cancel tool offset
        self.execute("G49")
        
        # change tool where ever we are
        # user sets toolchange position prior to toolchange
        # we will return here after

        try:
            self.selected_pocket =  int(self.params["selected_pocket"])
            emccanon.CHANGE_TOOL(self.selected_pocket)
            self.current_pocket = self.selected_pocket
            self.selected_pocket = -1
            self.selected_tool = -1
[... 113 lines cut ...]
            adj = workheight
            self.execute("G10 L2 P0 Z{}".format(adj))                    # restore G5x offset after correct tool measurement

            # apply tool offset
            self.execute("G43")

            self.execute("G53 G0 Z[#<_ini[AXIS_Z]MAX_LIMIT>+#<_ini[PROBE_SCREEN]Z_SAFE_TRAVEL_OFFSET>]")
            yield INTERP_EXECUTE_FINISH
            
            # return to recorded positon
            self.execute("G0 X[#4999] Y[#4998]")
            yield INTERP_EXECUTE_FINISH
            self.execute("G0 Z[#4997]")
            #self.execute("G0 Z[#4997-#<_ini[PROBE_SCREEN]Z_SAFE_TRAVEL_OFFSET>]")
 
        except InterpreterException as e:
            msg = "%d: '%s' - %s" % (e.line_number,e.line_text, e.error_message)
            print (msg)
            yield INTERP_ERROR

    except Exception as e:
        print (e)
        self.set_errormsg("tool_probe_m6 remap error: %s" % (e))
        yield INTERP_ERROR
```

alkabal, 2021-01-05:
well this one is more near to work

```
# REMAP=M6 modalgroup=10 python=tool_probe_m6
#
# auto tool probe on m6
# move to Z max - configurable offset
# move to tool change position for toolchange
# backup offset and position
# wait for acknowledge of tool change
# move to tool setter probe position
# probe tool on tool setter
# move back to tool change position
# set tool offsets
# move to Z max - configurable offset
# goto position backup
# based on Versaprobe remap
#
# param _backup_offset holds the work piece height
#
# required INI settings
# (Abs coordinates/ machine based units)
#
#[CHANGE_POSITION]
#Z_SAFE_TRAVEL_OFFSET = -0.2
#X = 0
#Y = 0
#Z = 0

#[TOOLSENSOR]
#X = -5.00
#Y = -1
#Z = -1

#SETTER_HEIGHT = 40
#MAXPROBE =  -3
#TS_LATCH = 0.3
#REVERSE_LATCH = 2
#SEARCH_VEL = 300
#PROBE_VEL = 10

# What about checking machine is homed before allow this?
# TODO ver's use a configurable setterheight that can be set manually or using probe

def tool_probe_m6(self, **words):

    # only run this if we are really moving the machine
    # skip this if running task for the screen
    if not self.task:
        yield INTERP_OK

    try:
        self.params["tool_in_spindle"] = self.current_tool
        self.params["selected_tool"] = self.selected_tool
        self.params["current_pocket"] = self.current_pocket
        self.params["selected_pocket"] = self.selected_pocket

        # cancel tool offset
        self.execute("G49")
        
        # change tool where ever we are
        # user sets toolchange position prior to toolchange
        # we will return here after

        try:
            self.selected_pocket =  int(self.params["selected_pocket"])
            emccanon.CHANGE_TOOL(self.selected_pocket)
            self.current_pocket = self.selected_pocket
            self.selected_pocket = -1
            self.selected_tool = -1
[... 114 lines cut ...]
            # apply tool offset
            self.execute("G43")

            self.execute("G53 G0 Z[#<_ini[AXIS_Z]MAX_LIMIT>+#<_ini[CHANGE_POSITION]Z_SAFE_TRAVEL_OFFSET>]")
            yield INTERP_EXECUTE_FINISH
            
            adj = workheight
            self.execute("G10 L2 P0 Z{}".format(adj))                    # restore G5x offset after correct tool measurement
            
            # return to recorded positon
            self.execute("G53 G0 X[#4999] Y[#4998]")
            yield INTERP_EXECUTE_FINISH
            #self.execute("G53 G0 Z[#4997]")                             # TODO something is wrong saved value is ok but when we go here now the value is something like 62.676 so this is the distance from tool to TS
            #self.execute("G53 G0 Z[#4997-#<_ini[CHANGE_POSITION]Z_SAFE_TRAVEL_OFFSET>]")
 
        except InterpreterException as e:
            msg = "%d: '%s' - %s" % (e.line_number,e.line_text, e.error_message)
            print (msg)
            yield INTERP_ERROR

    except Exception as e:
        print (e)
        self.set_errormsg("tool_probe_m6 remap error: %s" % (e))
        yield INTERP_ERROR
```

alkabal, 2021-01-05:
Restoring Z position is at this momennt imo not a good idea because result can differ if we come from tool 0 with only a G5x offset applied 
or also if we come from Toolchange position and if the tool height is highest the machine can't go to starting point.

c-morley, 2021-01-05:
So after doing some more testing -  linuxcnc does honour [EMCIO]TOOL_CHANGE_POSITION.
http://linuxcnc.org/docs/devel/html/config/ini-config.html#_emcio_section
Not sure what I did before.
I don't think we should reinvent the wheel with [CHANGE_POSITION].

Don't think we need to test for homing - one can't issue MDi commands without homing unless one specifically disabled that behavior in the INI. If you disable 'must home' and are using auto probing you are taking chances by choice.

Why do you need a run-time configurable setter height?

I think you are asking about a  (software) switch to not auto probe if desired? 

please explain some more:
>Restoring Z position is at this momennt imo not a good idea because result can differ if we come from tool 0 with only a G5x >offset applied
>or also if we come from Toolchange position and if the tool height is highest the machine can't go to starting point.

These should be G53 (machine coordinate) moves so tool offset doesn't matter. 
The tool goes to tool change point after tool probe - the exact same point it cam from.

Thanks

alkabal, 2021-01-05:
You do not have problem for use TOOL_CHANGE_QUILL_UP ? For me with or without this the 3 axis move at same time 

Oh yes you are right for homing I have do that in my config file and this is something temporary.

For the runtime setter I have reused the stuff like vers probe so if you have only a setter you can configure the height manually, if you have a touch probe you can do this with GUI, both way = value saved in the pref file 
maybe the best option is to allow both situation, 
I like the idea to keep the full use of vers screen and use new python remap method but I can keep the vers system and do not try to synch with your implementing.

I’m unsure about the best choice to do.

alkabal, 2021-01-05:
Yes the vers remap can be inhibited in the GUI and I think it’s a good option.

For restoring Z I need to do other test I have encountered some issue with first attempt that need to be done again.

c-morley, 2021-01-05:
tool_change_quill_up is not implemented in master yet - it is in my working branch. same as spindle off.

As for the toolsetter height - I guess I don't understand how it's used as I can't see why you would be changing it often AIUI that is the reference height so it should always be the same. so the INI file would be appropriate.

if you request no tool probe does the G43 offset still get applied?

alkabal, 2021-01-05:
Well

I use the 2.8 branch with copy past from your stglue, it is something normal that these feature does not work ?

Vers allow with touch probe to touch the table, after that to probe toolsetter for determinate the height I can understand why you found this strange but ???

Yes i restore G43 also without measurement the idea is also to be possible to only create tooltable with routine and if machine is enough accurate for some job you can bypass measurement, I need to agree this is not standard but i think doing somztging flexible

alkabal, 2021-01-05:
i try to isolate the third value from ini entry, have you any idea ?

            temp = (_ini[EMCIO]TOOL_CHANGE_POSITION[2])
            print(temp)
            self.execute("G53 G0 Z{:.5f}".format(temp))                                   #need to use third value

alkabal, 2021-01-06:
https://github.com/alkabal/linuxcnc/tree/patch-4

What did you think of that ?

That seem to work fine

alkabal, 2021-01-06:
adding this in machine unit allow to no more switch from G20 G21
TS_LATCH = 0.3
REVERSE_LATCH = 2  (only if we use G38.4)

alkabal, 2021-01-06:
I do not understand why this line is not executed 
                  self.execute("G1 Z{:.5f}".format(Z))                   # Value ar ok but this code seem to be not executed

https://github.com/alkabal/linuxcnc/blob/90d44b0bd34a1b77eee515d5fc85f84ec337e2aa/nc_files/remap_lib/python-stdglue/stdglue.py#L685

I hope you can check what i try to do mixing your idea with my idea.

c-morley, 2021-01-06:
>adding this in machine unit allow to no more switch from G20 G21
>TS_LATCH = 0.3
>REVERSE_LATCH = 2 (only if we use G38.4)

My idea is to do all probing in machine units. 
I check for current units compared to machine units. switch to machine units for probing.
after probing switch back to original units.

>I use the 2.8 branch with copy past from your stglue, it is something normal that these feature does not work ?
No some of the latest stuff (such as TOOL_CHANGE_QUILL_UP) is not in 2.8

>Vers allow with touch probe to touch the table, after that to probe toolsetter for determinate the height I can understand why you >found this strange but ???

If you reference the tool setter to the table. then you only need to do it once and enter it in the INI.
After that the table and the tool setter height don't change.
I don't want to use a bunch of params unless they are really needed.

I would like to keep things simple and general in the beginning then go from there.

alkabal, 2021-01-06:
Thanks a lot, now i think i understand why you wan't to keeep the switchUnitsFlag if user change current unit the value from ini are not changed so this is a possible isssue.  Many thanks for pointing that.

For TS height i agree with you, but this need to be changed in the fork PSNG if keep in synch is possible. 
For now i'm not sure about the good solution to choose, PSNG logic, your stdglue logic, my own logic or better something mixed and synched (but 3 variant and i suspect all of us wan't to keep own logic...)

Any idea why this line seem to be not executed ?
https://github.com/alkabal/linuxcnc/blob/90d44b0bd34a1b77eee515d5fc85f84ec337e2aa/nc_files/remap_lib/python-stdglue/stdglue.py#L685

Thanks

alkabal, 2021-01-06:
updated link for line not executed
https://github.com/alkabal/linuxcnc/blob/4206a98e58655c872e114d9ccab726d76fbef0a5/nc_files/remap_lib/python-stdglue/stdglue.py#L736

alkabal, 2021-01-06:
I have added logic for check the initial G90 G91 status and for restoring

Possible solution for manage error from G38.2 in the comment

Save restore offset separately than tool offset

Try to restore initial compensated position if first tool is not 0 but this does not work
https://github.com/alkabal/linuxcnc/blob/2023af79f88eeea12a00444221461497d3d608bd/nc_files/remap_lib/python-stdglue/stdglue.py#L763

closed 2021-01-06 by alkabal