LinuxCNC/linuxcnc #1034: Question about stdglue coding style

alkabal, 2021-01-08:
Hi

Just for be sure what did you think is better or is standard to use : 
```
def setspeed_prolog(self,**words):
    try:
        c = self.blocks[self.remap_level]
        if not c.s_flag:
            self.set_errormsg("S requires a value") 
            return INTERP_ERROR
        self.params["speed"] = c.s_number
    except Exception,e:
        self.set_errormsg("S/setspeed_prolog: %s)" % (e))
        return INTERP_ERROR
    return INTERP_OK
```

```
def setspeed_prolog(self,**words):
    try:
        c = self.blocks[self.remap_level]
        if not c.s_flag:
            self.set_errormsg("S requires a value")
            return INTERP_ERROR
        self.params["speed"] = c.s_number
        return INTERP_OK
    except Exception as e:
        self.set_errormsg("S/setspeed_prolog: %s)" % (e))
        return INTERP_ERROR
```

andypugh, 2021-01-10:
I can't really speak for which is the most "pythonic" but I prefer the second version.

alkabal, 2021-01-10:
Hi

Thanks a lot, so imo with your opion that nee to be changed in stglue for some entry
https://github.com/LinuxCNC/linuxcnc/blob/587781fcf75a8061b827dfde2634293bde17d278/nc_files/remap_lib/python-stdglue/stdglue.py#L33
and line : 55 70 89

Something is don't understand when i have try to add return INTERP_OK here, i get error
https://github.com/LinuxCNC/linuxcnc/blob/587781fcf75a8061b827dfde2634293bde17d278/nc_files/remap_lib/python-stdglue/stdglue.py#L197
https://github.com/LinuxCNC/linuxcnc/blob/587781fcf75a8061b827dfde2634293bde17d278/nc_files/remap_lib/python-stdglue/stdglue.py#L247

andypugh, 2021-01-11:
No, we tend to discourage code changes that are purely to correct coding style and (non functional) whitespace. 

If code is working properly it is best not to change it, if only to keep the commit history easier to follow.

alkabal, 2021-01-11:
ok i understand, thanks !

closed 2021-01-11 by alkabal