LinuxCNC/linuxcnc #1045: PNCconf with Mesa7i96 and 7i73 writes wrong 7i73 references to hal file

dieter-l-git, 2021-01-17:
I'm using 7i96 with 7i73 in Mode1 via Serial interface

When I configure an Input/Output or ENC of the 7i73, Pncconf writes 
hm2_7i96.0.7i73.0.**1**.xxx instead of hm2_7i96.0.7i73.0.**0**.xxx
to the hal file.

_Sample:_
Output from Pncconf:
net joint-select-a     <=  hm2_7i96.0.7i73.0.**1**.input-00

Should be:
net joint-select-a     <=  hm2_7i96.0.7i73.0.**0**.input-00

On the machine monitor all 7i73 pins are shown as 0.0., not as .0.1.

Same issue with 2.8.1 and 2.9

c-morley, 2021-01-21:
Could you post mesaflash's --readhmid print out?

dieter-l-git, 2021-01-21:
Done.

[readhmid.txt](https://github.com/LinuxCNC/linuxcnc/files/5852268/readhmid.txt)

dieter-l-git, 2021-02-01:
[sserial.txt](https://github.com/LinuxCNC/linuxcnc/files/5906137/sserial.txt)

added file with details from --sserial

dieter-l-git, 2021-02-01:
[sserial.txt](https://github.com/LinuxCNC/linuxcnc/files/5906151/sserial.txt)

output from  --sserial

c-morley, 2021-02-12:
Sorry for the delay.
I would guess you used pncconf's discovery option to configure the 7i96?
What version of lunuxcnc did you use to get the broken config files - when I tried master I could not get the 7i73 pins to be written to the HAL file at all.
I think I have a fix - I just need to test 2.8 yet

c-morley, 2021-02-12:
Ok I see that pncconf does cover the 7i96 with internal data.
I pushed a fix to 2.8/master so this should be right with internal data or with the discovery option.
If you could try it and confirm that would be great.

grandixximo, 2026-10-05:
Fixed by af1b37aa60 (2021-02-11): pncconf's internal 7i96 data reported the sserial channel as 1 instead of 0, so 7i73 pins came out as hm2_7i96.0.7i73.0.1.xxx. In 2.8.2 and later. Closing.

closed 2026-10-05 by grandixximo

dieter-l-git, 2026-11-16:
Thanks,
will do testing in the next few weeks.