LinuxCNC/linuxcnc #1048: Possible issue for aborting homing in progress

alkabal, 2021-01-19:
Hi

Using 2.8.1 RIP

Well like always i have a possible special issue unfortunately difficult to explain and reproduce without some code.

Using pin joint.0.homing connected to orient component for ask spindle homing when i ask machine homing
Also using motion.probe-input connected to orient for prevent all spindle move if probe is connected

So if the probe is connected the comp file call a MDI command
(ABORT,Program or Spindle or Orient is stopped due to 3D probe connected) 

Until here all work fine (even if i'm sure they are better way for "abort" from comp... (rtapi_print_msg(RTAPI_MSG_ERR seem to not do the job.)

But checking the signal from halcmd joint.0.homing i see he stay True after aborting, so at this point asking homing again i get 
 message like "homing sequence already in progress" 
Of course my test from pin joint.0.homing status did no stop the process anymore.

Like you know i'm not expert at all, but i suspect something can be patched for better homing aborting
Related occurence is here but i have no idea of what can be done for that
https://github.com/LinuxCNC/linuxcnc/blob/a8541d68fa72a79b16196e5aba91d14bb8665945/src/emc/motion-logger/motion-logger.c#L248
https://github.com/LinuxCNC/linuxcnc/blob/43ac9730f4d6be679bd28292d88fdfcdb8204907/src/emc/motion/homing.c#L242 
https://github.com/LinuxCNC/linuxcnc/blob/32a8f839f6f973b9a4381df61811a96cbadbe99b/src/emc/motion/command.c#L1460

[image]

Br

andypugh, 2021-01-19:
I think that using a HAL pin connection to toggle halui.abort would probably be a better way to abort motion. 
(And send the message by a different method) 

I agree that abort should reset homing state, however,

alkabal, 2021-01-19:
Thanks for the tips about halui.abort you are right i use this one for my lathe toolchanger but i did not remenber that.

From "long time" i have choose to use (MSG,) (DEBUG,) and (ABORT,) for replacing rtapi_print_msg (many time lost using machinekit and BBB with GUI that miss the use of rtapi_print..), so i don't know if all GUI for Linuxcnc work fine with rtapi_print_msg.

Thank you

alkabal, 2021-01-21:
I have do some test with halui.abort but this is more difficult to make it work correctly (need to be toggled like you says) and trying to send message at same time is also really difficult I have check the code of my lathe toolchanger and remember that I have fight and found workaround for get error mode working correctly.

I presume I have write without knowing the existence of (ABORT,message)  this was my first coding for computer use and cnc i think now it need to be reviewed...

So for the moment my opinion is : this is really more easy and accurate to use (ABORT,message) except with homing in progress. (And maybe other unknown situation)

If this is something that can be patched for correct homing state reset this is great ! but I can understand if nobody wan’t to check that (for now I think this is outside my possibility)

Br

alkabal, 2021-02-20:
Hi

Any idea for pointing out where to check the code ?

I have no idea where to start for try to patch the code.

Br

andypugh, 2021-02-26:
It looks like abort is handled here:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/task/emctaskmain.cc#L2095

HOME_ABORT is one of the stages in homing:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/motion/homing.c#L1246
(and it seems to be resetting a homing flag)

The joint.N.homing pin is probably created by some magic here:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/task/taskmodule.cc#L381

I think I would probably start at the other end, search the code for the _exact_ message that is displayed when you try to home after aborting.

alkabal, 2021-02-26:
Thanks a lot for reply.

I suspect when we ask abort from mdi_command (ABORT, ) with the homing started the "case HOME_ABORT" is not called

The exact error message is this one :
https://github.com/LinuxCNC/linuxcnc/blob/2388839f524a28207291ed1f4c82905fd0018a13/src/emc/motion/command.c#L1487

closed 2022-12-25 by andypugh