LinuxCNC/linuxcnc #1061: HAL_GLIB Task Saturation

neilwhelchel, 2021-01-28:
I have been using Gmoccapy for many years on my 3 axis mill without issue until I added a fourth axis to handle a rotary table that I setup as a pluggable accessory. This change broke the display for Gmoccapy. It loads a blank screen with only the Gremlin view working, but very laggy. Clicking on the blank screen where the buttons should be sometimes causes the expected reaction, sometimes not. Just for debugging, I tried the Axis interface, and it was even worse, blank screen except for Gremlin, and clicking where the buttons should be does nothing. I attached a debugger to glib, and I found that it is task saturated with the timer callbacks, so it never has enough CPU time to render the graphics before the next timeout. I looked, and the callback time is hard-coded to 100. 

hal_glib.py:243 GObject.timeout_add(100, self.update)

I changed this to 200 just for testing, and problem solved, it works fine with this change.

I looked at the Gmoccapy code, and it uses "CYCLE_TIME" from the [DISPLAY] section of the .ini file for its update task rather than being hard-coded. Maybe passing this into hal_glib will allow the user to tweak things for slower systems. Not sure if it is a good idea or not. (Also, there are a few timer callbacks that also default to 100 hal_glib.py for the HAL pins that I did not look at.)

It is clear that timeout_add() sits higher in the priority queue in glib than rendering the graphics. I would suggest that there needs to be a way to monitor this for saturation. One possible approach is to add a second callback via idle_add() which seems to be called by glib after the graphics are done rendering. Then a flag can be used to check if the callback from idle_add() has been called since update() was last called. (Maybe allow for a small amount of misses before a critical error is raised. But make it at least a warning if it is missed even once.) As an alternative, a PID or similar calculation can be used to automatically set the cycle time. The amount of idle time can be measured from the time that the idle_add() callback is done and the timeout_add() callback is done. (This should raise a warning if the cycle time gets slower than a particular rate.) This would make a system that self tunes to the available CPU cycles.

Either way, it is clearly a bug if the CPU gets saturated to the point that the GUI becomes unresponsive, and considering what this program does, it needs to be checking for task saturation at the very least.

I tested this with the latest version from GIT, as well as 2.8.0 and 2.8.1, all versions had the same issue. 
The computer is based on an Intel Atom D510 @ 1.66GHz with plenty of memory, and swap disabled.
The hardware interface is the Mesa FPGA card.
(Sorry for not following the normal bug report format, but since this has the bug pinned down, I did not think that was the optimal reporting method.)

c-morley, 2021-01-28:
My instinct says that something else happened when you added the 4th axis.
It doesn't seem reasonable that adding an axis suddenly stalls a display that worked fine before.
I would like to know more about what you mean by 'added an axis' - did you modify Gmoccapy to do this?
AXIS doesn't use GLIB so that needs explaining too.

That being said there is nothing wrong with the idea of configurable update. It could help with low power machines.
The gremlin plot update is actually has a 50 ms update internally as well as a 100ms thread running in the background.
HAL pins using Gladevcp's GPin updater have a separate 100 ms update

c-morley, 2021-01-28:
If the screen gets saturated it will not likely crash a running linuxcnc machine. the realtime task is separate.
I don't think one needs to add monitoring - it's pretty self evident if there is a problem - and if configurable easy to fix.

neilwhelchel, 2021-01-28:
I was a bit skeptical as well. I did some very careful debugging to arrive at the conclusion I discussed above. I went so far as to build a second computer using the exact same parts. I did a from scratch install, I rested it with the Gmoccapy simulator files included with the release with 3 axis, works fine, but fails with the 4 axis interface. I changed the callback timeout to 200 as listed above and it works. It actually works fine if I set it to 110, it is hit and miss at 105, complete fail at 100. When I use the 5 axis example, it has to be at about 120 to work reliably. 
I disagree about monitoring the GUI thread for one critical reason... The jog buttons are in the GUI. If the GUI gets behind, the buttons become really sluggish at responding. This means that when you take your finger off of the button, it might take several seconds to stop jogging. By then your tool is likely to be broken if you are lucky that is all the damage you did... In many cases, there is just not time to get to the EStop button.
I have my test setup available on-line if anyone wants to play with it via SSH/VNC.

c-morley, 2021-01-28:
So to be clear you see this on a non modified simulated config too? interesting.

What i mean about monitoring is it's not a thing that comes and goes - if it's a problem you fix it and that's the end of it.
No sense in monitoring after it's fixed. I suppose a test to confirm the problem may be handy.

Your changes won't help AXIS - the hal_glib is used in gladevcp, gscreen. gmoccapy and qtvcp.
AXIS already honours the INI  CYCLE time setting.

neilwhelchel, 2021-01-28:
I edited my post above, I added screenshots. The screenshots above are from the unmodified sim file gmoccapy_4_axis.ini, the 3 axis sim works fine. I am thinking that the extra axis of overhead is just enough to push timing out of bounds, as demonstrated by adding 20 ms to make it work.
The reason I went so far as to suggest making some sort of an auto timing feature, is that given the way that it works now, there is no way to tell how much timing margin you have. If it is sitting right on the edge, it may be unnoticed for a very long time, until someone is up burning the midnight oil and crond fires off a 2 AM maintenance task that causes the interface to lock up right when they are doing a setup. With the way that the software currently works, when a new machine is built, the only way you know how much timing overhead you have is to back it off to failure, and add some back in. There is no documentation for this, so in reality, how many people do you think are sitting on this edge? Also, frankly, even if it were documented, I likely would have missed it for sure when I added the config to my machine for the extra axis. Given that there is no error message, or warning, it was difficult for me to find. So even if it is in the ini file, it still could be an issue unless the software has some check for this as I suggested above.

c-morley, 2021-01-28:
It's an interesting problem. I don't doubt your observation and tests.
At the very least we should be able to make it configurable with little effort.

c-morley, 2021-01-28:
Ok I just added code in master to hal_glib to honour CYCLE_TIME which can actually be in ms or seconds!
Let me know if that is a step in the right direction.

neilwhelchel, 2021-01-28:
I just pulled, built, and tested. With a minor tweak to CYCLE_TIME in the .ini file, it works as it should. So definitely a good step.
Im my opinion another good step would be some documentation regarding CYCLE_TIME in general that is slightly more descriptive than all I was able to find "Cycle time in seconds that display will sleep between polls." I am not sure that this is enough to relay to the machine builder what this is really doing. Also, it is misleading as it specifically says that it is the "sleep time". Which if it were, we would not be discussing this. It is actually the update interval, and the actual sleep time depends on how fast all of the code in update() runs.

In that point, it would be trivial to convert it to sleep time, just by making update() return False, and just before returning, a fresh call to timeout_add(). The same could be done in the upper layers. This approach would be immune to saturation/thrashing regardless of system speed and userspace load.

c-morley, 2021-01-28:
I'll see about adding some more info into docs.
As for the docs being wrong - it probably was/is right for some screens.
Nothing audits linuxcnc to see if it actually follows conventions (nor is there really any official conventions - on screens at least)
I'm not interested in coding a change to sleep vrs update, but if a patch was submitted I see no immediate reason not to consider it.

neilwhelchel, 2021-01-29:
I will look into making a patch, but it is going to take some time as I have to become (more) familiar with all of the pieces involved. That is why I did not submit a patch right off, I was attempting to consult with people that are more familiar with the big picture than I am. Technically, c-morley's patch is a confirmed fix for the problem as it is reported, so maybe this ticket can/should be closed. I would ask that the project maintainers please decide this. However, it is still subject to a race condition should other userspace loads pop up. My opinion is that having any GUI that is subject to racing or thrashing based on system load is clearly a bug, and in the case of a safety related GUI, I think it is unacceptable, so I am happy to work on a solution for it.

c-morley, 2021-01-29:
Sounds good. please keep in mind that gscreen, gmoccapy, gladevcp and qtvcp use hal_glib (though qtvcp uses it's own gobject timer), so if patching, please test them to confirm they all still function.

Thank you for your report and discussion.

closed 2021-01-29 by c-morley