LinuxCNC/linuxcnc #1067: tool_mmap_user(): file open fail: No such file or directory (when using QtDesigner in master)

c-morley, 2021-02-06:
I see  commit 2dbb2f and the commit message seems to indicate that maybe it was committed to master by mistake.

When launching designer with  designer -qt=5 in a terminal it fails with the above error.

dngarrett, 2021-02-06:
1) I haven't been able to reproduce the message using a
rip build on debian buster:

$ uname -a
Linux t520 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5 (2019-06-19) x86_64 GNU/Linux

$ lsb_release -idrc
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

2) I have never used qt before and could not find the
package requirements where expected in the debian control files
in the debian  directory:

Ref:
https://github.com/LinuxCNC/linuxcnc/tree/master/debian

$ pwd
/home/git/linuxcnc-dev/debian
$ grep -i qt control*
$ 
(no references to qt packages)

3) So i installed qt stuff using a script i found on the forum
Refs:
https://forum.linuxcnc.org/qtvcp/38414-installing-qtvcp?start=0#157829
https://forum.linuxcnc.org/media/kunena/attachments/6129/qtvcp_install_2021-01-05-2.txt

4) after installing qt,
tried command designer -qt=5. it seemed to work and the
example sim qtdefault seemed to startup:

Video:
https://www.youtube.com/watch?v=A4ENiMXfh1Y&feature=youtu.be

5) more information is needed to indicate system and conditions
needed to show the reported message.

Suggest posting output from the command: linuxcnc_info

c-morley, 2021-02-06:
QT screens seem to load fine.
Qtdesigner does not - has error message as posted.
My guess is you don't have designer quite set up properly. - do you see the linuxcnc widgets ? I couldn't see from your video.

Tried Mint 19 and Debian 9 - wouldn't think that would matter.
Again my guess is a default-is-to-open-file problem - I am unsure why that is a problem when designer is used but I do play some tricks to get the graphics display to work without linuxcnc.

output from linuxcnc_info:

VIEWER=gedit
                Date: Sat Feb 6 11:14:28 PST 2021
            UTC Date: Sat Feb 6 19:14:28 UTC 2021
        this program: /home/chris/emc/scripts/linuxcnc_info
              uptime: 11:14:28 up 25 min, 1 user, load average: 0.07, 0.36, 0.50
     lsb_release -sa: Debian Debian GNU/Linux 9.12 (stretch) 9.12 stretch
      which linuxcnc: /home/chris/emc/scripts/linuxcnc
                 pwd: /home/chris/emc/src
                USER: chris
             LOGNAME: chris
                HOME: /home/chris
              EDITOR: 
              VISUAL: 
            LANGUAGE: 
                TERM: xterm-256color
           COLORTERM: truecolor
             DISPLAY: :0.0
             DESKTOP: lightdm-xsession
        display size: 1366x768 pixels (361x203 millimeters)
                PATH: /home/chris/emc/bin:/home/chris/emc/scripts:/home/chris/emc/tcl:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games

uname items:
         nodename -n: chris-laptop2
      kernel-name -s: Linux
      kernel-vers -v: #1 SMP PREEMPT RT Debian 4.9.210-1+deb9u1 (2020-06-07)
          machine -m: x86_64
        processor -p: unknown
         platform -i: unknown
      oper system -o: GNU/Linux

/proc items:
             cmdline: BOOT_IMAGE=/boot/vmlinuz-4.9.0-12-rt-amd64 root=UUID=174cdff9-c154-4518-804c-588cadc97716 ro quiet
          model name: Intel(R) Core(TM) i5-2520M CPU @ 2.50GHz
               cores: 2
             cpu MHz: 2499.847
             parport: 
              serial: 

Versions:
                 gcc: gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516
              python: Python 2.7.13
                 git: git version 2.11.0
          git commit: f70ef7729
                 tcl: 8.6
                  tk: 8.6
               glade: not_in_PATH
          glade-gtk2: glade3 3.8.0

linuxcnc_var all:

     LINUXCNCVERSION: 2.9.0~pre0
LINUXCNC_AUX_GLADEVCP: /usr/share/linuxcnc/aux_gladevcp
LINUXCNC_AUX_EXAMPLES: /usr/share/linuxcnc/aux_examples
            REALTIME: /home/chris/emc/scripts/realtime
                 RTS: uspace
          HALLIB_DIR: /home/chris/emc/lib/hallib
              PYTHON: /usr/bin/python2

dpkg -l '*linuxcnc*':
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                Version                      Architecture Description
+++-===================-============================-============-=====================================================================
un  linuxcnc            <none>                       <none>       (no description available)
un  linuxcnc-dev        <none>                       <none>       (no description available)
un  linuxcnc-doc        <none>                       <none>       (no description available)
ii  linuxcnc-doc-en     1:2.9.0~pre0.2311.gbb5d6a60a all          motion controller for CNC machines and robots (English documentation)
un  linuxcnc-sim        <none>                       <none>       (no description available)
un  linuxcnc-sim-dev    <none>                       <none>       (no description available)
ii  linuxcnc-uspace     1:2.9.0~pre0.2311.gbb5d6a60a amd64        motion controller for CNC machines and robots
ii  linuxcnc-uspace-dev 1:2.9.0~pre0.2311.gbb5d6a60a amd64        PC based motion controller for real-time Linux

dngarrett, 2021-02-06:
> My guess is you don't have designer quite set up properly.
> - do you see the linuxcnc widgets ? I couldn't see from your video.

As mentioned, i have never used qt and don't know what a linuxcnc widget looks like.
Questions:
1) Where are the instructions for setting up designer "properly"?
2) What operating systems are supported by qtvcp?
   (There are no qt packages listed in the git debian directory control.* files
    so it's hard to tell what operating systems are supported)
3) I have the following packages installed for buster after using the script from
the forum mentioned above:
https://forum.linuxcnc.org/media/kunena/attachments/6129/qtvcp_install_2021-01-05-2.txt

$ dpgk -l '*qt*' |grep ii
ii  libqscintilla2-qt5-13           2.10.4+dfsg-2.1        amd64        Qt5 port of the Scintilla source code editing widget
ii  libqscintilla2-qt5-l10n         2.10.4+dfsg-2.1        all          Scintilla source code editing widget for Qt5, translation files
ii  libqt4-dbus:amd64               4:4.8.7+dfsg-18        amd64        Qt 4 D-Bus module
ii  libqt4-network:amd64            4:4.8.7+dfsg-18        amd64        Qt 4 network module
ii  libqt4-opengl:amd64             4:4.8.7+dfsg-18        amd64        Qt 4 OpenGL module
ii  libqt4-script:amd64             4:4.8.7+dfsg-18        amd64        Qt 4 script module
ii  libqt4-xml:amd64                4:4.8.7+dfsg-18        amd64        Qt 4 XML module
ii  libqt4-xmlpatterns:amd64        4:4.8.7+dfsg-18        amd64        Qt 4 XML patterns module
ii  libqt5concurrent5:amd64         5.11.3+dfsg1-1+deb10u4 amd64        Qt 5 concurrent module
ii  libqt5core5a:amd64              5.11.3+dfsg1-1+deb10u4 amd64        Qt 5 core module
ii  libqt5dbus5:amd64               5.11.3+dfsg1-1+deb10u4 amd64        Qt 5 D-Bus module
ii  libqt5designer5:amd64           5.11.3-4               amd64        Qt 5 designer module
ii  libqt5designercomponents5:amd64 5.11.3-4               amd64        Qt 5 Designer components module
ii  libqt5gamepad5:amd64            5.11.3-2               amd64        Qt 5 gamepad module
ii  libqt5gui5:amd64                5.11.3+dfsg1-1+deb10u4 amd64        Qt 5 GUI module
ii  libqt5help5:amd64               5.11.3-4               amd64        Qt 5 help module
ii  libqt5multimedia5:amd64         5.11.3-2               amd64        Qt 5 Multimedia module
ii  libqt5network5:amd64            5.11.3+dfsg1-1+deb10u4 amd64        Qt 5 network module
ii  libqt5opengl5:amd64             5.11.3+dfsg1-1+deb10u4 amd64        Qt 5 OpenGL module
ii  libqt5opengl5-dev:amd64         5.11.3+dfsg1-1+deb10u4 amd64        Qt 5 OpenGL library development files
ii  libqt5positioning5:amd64        5.11.3+dfsg-2          amd64        Qt Positioning module
ii  libqt5printsupport5:amd64       5.11.3+dfsg1-1+deb10u4 amd64        Qt 5 print support module
ii  libqt5qml5:amd64                5.11.3-4               amd64        Qt 5 QML module

dngarrett, 2021-02-06:
I've made a test branch that may address the message (i cannot duplicate on buster)

Please build and test the branch dgarr/tst  (at commit d6e9186e2)

c-morley, 2021-02-06:
Docs for qtvcp/designer:
http://linuxcnc.org/docs/devel/html/gui/qtvcp.html#_get_designer_to_include_linuxcnc_widgets

README in source:
https://github.com/LinuxCNC/linuxcnc/blob/master/lib/python/qtvcp/designer/README.txt

Qtvcp supports any system with qt5 and linuxcnc (py2 and py3)
Tested on mint 18-20,  Debian 9 and I'm sure I tried Debian 10.
Python 3 is least tested and broken from time to time when incompatible code slips in.

I know almost nothing about debian control files - someone was going to add the requirements but apparently didn't.

in Designer after launch, select 'create' in the dialog now you can interact with the ui.
The righthand side shows the available widgets.
If you scroll down you should see three heading for custom widgets: Linuxcnc-HAL, linuxcnc-controller, linuxcnc-dialogs.
If you don't see the widgets - something is set up wrong. Some hints can usually be found in the terminal.

Your test branch tst did allow designer to load with this output in the terminal:
(first two lines are normal for designer)

('Qtvcp python plugin found:', '/home/chris/.designer/plugins/python/qtvcp_plugin.pyc')
[QtDesigner][INFO]  Logging to "/home/chris/qtdesigner.log" (logger.py:100)
tool_mmap_user(): file open fail: No such file or directory
tool_mmap_user no mmap file </tmp/.tool.mmap>
mmap tool data not available, continuing emc/usr_intf/axis/extensions/emcmodule.cc

dngarrett, 2021-02-06:
>Your test branch tst did allow designer to load with this
> output in the terminal:
>...

What operating system?

If the commit in the test branch allows the designer to load
what else needs to be addressed?

I don't have any interest in learning how to use qt or its designer.

c-morley, 2021-02-06:
I tested on mint 19

I think that would be all you need to do - if other problems show up I'll put a ticket in.

Well with about 5 seconds more work you could have told me if the widgets showed up - which is surely the reason you couldn't see the problem. You asked on how to cause the error so I added that so we could get the info we needed.
Like it or not qtvcp is part of linuxcnc.

By the way what is this code for? Did I miss a discussion or description email?
Your commit message made it sound like it was accidentally committed (the force push part)

Thank you for the quick fix - now I can fix my reported bugs :)

dngarrett, 2021-02-06:
Apologies, i don't see anything that looks like a linuxcnc widget using designer -qt=5 and debian buster.

If the debian control files don't specify the packages needed to use LinuxCNC facilities and configs, it is unlikely that users will be able to install a LinuxCNC deb package and expect them to work without additional package installs since related packages are not specified (or recommended) by the usual debian methods.

I will update the master branch soon with the commits from the test branch.
> By the way what is this code for? 
The primary commit message for 2dbb2f640fb87e describes in some detail

c-morley, 2021-02-06:
I have twice told developers (who asked me and said they would handle it ) who knew more then me on how to add what packages are needed for qtvcp - Andy was one.
More then likely he added just what was needed for the screen to display rather then allow editing them.
Is there docs you can recommend to help me figure out how to fix this problem?

Your new code still allows legacy tool handling but do you anticipate I will need to remake any tooloffset manipulating code I have for this new code you have done? The tooloffset code in gladevcp and qtvcp change the tool file directly, then has linuxcnc reload that file. Your description in the commit doesn't make how this now works clear to me.

It might have been nice to have a discussion in the maillist before committing so I could know if I need to get my code in line if required.

c-morley, 2021-02-06:
Actually after i thought if the problem more, Qtvcp uses non standard python2 libraries (for designer plugins) that no distribution supplies.
We supply precompiled versions in linuxcnc.
This is likely why it'd not found in debian control file.

When linuxcnc fully supports python3 we can fix this properly.

I'll close this issue now. Thaks again.

closed 2021-02-06 by c-morley

andypugh, 2021-02-06:
I think I would expect the qtvcp dependencies to be added here: 
https://github.com/LinuxCNC/linuxcnc/blob/master/debian/control.top.in
Probably in the linuxcnc-dev section. 

It's just a plain text file. 

If there are packages required that are served from a LinuxCNC repository, that's fine. But they need to be added to the repository indices and signed. (I can do that)

c-morley, 2021-02-06:
Thanks Andy.
I was sure you added these requirements when you made the buster ISO.
but that might have been so you can use the screens rather then edit the screens - I can't remember now.

I don't see any files that refer to GTK/GLADE (for a reference of what to include for qtvcp) in there so is there some other file too?
The precompiled files are just the files - we didn't make packages. I am hoping that python3-only happens before next release so that packaging is not needed.

andypugh, 2021-02-06:
In 2.8 the packages are referenced in control.bottom.in
https://github.com/LinuxCNC/linuxcnc/blob/2.8/debian/control.bottom.in
But that file does not seem to exist in Master. It seems that there has been some reorganisation. 
https://github.com/LinuxCNC/linuxcnc/commit/dd52b72e1692dc407c3ebc4453918d396267ba11
So the merge presumably failed to add those dependencies to the newly-named control.main-pkg.in

andypugh, 2021-02-07:
cb8056fedcef8da4a3764f7e3252ecfc376e99a6