LinuxCNC/linuxcnc #1070: Build LinuxCNC 2.8.1  with gcc 10

midyukov-anton, 2021-02-07:
Problem building with gcc10:
Linking classicladder
ld: objects/hal/classicladder/files_project.o:/usr/src/RPM/BUILD/linuxcnc-2.8.1/src/hal/classicladder/files_project.c:163: multiple definition of `FileName'; objects/hal/classicladder/files.o:/usr/src/RPM/BUILD/linuxcnc-2.8.1/src/hal/classicladder/files.c:1303: first defined here

andypugh, 2021-02-07:
Is this purely a gcc-10 problem?

midyukov-anton, 2021-02-07:
Distributiv ALT Linux Sisyphus. gcc-10.2.1. Patches for gcc:
https://packages.altlinux.org/en/sisyphus/srpms/gcc10/patches

havardAasen, 2021-02-08:
This has been fixed in master with this commit 4c7b154720ad3720dcc1a435330a1d586a6a7fcb  PR #827 with some more info. I think this issue was mentioned when 2.8.0 was being prepared for release. Don't remember where though.

rene-dev, 2021-05-26:
I can confirm that linuxcnc builds with gcc 10.

closed 2021-05-26 by rene-dev