LinuxCNC/linuxcnc #1071: WJ200_vfd not compatible with parity argument for latest libmodbus

damiodj, 2021-02-07:
While trying using the user component wj200_vfd and setting the serial port parity, the component provide to a "parity" argument.
The code filter for Y or N:

line 434 of wj200_vfd.c:
if(parity != 'Y' && parity != 'N')

while the new libmodbus expect N, E, O (No, Even, Odd - https://libmodbus.org/docs/v3.0.8/modbus_new_rtu.html)

changing line 434 to:
if(parity != 'N' && parity != 'E' && parity!='O')

fixes the issue

also the message in line 437 need to be changed to reflect this.

andypugh, 2021-02-08:
Which version of LibModbus? 
With reference to http://wiki.linuxcnc.org/cgi-bin/wiki.pl?MinimumSoftwareVersions
Which LinuxCNC versions are affected?

damiodj, 2021-02-08:
Tested on a clean Debian 10, both LinuxCNC 2.8.1 and 2.9-pre are affected.
Both libmodbus current versions 3.0.8 and 3.1.6 require N, O or E.
Looking at libmodbus repository I cannot find any old version that accepts an 'Y' as parity, the only mention of parity in changelogs are from 2010.

I'm thinking that the bug was always there, maybe most of the installations use parity == 'N', keep in mind that the error returned setting parity == 'Y' is generic and was not clear for me what was wrong before looking at libmodbus code

andypugh, 2021-02-08:
OK, that seems pretty clear. 
Do you want to create a pull-request? (If not, then we can hope that someone else does)

damiodj, 2021-02-08:
I will.
thanks

andypugh, 2021-02-08:
If it is just one change to just one file then you can make the PR directly on this web site. Each file has an edit icon in the full code view that can then allow a PR to be made. 
(For anything more complicated, especially toucching multiple files, then I think you need to make a local Git repository and work in the normal way)

oMtQB4, 2021-06-04:
This issue can be closed since the pull request already merged.

closed 2021-06-04 by damiodj