LinuxCNC/linuxcnc #1072: Unable to enable axis when setting `CYCLE_TIME` in INI: [DISPLAY]

hansu, 2021-02-07:
When setting the value `CYCLE_TIME` in the DISPLAY section of the INI file, axis cannot be enabled any longer by the gui buttons
```
[DISPLAY]
DISPLAY = axis
...
CYCLE_TIME = 100
```
In gmoccapy and qtaxis this value makes no problem.

Tested with LinuxCNC 2.8.1 on Debian buster.

c-morley, 2021-02-07:
CYCLE_TIME in AXIS is in seconds, in other GUIs it is (always?) milliseconds

andypugh, 2021-02-07:
Which seems slightly crazy. 
And means that you can't just change the DISPLAY= to change between GUIs.

c-morley, 2021-02-08:
Yes It's my fault I suppose. when making gscreen ms made much more sense then seconds.
Qtvcp will accept either - we could probably modify the other screens to as well.

hansu, 2021-02-09:
Ah sorry my fault. I only looked up if this value is still documented in 2.8 but I didn't check the unit because in my INI it is commented as milliseconds.
But the doc is misleading at this point. I think axis is the only GUI that uses I think. Is that right?
For the other GUIs that is not documented, only for inside gladevcp's, but that's a different value I guess:
```
cycle_time

    The time the DRO waits between two polls, the value must be an integer in the range of 100 to 1000,

c-morley, 2021-02-24:
Ok I changed AXIS in master to accept seconds or ms.
Feel free to open this again if there is still a problem.

closed 2021-02-24 by c-morley

hansu, 2021-02-26:
Nice, thanks!
So `CYCLE_TIME = 1.0` would be 1 ms and `CYCLE_TIME = 0.9` would be 0.9 seconds.
So it is important that we don't forget to document that!

c-morley, 2021-02-27:
Ok thanks for the reminder. I updated the docs I found in master

hansu, 2021-02-27:
:+1: 
Something other: To which version does the docs page http://linuxcnc.org/docs/html/ point? Shouldn't that be the current one, thus the 2.8?

andypugh, 2021-02-27:
I think I fixed the /docs/html link

hansu, 2021-02-27:
> I think I fixed the /docs/html link

Yes, you did. Thanks!

hansu, 2021-04-11:
Just for reference:
https://github.com/LinuxCNC/linuxcnc/commit/bb1a1c4d70e1faa84819a7e6a0f6401165613bf8 **qtvcp -fix conversion of seconds/ms in CYCLE_TIME**
https://github.com/LinuxCNC/linuxcnc/commit/b64185b44f7579c6c35f0abb554b63b733484bc4 **docs -add info about INI's [DISPLAY]CYCLE_TIME**

hansu, 2021-04-11:
@c-morley should not https://github.com/LinuxCNC/linuxcnc/commit/b64185b44f7579c6c35f0abb554b63b733484bc4  also in 2.8?

c-morley, 2021-04-11:
The CYCLE_TIME changes were not added to 2.8.
I suppose we could document that AXIS and everyone screen use different units

hansu, 2021-04-11:
> I suppose we could document that AXIS and everyone screen use different units

Yes that's exactly what I meant and what you have already done in this commit.
Maybe except ` docs/src/gui/qtvcp.txt `