LinuxCNC/linuxcnc #1075: Please include this GUI.

Supermagnum, 2021-02-10:
https://youtu.be/lRZ_AMuayoM

saxa, 2021-02-10:
Oh yeah, really nice one. Marketing is a good thing for software :)

andypugh, 2021-02-10:
The problem is that this isn't a working user interface as presented. 
There is something vaguely similar here though. 
http://linuxcnc.org/docs/2.8/html/gui/qtvcp.html
What is it about this interface that makes you desire it so much more than the other QtVCP screens, or the QTPtVCP screens here? https://www.qtpyvcp.com/showcase/mill_vcps.html

saxa, 2021-02-11:
I dont know, but the colors and design on the requested GUI is much more pleasant to the eye than Qtvcp.

gmoccapy, 2021-02-11:
You can use also QtpyVCP Guis, witch are also very pleasant in their look,Or yiu use gmoccapy, what has been designed special to compare to industrial applications. You can use themes to modify the look of most GUI incl. Gmoccapy and QtVCP screens and also most GUI can be modified by adding additional widgets or hiding other stuff.But you are very wellcome to propose changes or additions. In this case, it would need someone to take over to make out of a proposal an code to be added.I would vote to include the complete QtPyVCP project to LinuxCNC, and IMHO the people who are coding that one are willing to put some effort on it. (Most difference between that one and QtVCP ist the signal slot handling, the documantation and some coding stuff. Chris will be able to give more information.)-------- Ursprüngliche Nachricht --------Von: saxa <notifications@github.com>Datum: Do., 11. Feb. 2021, 03:02An: LinuxCNC/linuxcnc <linuxcnc@noreply.github.com>Cc: Subscribed <subscribed@noreply.github.com>Betreff: Re: [LinuxCNC/linuxcnc] Please include this GUI. (#1075)
I dont know, but the colors and design on the requested GUI is much more pleasant to the eye than Qtvcp.

c-morley, 2021-02-11:
This isn't a working gui as Andy said. if someone would like to continue the development of qtvcp's similar concept example then that would be great. untill we have a volunteer for that it's not going to happen unfortunately. We are short on developers.
The example you pointed to was a good concept but was not built with widgets - it's artistic.
Qtvcp can be themed and customized for radically different looks.
PyQtvcp is a different project and it's best to discuss it in it's own forum.

I'm going close this.

closed 2021-02-11 by c-morley