LinuxCNC/linuxcnc #1078: LinuxCNC 2.9 Stepconf Wizard Parallel Port Setup Issue

BSHoekstra, 2021-02-11:
I have installed LinuxCNC 2.9.  When running Stepconf Wizard I noticed that only Parport 0 can be selected.

## This is what I expected to happen:

I expected to be able to select Parport 0, 1 or be able to enter the parport address similar to all previous Linux CNC versions.

## It worked properly before this:
Parport selection in Stepconf Wizard worked properly in all previous LinuxCNC releases.  Linux 2.9 is in the development phase.  

## Information about my hardware and software:
 * I am using this Linux distribution and version:  Debian 10 Buster
 * I am using this kernel version (shown by `uname -a`):  4.19.0-14-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux

 * I am running:  A binary version from linuxcnc.org (including buildbot.linuxcnc.org)

 * I am using this LinuxCNC version: 1:2.9.0-pre0-3505-g7e326429e
  
 * I am using this user interface (GUI):  AXIS

 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 
Host bridge: Intel Corporation 4th Gen Core Processor DRAM Controller (rev 06)
Subsystem: Dell 4th Gen Core Processor DRAM Controller
SMBus: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller (rev 04)
Subsystem: Dell 8 Series/C220 Series Chipset Family SMBus Controller
Parallel controller: Oxford Semiconductor Ltd Device c110 (prog-if 02 [ECP])
Subsystem: Oxford Semiconductor Ltd Device c110

c-morley, 2021-02-11:
This is expected behavior if you have only one parport available on the system.
Stepconf now checks the system to see what is available and then only allows those settings.
I'm not sure why only those are allowed but probably to make it simpler for a new user to get things right.

Is there a parport on your system that is not available because of these changes?

BSHoekstra, 2021-02-11:
Thank you.  I just wasn't expecting this change from earlier LinuxCNC releases.  The setup worked as designed then and you are right, it will be a very nice upgrade for new users.  I've spent a lot of time over the years figuring this out each time computers were changed out in my shop.

Thank you again.

Have a nice day,

Barry

closed 2021-02-11 by BSHoekstra