LinuxCNC/linuxcnc #1079: LinuxCNC 2.9 Program Organization Issue Under Applications Tab

BSHoekstra, 2021-02-11:
I have installed LinuxCNC 2.9.  Many of the standard LinuxCNC programs were auto-loaded under the category Applications/ Accessories instead of Applications/ CNC.

## This is what I expected to happen:
LinuxCNC files would be auto-loaded under Applications/CNC 

## It worked properly before this:
LinuxCNC files auto-loaded under Applications/CNC in previous LinuxCNC Releases.    

## Information about my hardware and software:
 * I am using this Linux distribution and version:  Debian 10 Buster
 * I am using this kernel version:  4.19.0-14-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux

 * I am running:  A binary version from linuxcnc.org (including buildbot.linuxcnc.org)

 * I am using this LinuxCNC version: 1:2.9.0-pre0-3505-g7e326429e
  
 * I am using this user interface (GUI):  AXIS

 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): 
Host bridge: Intel Corporation 4th Gen Core Processor DRAM Controller (rev 06)
Subsystem: Dell 4th Gen Core Processor DRAM Controller
SMBus: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller (rev 04)
Subsystem: Dell 8 Series/C220 Series Chipset Family SMBus Controller
Parallel controller: Oxford Semiconductor Ltd Device c110 (prog-if 02 [ECP])
Subsystem: Oxford Semiconductor Ltd Device c110

[image]

andypugh, 2021-02-11:
I think I see the problem.
https://github.com/LinuxCNC/linuxcnc/blob/master/debian/extras/etc/xdg/menus/applications-merged/CNC.menu
So anything of type X-CNC, X-CNC-REF or X-CNC-DOC should be in the CNC tree. 
But the desktop files have _ instead of - in places. And in some cases the wrong category. 
https://github.com/LinuxCNC/linuxcnc/blob/master/debian/extras/usr/share/applications/linuxcnc.desktop
But I can't find the commit that inserted that error: 
https://github.com/LinuxCNC/linuxcnc/commits/master/debian/extras/usr/share/applications/linuxcnc.desktop

andypugh, 2021-02-11:
I am curious id https://github.com/LinuxCNC/linuxcnc/commit/8b5697ce4fceb6da859f613ec43f4ae55a17c76b helps

BSHoekstra, 2021-02-11:
In reading through the reply I did some experimenting with the ideas and figured out how to manually brute force fix this issue.  I'm hoping that this info. helps develop a more elegant way to approach this so the files don't auto-install this way.....Please note that I am by no means a programmer.

1) Navigated to /usr/share/applications/   
2) Used Kate to open each "desktop file" showing up under Applications/Accessories and showing up under Applications/CNC/Documentation (e.g. G-Code Quick Reference - EN)
3)  Some files were coded as follows, so I made this change
    (-) Categories=Utility;Engineering;X-CNC-DOC;
    (+) Categories=X-CNC-DOC;
4) Some files were coded as follows, so I made this change
    (-) Categories=Science;X-CNC-DOC;
    (+) Categories=X-CNC-DOC;

Now all files show up in only one place (i.e. under Applications/CNC/Documentation) where they should reside versus 2 places (i.e. under Applications/Accessories and under Applications/CNC/Documentation).

Thank you.

Barry

andypugh, 2021-02-11:
I think that applications have to be allocated to a recognised category for the package to be compliant with the Debian standards. (Or, at least to stop Lintian complaining)

BSHoekstra, 2021-02-11:
Thank you.  : )

Have a nice day,

Barry

closed 2021-02-11 by BSHoekstra