LinuxCNC/linuxcnc #1080: LinuxCNC 2.9 File Selection Issue

BSHoekstra, 2021-02-11:
I have LinuxCNC 2.9 installed.  After launching AXIS.......Click on File, Click Open, browse to the desired .ngc file and then double click on the selected .ngc file....nothing happens.  You can only single click to select the .ngc file and then you have to single click Open to Open the file in AXIS. 

This is what I expected to happen:

Double clicking the selected .ngc file would automatically select and Open the file in AXIS without having to single click the file to select it and then single click on Open to Open it in AXIS. 

Information about my hardware and software:

    I am using this Linux distribution and version: Debian 10 Buster

    I am using this kernel version: 4.19.0-14-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux

    I am running: A binary version from linuxcnc.org (including buildbot.linuxcnc.org)

    I am using this LinuxCNC version: 1:2.9.0-pre0-3505-g7e326429e

    I am using this user interface (GUI): AXIS

    I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):
    Host bridge: Intel Corporation 4th Gen Core Processor DRAM Controller (rev 06)
    Subsystem: Dell 4th Gen Core Processor DRAM Controller
    SMBus: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller (rev 04)
    Subsystem: Dell 8 Series/C220 Series Chipset Family SMBus Controller
    Parallel controller: Oxford Semiconductor Ltd Device c110 (prog-if 02 [ECP])
    Subsystem: Oxford Semiconductor Ltd Device c110

phillc54, 2021-02-11:
I have seen that behaviour before if the CYCLE_TIME in the [DISPLAY] section of the ini file is not quite long enough.

BSHoekstra, 2021-02-11:
Wow!  You solved that problem! Thank you very much!!  :)  I changed the CYCLE TIME from 0.01 to 0.02 as shown below.  Sorry for posting this issue on this forum but I thought it was a program issue because I had not seen this until I upgraded to LinuxCNC 2.9.  No further action is required.  Please close this ticket.  

# Cycle time, in seconds, that display will sleep between polls
# CYCLE_TIME =         0.0100
CYCLE_TIME =            0.0200

Thank you again.

Barry

closed 2021-02-11 by BSHoekstra