LinuxCNC/linuxcnc #1088: Calling the python linuxcnc.command.mdi("o<myfunction> call") causes milltask to segfault after the ~1000 time

ZincBoy, 2021-02-18:
## Here are the steps I follow to reproduce the issue:

 1. Open a mill simulation using Gmoccapy (other interfaces may show this behavior but I have not tested them)
 2. Find the <process number for milltask>
 3. lsof | grep <process number of milltask> | wc -l
 4. Record the number of open files for milltask
 5. Run a macro using one of the MDI screen macro buttons.  These use the python linuxcnc.command.mdi interface to call the o word function.
 6. lsof | grep <process number of milltask> | wc -l
 7. See that the number of open files for milltask has increased by one
 8. Repeat steps 6-7 and see that the files open increase by one for every button press.
 9. If the button is pressed about 1024 times (or whatever the system open file limit is), milltask will segfault with a null pointer read
 10. I didn't press the button 1024 times, but I have a python script that repeatedly calls an o word function and it crashes after about 1024 iterations.
 11. The file open problem does not happen if the o word function is called through the EMC_MDIHistory widget or by repeatedly running a program through the cycle start.

## This is what I expected to happen:

I expected the o-word.ngc files to be closed after access like what happens when using the EMC_MDI_History widget.  

## This is what happened instead:

Each o-word.ngc call opened a new instance of the file and kept it open.  Eventually hitting the system open file limit and causing milltask to segfault

## It worked properly before this:

Tested on 2.8.1 in simulation and on hardware.  Not tested on other versions.

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian Buster
 * I am using this kernel version (shown by `uname -a`): Linux VK45 4.19.0-14-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux
 * I am running ...
   * [ X] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 2.8.1
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): gmoccapy
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):  Mesa and Simulation

Segfault info:
Feb 17 23:10:31 VK45 kernel: [132086.124829] milltask[10042]: segfault at 0 ip 00007fd2b18886a3 sp 00007fff116632a8 error 4 in libc-2.28.so[7fd2b1833000+148000]
Feb 17 23:10:31 VK45 kernel: [132086.124836] Code: 1f 84 00 00 00 00 00 c3 0f 1f 80 00 00 00 00 8b 42 04 83 e1 cf 8d 70 01 89 72 04 89 0f eb 99 0f 1f 84 00 00 00 00 00 48 89 fa <8b> 07 f6 47 74 80 0f 84 b9 00 00 00 89 c1 81 e1 00 80 00 00 0f 85

gmoccapy, 2021-02-19:
I checke dthat and I am able to reproduce the bahavior.

But I needed to use: 
lsof | grep milltask | wc -l

After each call of a subroutine the number of files increases by one as mentioned. I checked also with axis, there it does not!

With gscreen the behavior is the same as with gmoccapy, so the number increases. 

IMHO tha means it is related the the GladeVCP part of linuxcnc

ZincBoy, 2021-02-19:
I found that this only happens if there is no .ngc file opened in the auto mode page.  If you set a file to load on start this issue will not occur.  A fix for now is to always load a .ngc file before trying to call any o word macros.

This is why I thought it didn't happen when calling an o-word through the EMC_MDIHistory widget.  It actually does but when I was testing, I had run a .ngc file in auto mode first.  On re-start of the system I found that EMC_MDIHistory widget was causing the open files to remain.  Once an .ngc file is opened no further files are kept open.   Opening the .ngc does not close any of the files left open.

gmoccapy, 2021-02-19:
The very strange thing is:
- I do in MDI History         o<increment> call  [1] [1]
- The machine moves as expected 1 mm in each axis
- lsof | grep increment | wc -l              gives 1
- after doing the same macro again with o<increment> call  [1] [1]     gives 2
- so the number of files is increasing.
I checked hal_mdihsitory and was not able do find something strange neither in hal_glib nor hal_actions seems to do strange stuff. I do not know where to look further, may be someone with C knowledge can check the python interface

Norbert

ZincBoy, 2021-02-20:
I reproduced this issue in Axis but only if the OPEN_FILE="" is set in the INI.  @dngarrett's fix works for me.  I tested it in 2.9pre and the issue was resolved.  That was a fast fix!

commit 65e2d92b (2021-02-25): interp_o_word.cc: mdi-opened files leak #1088

Observed:
with previous_frame->positon==-1, subroutine files
opened by mdi command not closed, so conditionally
close settings->file_pointer if non-null

fixes #1088
closed 2021-02-25 by dngarrett