LinuxCNC/linuxcnc #1089: Halcompile: personality does not appear to work in userspace comps

andypugh, 2021-02-21:
## Here are the steps I follow to reproduce the issue:

Attempt to compile the following component:

```
component test;
option userspace;
license "GPL";
pin out unsigned apin-##[16:personality];

;;

void user_mainloop(void) {

    while (1) {
        FOR_ALL_INSTS(){
        }
    }
}
```

## This is what I expected to happen:

Compilation and a working do-nothing component. 

## This is what happened instead:

Compiler error:

```
/tmp/tmpJASh03/test.c:63:32: error: expected ‘)’ before numeric constant
 RTAPI_MP_ARRAY_INT(personality, 64, "personality of each test");
                                ^~~
                                )
```

## It worked properly before this:

Maybe never. It doesn't work with 2.7 either. 

## Information about my hardware and software:

 * I am using this Linux distribution and version: Buster / Wheezy 
 * I am using this kernel version (shown by `uname -a`): RTAI in both cases
 * I am running ...
   * [ ] A binary I built myself

andypugh, 2021-02-21:
@jepler Can you remember if this ever worked?

jepler, 2021-02-21:
I don't remember.