LinuxCNC/linuxcnc #1090: Wrong error message about second parallel port

hansu, 2021-02-21:
When using a second parallel port on axis or qtvcp
configured with
`loadrt hal_parport cfg="0xcc00 out 0xa880 in"`
an error message is shown like
[image]

But  the port is quite working so the error message is a little bit misplaced. 
In gmoccapy the message doesn't appear.

Moreover it is a little bit irritating that the address is shown as decimal.

LinuxCNC 2.8.1
Debian 10

andypugh, 2021-02-21:
It's in decimal because the same message will report the index for smaller numbers. 
(But, then, I doubt anyone uses more than 9 parports by index, so it easilty could be hex)

https://github.com/LinuxCNC/linuxcnc/blob/master/src/rtapi/uspace_rtapi_parport.cc#L74

I recall a similar issue with the Mesa driver (for 7i43 and 7i90) and it was due to something automatically choosing default addresses.

hansu, 2021-02-26:
Ah okay. But then it is still strange that this message occur and the parallel port still works.