LinuxCNC/linuxcnc #1093: HALUI man pages HTML docs use wrong hyphen

c-morley, 2021-02-27:
At least 2.7, 2.8, and master HTML man pages for HALUI render the wrong hyphen.
This shows up as an error when running copy/pasted code.

This is a typical man page line:
.B halui.feed\-override.count\-enable\fR bit in \fR (default: \fBTRUE\fR)

The hyphen here is right. When rendered it's a longer one.
It seems to be the backslash-ed hyphen 
mitsub_vfd renders correctly but doesn't use backslash.

Does anyone know why the hyphen was backslash-ed?

andypugh, 2021-02-27:
https://www.gnu.org/software/groff/manual/groff.html says that \- is the minus character. (look for \[-] on that page)
I think that it is used to specify that the character is a minus rather than a hyphen. 

Your description doesn't make it clear to me which are right and which are wrong.

c-morley, 2021-02-27:
Seems we use minus as a hyphen then.

wrong:
http://linuxcnc.org/docs/devel/html/man/man1/halui.1.html

right:
http://linuxcnc.org/docs/devel/html/gui/halui.html#_max_velocity

andypugh, 2021-02-27:
Well, it's next to + on the keyboard. And gives you a minus in a text editor or when coding. 

The complication is that word processors use the key to insert hyphens, which are a different character. 

The HAL pin names definitely have minus as the dash.

Sigma1912, 2021-03-02:
@andy
Do the two pages posted above by c-morley give you the same hyphens on your machine?

andypugh, 2021-03-02:
Following the links, copying and pasting shows the first one as unicode hyphens (E2 88 92)
The second is a (correct) 2D. 
But, curiously, if I look on my own machine at the HTML generated when I build the docs locally, the hyphens are all correct there too. 
`<b>halui.tool.length-offset.b</b>` (2D)

(Looking at the raw html served from www.linuxcnc.org) 
`<b>halui.tool.length−offset.b</b>` (E2 88 92)

So the same tools seem to generate different output from the same input files on different machines.

hansu, 2021-03-05:
> But, curiously, if I look on my own machine at the HTML generated when I build the docs locally, the hyphens are all correct there too.

Yes, on my machine it is generated correctly too.

hansu, 2021-03-05:
Were there made some changes in the build process from 2.6 to 2.7?
Because the docs from 2.6 seems to be correct regarding this.

(http://linuxcnc.org/docs/2.6/html/man/man1/halui.1.html <-->http://linuxcnc.org/docs/2.7/html/man/man1/halui.1.html)

andypugh, 2021-03-05:
Looking around, it seems that manpages that escape the hyphen "\-" (Which is actually meant to be correct) end up with html containing "&minus;" which then gets rendered as a dash. 
Interestingly the shuttle manpage does _not_ escape the hyphen, and that one comes out right. 
So, possibly, the trick for manpages is to unescape (imprison?) the minus signs everywhere.

c-morley, 2021-03-05:
Yes I mentioned that in first post.
The question was is there a good reason that it was back slashed - seems probably not.

c-morley, 2021-03-05:
I guess I should have said backslashed minus sign in the first post though :)

andypugh, 2021-03-06:
I tried removing the escapes in the halui manpage in Master, and the docs were generated correctly:
http://linuxcnc.org/docs/devel/html/man/man1/halui.1.html
Compare to the unchanged 2.8 docs:
http://linuxcnc.org/docs/2.8/html/man/man1/halui.1.html

So, that seems to be a fix for manpages written in troff, but not necessarily for manpages written in asciidoc

Though it would be neater to have a2x (or whatever does it) not convert /- to &minus

andypugh, 2021-03-06:
Curiously the problem appears to have been introduced by an attempt to fix this very problem:
https://github.com/LinuxCNC/linuxcnc/commit/4a897203bcdfc1fb57438e98ea56a7417c9c88ad#diff-cfc9ad27e8d4ec539bd151f07d436c5383003b61ed81c1c6ffbd59bb237b46d6

hansu, 2021-03-10:
Did I miss a commit regarding this issue? Or has someone modified the build process? Because now it is displayed correctly. Even the [images](http://linuxcnc.org/docs/html/man/man9/toggle2nist.9.html) are displayed now :heart_eyes:

SebKuzminsky, 2021-03-10:
I moved the building of the docs from Wheezy to Buster.  The newer version of the docs toolchain in Buster gets this particular thing correct.

For posterity, manpages should use:
* `\-` where they want the normal "minus" character (unicode 2d, what we used to call ASCII 0x2d). This includes HAL pin names and command-line arguments like `--verbose`.
* `-` where they want a hyphen, for example as in the phrase "command-line".

andypugh, 2021-03-10:
https://sourceforge.net/p/emc/mailman/emc-users/thread/a11d3b7a-1736-bb6c-77cc-3112246f8d86%40gmail.com/#msg37235008

hansu, 2021-03-10:
Nice done!

closed 2021-05-13 by andypugh