LinuxCNC/linuxcnc #1100: Gmoccapy optimizations

hansu, 2021-03-05:
Here some annotations about gmoccapy:

**1. Icons**

The buttons of the new search and replace function are some kind of experimental I think:
[image]
To make it fit to the minimal screen resolution here a proposal:

[image]

Furthermore I optimized some icons (red) to better distinguish between then.
If you agree to this I can do a PR from: https://github.com/hansu/linuxcnc/tree/gmoccapy-icons

**2. Functional**
- A little bug in auto mode: When choosing "load new program" and then return without selecting a file, it results in a fullscreen preview screen [(video)](https://user-images.githubusercontent.com/67957916/110139371-3d015100-7dd3-11eb-986a-4f62b6f7c987.mp4).
- When editing a file without explicitly saving the changes, the changes are lost on pressing the return button. A warning dialog like "Save changes? [Yes] [No] [Cancel]" would make sense I think.
- Allow keyboard jogging in the [touch view](https://user-images.githubusercontent.com/67957916/110141271-468bb880-7dd5-11eb-93ba-60852c543947.png). It's a little bit annoying to always leave that mode to jog to the next edge...
Of course it should be disabled again when be in offset editing mode.
In my opinion keybord jogging could be also enabled in auto mode like it is done in axis. But I can image that I am maybe alone with this opinion :smile: 

**3. Nice to have**
- Add new `EMBED_TAB_LOCATION`
I think it is useful to be able to use [this space](https://user-images.githubusercontent.com/67957916/110143344-70de7580-7dd7-11eb-9767-08207db1476b.png)  also for customized panels as `EMBED_TAB_LOCATION`
- Maybe a counter that counts the run time of the program can be added to the lower right corner where the current time and date is displayed. (But can also be done with the point above ;-))

hansu, 2021-03-07:
Concerning 2.2:

It could also be made clear by the buttons that it is not automatically saved.
Maybe in this way, but that would violate the concept of operation:

[image]

Or a little bit modified:
[image]

What do you think @gmoccapy  ?

gmoccapy, 2021-03-07:
Hallo Hans, 

**Icon Search and replace**
thanks for reporting. I corrected the Search button line by shortening the translations, so it now fits again also on minimal screen resolution.

**Icon different icon**
The ICON you showed in picture 2 are both pictures witch are taken from gtk-source, so they depend on what window manager you use. In my new GUI i can make tehem interchangable by the user. I will not modify the code of gmoccapy in this manner at this state.

**Functional bug Back from FileChooser**
Thanks for reporting, I fixed that now and have pushed the changes to 2.8 branch

**Functional : Ask user to save changes**
Also this has been added today, also pushed to 2.8. I needed to add a new signal to sourceview. If you only click once in the text, even without changing anything will react to a dialog if you want to save changes. It is not very easy to compare the text if a change has been made. I will check if I can do that more perfect.

**Functional : Jogging in Touch off mode,** you are right, I have not looked further at this, do you now, that you can click on the DRO Axis letter to set an axis value? This way you do not need to go to touch off button. A click on the numbers will toggle the DRO and a click on the letter will open the dialog to enter a value. (May be I have not documented that, shame on me)

**Functional : Jog Speed** I habe never noted that, how can you jog faster than max vel? I need to check on this asap.

**Nice to have: additional tab places**
you can use something like this to use that space:
`EMBED_TAB_NAME = tool_replacement_panel
EMBED_TAB_LOCATION = box_tool_and_code_info
EMBED_TAB_COMMAND = gladevcp -x {XID} -u signals.py -H signals.hal signals.glade`
See gnmoccapy plasma for this, I should document that one.

Do you know, that you can use also ntb_setup as place to put panels in?

**Nice to have counter**
do you mean something like this?
[https://forum.linuxcnc.org/gmoccapy/40709-gmocappy-clock?start=10#201002](url)

Norbert with thanks for reporting

hansu, 2021-03-08:
Hallo Norbert,
thanks for your fast reply and fixing!
Wenn ich mich irgendwie revanchieren kann - sag bescheid ;-)

**Icon Search and replace**
That fits now but looks a little bit squeezed. You don't want icons in this area?

**Icon different icon**
For the save-icons I only added the icons that already exists in the images folder. If you prefer using the icons from the theme, than these icons (like [this](https://github.com/LinuxCNC/linuxcnc/blob/master/share/gmoccapy/images/save.png)) can be removed from the repo. 
But didn't you like my modified configure icon? In my opinion it was too colourful before so it could be confused with the auto mode icon. 

**Functional bug Back from FileChooser**
> Thanks for reporting, I fixed that now and have pushed the changes to 2.8 branch

:+1:

**Functional : Ask user to save changes**
Yes it's a little bit irritating that it warns that it isn't saved even though it is. But even better than no warning.
And the buttons in this messagebox could be increased for a etter touch screen experience.
In this context - do you noticed  https://github.com/LinuxCNC/linuxcnc/pull/1029 ?

**Functional : Jogging in Touch off mode**
Oh nice I only knew clicking on the numbers not on the axis letters. No chance for the user to use this intuitively.
But a nice possibilty! Anyway may you could still have a look at this.

**Functional : Jog Speed**
>I habe never noted that, how can you jog faster than max vel? I need to check on this asap.

Yes, but only if you set halui.max-velocity to a value lower than your machine's max velocity. 

**Nice to have: additional tab places**
So the tabs names as for plasma are working for the regular version too? Cool. Good that we talked about this. Thanks for that hint.

>Do you know, that you can use also ntb_setup as place to put panels in?

No. But I think I would like to have it on the main screen.

**Nice to have counter**
Funny that somebody else had the same wish at the same time :-)

**Something else**
Could you enable the horizontal scoll bar when the program is running respectively when it is paused? I recently stepped through a program and couldn't see the whole line...

hansu, 2022-06-16:
> **3. Nice to have**
> 
>  * Add new `EMBED_TAB_LOCATION`
>       I think it is useful to be able to use [this space](https://user-images.githubusercontent.com/67957916/110143344-70de7580-7dd7-11eb-9767-08207db1476b.png)  also for customized panels as `EMBED_TAB_LOCATION`

https://github.com/LinuxCNC/linuxcnc/commit/8d9d9cbdc5e9050672eff81cc39093bd894d195f resolves the last open point, so this can be closed.

closed 2022-06-16 by hansu