LinuxCNC/linuxcnc #1106: compile error when doing a dpkg-buildpackage on master version.

grotius-cnc, 2021-03-13:
I think a tiny bug. But it stop's the compile process.

Making halcompile manpage abs.9
Making halcompile manpage thermistor.1
make[1]: *** No rule to make target '../docs/man/man1/halstreamer.1', needed by 'depends/man1/halstreamer.1.d'.  Stop.
make[1]: *** Waiting for unfinished jobs....
Depending linuxcnc.1
make[1]: Leaving directory '/opt/linuxcnc/src'
make: *** [debian/rules:36: build-stamp] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2

andypugh, 2021-03-14:
The buildbot seems to be coping:
http://buildbot.linuxcnc.org/buildbot/builders/4041.deb-buster-rtpreempt-amd64/builds/817

halstreamer is the first in the list of userspace comps with a manpage authored in asciidoc. 

What OS are you creating debs under?

andypugh, 2021-03-14:
I just tried a build here. Everything seems fine? 

dpkg-deb: building package 'linuxcnc-doc-fr' in '../linuxcnc-doc-fr_2.9.0~pre0_all.deb'.
dpkg-deb: building package 'linuxcnc-dev' in '../linuxcnc-dev_2.9.0~pre0_amd64.deb'.
dpkg-deb: building package 'linuxcnc' in '../linuxcnc_2.9.0~pre0_amd64.deb'.
dpkg-deb: building package 'linuxcnc-doc-es' in '../linuxcnc-doc-es_2.9.0~pre0_all.deb'.
dpkg-deb: building package 'linuxcnc-doc-cn' in '../linuxcnc-doc-cn_2.9.0~pre0_all.deb'.
dpkg-deb: building package 'linuxcnc-dbgsym' in '../linuxcnc-dbgsym_2.9.0~pre0_amd64.deb'.
dpkg-deb: building package 'linuxcnc-doc-en' in '../linuxcnc-doc-en_2.9.0~pre0_all.deb'.
 dpkg-genbuildinfo
 dpkg-genchanges  >../linuxcnc_2.9.0~pre0_amd64.changes
dpkg-genchanges: info: including full source code in upload
 dpkg-source --after-build .
dpkg-buildpackage: info: full upload; Debian-native package (full source is included)

grotius-cnc, 2021-03-14:
Hi Andy,

Debian 11 Bullseye. I get lcnc compiled from the src directory without problems.

I will test some without compiling the userspace comps. Maybe i find a solution. Thanks for the hint.

rene-dev, 2021-08-19:
@grotius-cnc has this been resolved? I have successfully compiled on Bullseye.