LinuxCNC/linuxcnc #1108: iocontrol - io - iov2 man pages

hansu, 2021-03-14:
I assume that the man page for iocontrol is still included because of compatibility reasons?
If yes I would suggest only add a note like "See IO man page" instead of duplicating the content. So it looks as if there are two components.
And what does IOv2 differently? It would be nice if that could be documented.

andypugh, 2021-03-14:
There might be some clues here: http://wiki.linuxcnc.org/cgi-bin/wiki.pl?ToolchangerProtocolProposal
But Michael didn't get round to documenting iov2 before splitting off to start Machinekit.

hansu, 2021-03-17:
Okay thanks I can add some of this information to the iov2 man page.
What exactly is Machinekit? On first sight it looks like a LinuxCNC clone...

andypugh, 2021-03-17:
Machinekit is a Fork, so diverged from LinuxCNC when a group of developers did not like the focus on maintaining a stable and functional code base in LinuxCNC. They had some good ideas and I followed the project for a while, until it was made clear that my input was not wanted. I don't know how far they got with those ideas.

PR #1127 (2021-04-18): docs: reworked iocontrol/io/iov2 man pages
closes #1108