LinuxCNC/linuxcnc #1111: Gmoccapy: New translations causes wrong default theme

hansu, 2021-03-17:
The new translation files in b8ce6c823011cd8bb91dea140cf546e26de4be00 causes gmoccapy to use a wrong theme if using "Follow System Theme".
In detail this entry: https://github.com/LinuxCNC/linuxcnc/blob/b8ce6c823011cd8bb91dea140cf546e26de4be00/src/po/gmoccapy/de.po#L288

Maybe at this position the translated string is compared with the untranslated string?
https://github.com/LinuxCNC/linuxcnc/blob/b8ce6c823011cd8bb91dea140cf546e26de4be00/src/emc/usr_intf/gmoccapy/gmoccapy.py#L4168

hansu, 2021-03-17:
And when you are editing the translation files, you can add a _g_ here :wink: 
 https://github.com/LinuxCNC/linuxcnc/blob/b8ce6c823011cd8bb91dea140cf546e26de4be00/src/po/gmoccapy/de.po#L1590

hansu, 2021-03-17:
Issue in second comment is fixed in https://github.com/LinuxCNC/linuxcnc/pull/1112
Looks now so:

[image]

hansu, 2021-03-17:
@gmoccapy

hansu, 2021-03-17:
I just detected another thing: The pause button was not updated and the spindle buttons not deactivated/activated when using an external button for pause/resume. Fixed in https://github.com/LinuxCNC/linuxcnc/pull/1112/commits/3779d68617db1c360290f8847b2fed6c5e30fd5c by using the `program-pause-changed`signal instead of the button signal from the GUI. But may you have a better solution as I handled all items separately.

hansu, 2021-03-20:
The thing with the not updated pause button I could only reproduce on my developing system. At the machine this is not showing up. Further I recognized that in my version the refresh of the Pause button is a little bit delayed compared to your solution with the signal from the GUI.

gmoccapy, 2021-05-14:
@hansu 
I checked here all possibilities related to the PAUSE button. I am not able to reproduce your comments.
If I start a sim config and set halui.program.pause the running program wait for the resume of the external button. All widgets are updated as expected, so spindle and cooling are accessible. Not scrolling in the code, that is an other item!

Norbert

gmoccapy, 2021-05-14:
The wrong System Theme stuff has been solved in https://github.com/LinuxCNC/linuxcnc/commit/69fe5b61e2122edba8a2884aa75a9f7b1d7ce7a4

Will Update the translation files in a different commit

gmoccapy, 2021-05-14:
ERROR string : deleted unnecessary ERROR in translation strings
https://github.com/LinuxCNC/linuxcnc/commit/83e6e39ea7b49280a9242ba4063f04ee08f36ba5

hansu, 2021-05-14:
> ERROR string : deleted unnecessary ERROR in translation strings
> [83e6e39](https://github.com/LinuxCNC/linuxcnc/commit/83e6e39ea7b49280a9242ba4063f04ee08f36ba5)

Not completely fixed I think.
1. setting 
`setp joint.0.amp-fault-in 1`
`setp spindle.0.amp-fault-in 1`
and turn on machine results in:
[image]
So the amp-fault string isn't translated correctly.

2. setting **additionally** 
`setp joint.0.pos-lim-sw-in 1`
an turn on machine results in:
[image]
So
   1. one "Fehler" string is still to much
   2. the amp-fault error message doesn't appear any longer even though it is still active

hansu, 2021-05-14:
> @hansu
> I checked here all possibilities related to the PAUSE button. I am not able to reproduce your comments.
> If I start a sim config and set halui.program.pause the running program wait for the [...]

I think we can press that into the skat.

gmoccapy, 2021-05-14:
OK, ist in Skat gedrückt ;-)

gmoccapy, 2021-05-14:
OK, I see, the missing / wrong translation  are caused due to de.po translation files not related to gmocapy. neverthereless, I will fix that. May be even this afternoon.

Norbert

hansu, 2021-05-14:
> OK, I see, the missing / wrong translation are caused due to de.po translation files not related to gmocapy. neverthereless, I will fix that. May be even this afternoon.

Maybe get some inspiration from https://github.com/LinuxCNC/linuxcnc/pull/1112/files

gmoccapy, 2021-05-14:
Halo Hans,

I just pushed some translation corrections and additions to **master**
The not appearing error after setting the limit switch is cause somewhere in the error channel, the message is not thrown if the limit switch is active, so this is not a gmoccapy issue.

I merged the changes I made in 2.8 also to master.

Please confirm I can close your pull request.

Norbert

hansu, 2021-05-14:
Why didn't you fix that in 2.8 as well?
[image]

gmoccapy, 2021-05-14:
I was working in master at that moment, so I merged In master, I will try to fix that also in 2.8 as soon as possible, my fault😢

hansu, 2021-06-07:
I think you can just cherry pick your commit 65503ba214cd8e126b6520857e786115dcb8e262 to 2.8. That would be nice because  I could close the issue and PR.

closed 2021-12-12 by hansu