LinuxCNC/linuxcnc #1117: Linking libtirpc fails on Ubuntu 20.10

cstrahan, 2021-03-26:
## Here are the steps I follow to reproduce the issue:

Attempt build on Ubuntu 20.10:

```bash
$ export PYTHONPATH=/usr/local/lib/python2.7/dist-packages/gtk-2.0:/usr/local/lib/python2.7/site-packages/:/usr/local/lib/python2.7/site-packages/gtk-2.0
$ ./configure --with-realtime=uspace --with-python=python2 --with-boost-python=boost_python27
$ BUILD_VERBOSE=1 make
```

Find build fails with:

```
Linking linuxcncrsh
g++ -L/home/cstrahan/src/linuxcnc/lib -Wl,-rpath,/home/cstrahan/src/linuxcnc/lib -ltirpc -o ../bin/linuxcncrsh -Wall -g -I. -I/home/cstrahan/src/linuxcnc/src/include -DULAPI -D_GNU_SOURCE -Os -DLOCALE_DIR=\"/usr/share/locale\" -DPACKAGE=\"linuxcnc\" objects/emc/usr_intf/emcrsh.o objects/emc/usr_intf/shcom.o ../lib/liblinuxcnchal.so.0 ../lib/liblinuxcnc.a ../lib/libnml.so.0 ../lib/liblinuxcncini.so.0 -lpthread
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_bytes'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_u_long'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_int'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_vector'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_double'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_long'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_float'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdrmem_create'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_short'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_u_char'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_u_int'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_u_short'
/usr/bin/ld: ../lib/libnml.so.0: undefined reference to `xdr_char'
collect2: error: ld returned 1 exit status
make: *** [emc/usr_intf/Submakefile:25: ../bin/linuxcncrsh] Error 1
```

If I manually tweak and run the last command so that `-ltirpc` comes after `libnml.so.0` then the error goes away and `linuxcncrsh` is linked successfully:

```
$ g++ -L/home/cstrahan/src/linuxcnc/lib -Wl,-rpath,/home/cstrahan/src/linuxcnc/lib -ltirpc -o ../bin/linuxcncrsh -Wall -g -I. -I/home/cstrahan/src/linuxcnc/src/include -DULAPI -D_GNU_SOURCE -Os -DLOCALE_DIR=\"/usr/share/locale\" -DPACKAGE=\"linuxcnc\" objects/emc/usr_intf/emcrsh.o objects/emc/usr_intf/shcom.o ../lib/liblinuxcnchal.so.0 ../lib/liblinuxcnc.a ../lib/libnml.so.0 ../lib/liblinuxcncini.so.0 -lpthread -ltirpc ; echo $?
0
```

With the following patch, I can successfully build the entirety of LinuxCNC:

```patch
diff --git a/src/emc/iotask/Submakefile b/src/emc/iotask/Submakefile
index e01cf8b11..5ec116e72 100644
--- a/src/emc/iotask/Submakefile
+++ b/src/emc/iotask/Submakefile
@@ -4,11 +4,11 @@ USERSRCS += $(IOSRCS) $(IOV2SRCS)
 
 ../bin/io: $(call TOOBJS, $(IOSRCS)) ../lib/liblinuxcnc.a ../lib/libnml.so.0 ../lib/liblinuxcnchal.so.0 ../lib/liblinuxcncini.so.0
 	$(ECHO) Linking $(notdir $@)
-	@$(CXX) $(LDFLAGS) -o $@ $^ 
+	@$(CXX) $(LDFLAGS) -o $@ $^ -ltirpc
 
 ../bin/iov2: $(call TOOBJS, $(IOV2SRCS)) ../lib/liblinuxcnc.a ../lib/libnml.so.0 ../lib/liblinuxcnchal.so.0 ../lib/liblinuxcncini.so.0
 	$(ECHO) Linking $(notdir $@)
-	@$(CXX) $(LDFLAGS) -o $@ $^
+	@$(CXX) $(LDFLAGS) -o $@ $^ -ltirpc
 
 TARGETS += ../bin/io ../bin/iov2
 
diff --git a/src/emc/sai/Submakefile b/src/emc/sai/Submakefile
index 9bc28615a..ca64c6f47 100644
--- a/src/emc/sai/Submakefile
+++ b/src/emc/sai/Submakefile
@@ -9,7 +9,7 @@ INCLUDES += emc/sai
 ../bin/rs274: $(call TOOBJS, $(SAISRCS)) ../lib/librs274.so.0 ../lib/liblinuxcnc.a ../lib/libnml.so.0 \
 	../lib/liblinuxcnchal.so.0 ../lib/liblinuxcncini.so.0 ../lib/libpyplugin.so.0
 	$(ECHO) Linking $(notdir $@)
-	$(Q)$(CXX) $(LDFLAGS) -o $@ $^ $(ULFLAGS) $(BOOST_PYTHON_LIBS) -l$(LIBPYTHON) $(READLINE_LIBS)
+	$(Q)$(CXX) $(LDFLAGS) -o $@ $^ -ltirpc $(ULFLAGS) $(BOOST_PYTHON_LIBS) -l$(LIBPYTHON) $(READLINE_LIBS)
 
 $(patsubst ./emc/sai/%,../include/%,$(wildcard ./emc/sai/*.h)): ../include/%.h: ./emc/sai/%.h
 	cp $^ $@
diff --git a/src/emc/task/Submakefile b/src/emc/task/Submakefile
index 0db167a92..0aabc3fe2 100644
--- a/src/emc/task/Submakefile
+++ b/src/emc/task/Submakefile
@@ -5,7 +5,7 @@ USERSRCS += $(EMCSVRSRCS)
 
 ../bin/linuxcncsvr: $(call TOOBJS, $(EMCSVRSRCS)) ../lib/liblinuxcnchal.so.0 ../lib/liblinuxcnc.a ../lib/libnml.so.0 ../lib/liblinuxcncini.so.0
 	$(ECHO) Linking $(notdir $@)
-	@$(CXX) $(LDFLAGS) -o $@ $^
+	@$(CXX) $(LDFLAGS) -o $@ $^ -ltirpc
 TARGETS += ../bin/linuxcncsvr
 
 # disabled:	emc/task/iotaskintf.cc
diff --git a/src/emc/usr_intf/Submakefile b/src/emc/usr_intf/Submakefile
index 1ddd6de48..6254c9b4c 100644
--- a/src/emc/usr_intf/Submakefile
+++ b/src/emc/usr_intf/Submakefile
@@ -22,20 +22,20 @@ TARGETS += ../tcl/linuxcnc.so
 
 ../bin/linuxcncrsh: $(call TOOBJS, $(EMCRSHSRCS)) ../lib/liblinuxcnchal.so.0 ../lib/liblinuxcnc.a ../lib/libnml.so.0 ../lib/liblinuxcncini.so.0
 	$(ECHO) Linking $(notdir $@)
-	$(Q)$(CXX) $(LDFLAGS) -o $@ $(ULFLAGS) $^ -lpthread
+	$(Q)$(CXX) $(LDFLAGS) -o $@ $(ULFLAGS) $^ -ltirpc -lpthread
 TARGETS += ../bin/linuxcncrsh
 
 ../bin/schedrmt: $(call TOOBJS, $(EMCSCHEDSRCS)) ../lib/liblinuxcnchal.so.0 ../lib/liblinuxcnc.a ../lib/libnml.so.0 ../lib/liblinuxcncini.so.0
 	$(ECHO) Linking $(notdir $@)
-	$(Q)$(CXX) $(LDFLAGS) -o $@ $(ULFLAGS) $^ -lpthread
+	$(Q)$(CXX) $(LDFLAGS) -o $@ $(ULFLAGS) $^ -ltirpc -lpthread
 TARGETS += ../bin/schedrmt
 
 ../bin/linuxcnclcd: $(call TOOBJS, $(EMCLCDSRCS)) ../lib/liblinuxcnchal.so.0 ../lib/liblinuxcnc.a ../lib/libnml.so.0 ../lib/liblinuxcncini.so.0
 	$(ECHO) Linking $(notdir $@)
-	$(Q)$(CXX) $(LDFLAGS) -o $@ $(ULFLAGS) $^
+	$(Q)$(CXX) $(LDFLAGS) -o $@ $(ULFLAGS) $^ -ltirpc
 TARGETS += ../bin/linuxcnclcd
 
 ../bin/halui: $(call TOOBJS, $(HALUISRCS)) ../lib/liblinuxcnc.a ../lib/liblinuxcncini.so.0 ../lib/libnml.so.0 ../lib/liblinuxcnchal.so.0
 	$(ECHO) Linking $(notdir $@)
-	$(Q)$(CXX) $(LDFLAGS) -o $@ $(ULFLAGS) $^ 
+	$(Q)$(CXX) $(LDFLAGS) -o $@ $(ULFLAGS) $^ -ltirpc
 TARGETS += ../bin/halui
```

## This is what I expected to happen:

I expected LinuxCNC to build fine. 

## This is what happened instead:

Instead I run into a linking issue.

## Information about my hardware and software:

 * I am using this Linux distribution and version: Ubuntu 20.10
 * I am using this kernel version: 5.8.0
 * I am using this compiler: gcc 10.2.0
 * I am using this linker: GNU ld 2.35.1
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version: 2.8 tag
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):  N/A
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): N/A

rene-dev, 2021-03-26:
Are you building 2.8 or master? Afaik this is already fixed in master.

cstrahan, 2021-03-26:
2.8 (have a git clone and I checked out the 2.8 tag). Sorry for the noise if this is already fixed -- tried searching through the issues/PRs but couldn't find any mention of work in this area. Would you be able to drop a reference to the commit/PR, please? Curious what the fix looks like.

cstrahan, 2021-03-26:
Never mind, found it: https://github.com/LinuxCNC/linuxcnc/commit/0dc3ccf8355e7623b84d171eb34ef8054f45833a

cstrahan, 2021-03-26:
Since this is fixed in master, I'll go ahead and close. Thanks @rene-dev !

closed 2021-03-26 by cstrahan

rene-dev, 2021-03-26:
I can merge those changes into 2.8 if you require 2.8

cstrahan, 2021-03-26:
> I can merge those changes into 2.8 if you require 2.8

That would be much appreciated -- please and thanks!