LinuxCNC/linuxcnc #1120: G70 2nd Pass Cutter Return Path Issue

BSHoekstra, 2021-03-31:
I am learning to use the new G70 command in Linuxcnc 2.9  on my CNC Lathe and found what may be a bug.  When using P=2 to execute two profile finish passes, the cutter return path on the 2nd pass is not the same as the return path on the 1st pass.  On the 2nd cutter return pass, the cutter digs into the work piece and then drags across the workpiece...it's like it is ignoring the retract distance.  There are no issues with cutter return path on the 1st pass.  

Example:   G70 Q100 X15 Z3.041 E0 D0.125 P2

## This is what I expected to happen:  When using G70, the cutter return pass on all additional profile finish passes P>1 follow the exact same pass as the cutter return path taken on the 1st pass P=1.

## Information about my hardware and software:

 * I am using this Linux distribution and version:  Buster 10
 * I am using this kernel version:  4.19.0-14-rt-amd64 #1 SMP preempt RT Debian 4.19.171-2
 * I am running ...
   * [X ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI): AXIS
 * I am using this interface hardware vendor and chipset:  parallel port, Dell 3020

[image]
[G71 King_ngc.txt](https://github.com/LinuxCNC/linuxcnc/files/6232998/G71.King_ngc.txt)

mark-v-d, 2021-04-10:
I've tried to reproduce this error, but the attached file does not match
the attached picture. It also produces (and correctly so) a "Straight
feed in concave corner cannot be reached by the tool without gouging".

Can you attach the file which re-produces the error?
Does the error still occur if you turn cutter compensation
(G42.1) off?

BSHoekstra, 2021-04-11:
Thank you for looking into this and replying.  I really do appreciate the help and I hope the additional information in this reply 
[image]
[image]

will be helpful.

1)  As for the "Straight feed in concave corner cannot be reached by the tool without gouging"...please use the tool table that I have attached.  Note that I use 35 degree diamond shaped cutters so this contour gouging error is not an issue.

2) Sorry for any confusion created where the ngc file vs. the picture attached in the initial help request did not match.  The ngc file attached was the one that I actually used on the lathe, but the cut depth was so shallow that it made it difficult to show the issue on the picture.  So I basically made the cut depth deeper to create the picture.   Regardless the file attached in this response matches the picture.

3)  Turning G42.1 off does resolve the issue where the cutter digs in on the 2nd return path; however, the 2nd return path is NOT identical to the 1st return pass and I think it should be.   This is shown pictorially in the attached file.
[image]
[image]
[G71 King_QUICK_ngc.txt](https://github.com/LinuxCNC/linuxcnc/files/6291068/G71.King_QUICK_ngc.txt)

[image]

mark-v-d, 2021-04-11:
Thanks for the updated file. I still cannot get cutter compensation on
because of the gouging error, and I am using your tooltable. This is weird
and something I will look into.

Without the cutter compensation however the cycle works as intended. The
final move in your O100 sub is from z=-63 -> z=-75. After that it moves
to the starting point.

Please notice that the program makes a total of three passes along the
final contour, the first pass is actually from the G71.2 cycle. After
a G7[12] cycle finishes the path it always returns to the starting X
(for G71) or Z (for G72) before returning to the starting point.  This is
something I added to prevent bumping into features left on the workpiece.
The G70 cycle does not do this, since it does not know what direction
is safe, so it's up to the user to add this to the toolpath. (See attached file)

[G71_king.ngc.gz](https://github.com/LinuxCNC/linuxcnc/files/6291546/G71_king.ngc.gz)

BSHoekstra, 2021-04-11:
Thank you.  :) +1: 

G42.1 Tool Compensation:   I'm not sure why you are getting a gouging error using my tool table and I am not getting that error.... that is weird.  However, I had 3 thoughts that you could consider to test the NGC file with G42.1 turned ON.  1)  Ensure Tool 1 is in orientation 2 and not 6 in the tool table, or 2) Just change all tools to cutter 0 (I think tool 0 ignores all errors like this) and 3) I assume you just commented out the M112 messaging code "To move the tool post" (I probably should have done this before I attached the file....sorry).  Regardless, I tested the file you sent me with G42.1 turned ON and turned OFF and there was no difference in the tool paths between these 2 configurations.

G70 2nd Return Path Issue:  The code lines you added to the file solved the gouging problem and everything works perfectly with G42.1 turned ON and OFF.  First...THANK YOU!  :)  Second...I have now idea what your code does exactly nor why it works.  Can you please quickly explain?  and Can you please point me to where this type of code is referenced in the Linux G-code documentation so I can learn more about it?

O100 SUB   	
#1=#<_X> (Mark added this code)
G01 X3.5 Z1.0
G01 Z-7.0
X5.0
Z-10.0
X4.5 Z-13.0
Z-15.0
X9.3
Z-17
X5.3 Z-24
X5.3
Z-30
X9.3
G01 X9.3 Z-35
G02 X8.0 Z-55.0 R15
G01 X12.5 Z-63.0
Z-75
X#1 (Mark added this code)
O100 ENDSUB

The two G70 tool return paths (with P=2) using the code Mark added are shown below.  These tool paths clear all work piece features and resolve the tool gouging issue.  Mark please consider having the developers add your code lines in the G70 example files provided with the LinuxCNC download because other beginners like me will likely run into the same issue.

[image]
[image]

THANK YOU VERY MUCH for your help with this!  :)  This was VERY helpful...Wow.

Thank you again.

Have a nice day,

Barry

mark-v-d, 2021-04-11:
The first line store the current value of the X position in parameter
number 1. See http://www.linuxcnc.org/docs/html/gcode/overview.html#_parameters
Numbers 1 to 31 are intented as local parameters.

The value of this parameter then becomes 15 in "G70 Q100 X15 Z3.041
D0.05 E0 P2" (Assuming my latest patch has been applied)

After the final Z move, it just moves the X to whatever was stored in #1.

For your amusement I've attached a more elaborate example. This allows
you to cut an internal morse taper.

As for the gouging error, I really have no idea what is going
on. Basically I should have gotten it without the final X move, but not
with that move. It is also a mystery to me why you don't get that error.
[Morse_taper_internal.ngc.gz](https://github.com/LinuxCNC/linuxcnc/files/6291998/Morse_taper_internal.ngc.gz)

BSHoekstra, 2021-04-12:
Thank you for all the help and resolving the issue.

Have a great week!  :)

closed 2021-04-12 by BSHoekstra