LinuxCNC/linuxcnc #1147: halmodule: Improvements and `halcmd` parity discussion

bendemott, 2021-05-06:
@c-morley - I've been following all your great work on the halmodule, great to see some improvements coming to it.

I had some thoughts before this is put into the next release and was hoping you might be open to some discussion.

I haven't been very involved in this project previously, but I customize a lot of things privately.
If your open to any of these ideas I'm happy to get them working / help / work with you.

## `hal.get_info_comps()`
- would be a nice to have to provide a component api call... similar to `halcmd show comp`
```
Loaded HAL Components:
ID      Type  Name                                            PID   State
    87  User  halcmd26632                                     26632 ready
    53  User  axisui                                          24597 ready
    51  User  inihal                                          24594 ready
    44  User  mega2560                                        24577 ready
    40  RT    toggle2nist                                           ready
    37  RT    toggle                                                ready
    34  RT    estop_latch                                           ready
    31  RT    pid                                                   ready
    28  RT    hm2_pci                                               ready
    25  RT    hostmot2                                              ready
    22  RT    __servo-thread                                        ready
    21  RT    motmod                                                ready
    18  RT    trivkins                                              ready
    12  User  halui                                           24562 ready
     6  User  iocontrol                                       24560 ready
```

## `hal.get_info_signals()`
Would it be good if this had parity with `halcmd show sig`
- this shows `DRIVER`, `READER` or `bi-directional` 
- it also contains the data type so you don't have to make 2 calls 
```
bit           FALSE  input-z-axis-limit
                         ==> axis.2.home-sw-in
                         ==> axis.2.neg-lim-sw-in
                         ==> axis.2.pos-lim-sw-in
                         <== mega2560.input-20
bit           FALSE  z-index-enable
                         <=> axis.2.index-enable
                         <=> hm2_5i25.0.encoder.02.index-enable
                         ==> pid.z.index-enable

```

## `hal.get_info_pins()`
- the `halcmd show pin` displays `owner`, `type`, `direction`, `value`, `name`, and `signal` (if connected)
- would providing the same / similar information to the cli make sense?
- including the attached signal is nice (it may be a bit too much code in C ?)
```
    44  bit   OUT         FALSE  mega2560.input-03 ==> input-servo-power
    44  bit   OUT          TRUE  mega2560.input-03-not
    44  bit   OUT         FALSE  mega2560.input-04 ==> input-front-door-closed
    44  bit   OUT          TRUE  mega2560.input-04-not
    44  bit   OUT         FALSE  mega2560.input-05 ==> input-front-door-open
    44  bit   OUT          TRUE  mega2560.input-05-not
    44  bit   OUT         FALSE  mega2560.input-06 ==> input-spindle-stopped
    44  bit   OUT          TRUE  mega2560.input-06-not
    44  bit   OUT         FALSE  mega2560.input-07 ==> input-spindle-speed-attained
```

c-morley, 2021-05-06:
Hey thanks for comments.
I added what I thought I needed for a future project.
I was too lazy to add everything that halcmd offers.
(I'm not very practiced with c++)
But I did use a dict of information so more could be added in the future without breaking anything.

I am always interested in new code/ideas.

Do you think the dict of information is the wrong approach?

What other mods have you done for yourself?

bendemott, 2021-05-06:
I think the `dictionary` approach is fine, especially for getting your code working quickly  :-)

It may be nice to have a proper python object with attributes, the reason being is calling `help()` in Python on the object or response of `hal.get*` would then produce the list of properties with doc-strings automatically inserted.
I think code with properties / real objects is a bit cleaner. 

I privately added to the `hal` module previously so I could write "io panels" like below for debugging.

In the next couple weeks I'll pull the latest and see if I can integrate some of my work for signals, parameters, pins and the rest of the halcmd functionality like I talked about above. Maybe I'll just start with signals (readers) and see if we can get that change.

[image]

Thanks for the response @c-morley - I'll update here, or create another issue when I have some code ready.