LinuxCNC/linuxcnc #1153: Qt Foam Display

satiowadahc, 2021-05-14:
I'm attempting to use gcodegraphics from QTVCP as a foam wire cutter. I've figured out how to load the model properly, but can't seem to find the code that moves the second axis.

I'm assuming its somewhere around rs274/glcanon.glcanondraw

Any suggestions on how to continue?

https://github.com/satiowadahc/linuxcnc/commit/4ad2861f90d0d9fbb5cc8e2a9043f364a2c9275c

c-morley, 2021-05-14:
I don't have any experience with foam capabilities.
i see you must set FOAM = 1 in the INI
and some the AXIUS sample uses magic comments in the Gcode:
(AXIS,XY_Z_POS,5)
(AXIS,UV_Z_POS,30)

magic comments are processed in glcanon.py

satiowadahc, 2021-05-14:
It currently loads the outlines properly on gcodegraphics and I see the comments are being processed. But when I run the program its as if the interpreter sees no XY.

Mine
[image]

Axis
[image]

c-morley, 2021-05-14:
well I haven't got the outlines to display, but I figured out the plotting problem while moving.

in qt_graphics.py the position_logger doesn't get set into foam mode.

c-morley, 2021-05-14:
I pushed to master some work that should help

satiowadahc, 2021-05-17:
Seems to work Good ! Thanks

I was halfway there, looks like but you figured it out way faster than I could've.

Thank you.

closed 2021-05-17 by satiowadahc

c-morley, 2021-05-18:
We aim to please :) Thanks for the report. It means foam doesn't work in gtk either.