LinuxCNC/linuxcnc #1158: M70/M72 restore of modal state fails with comma ',' in feed paramte

Luddi1, 2021-05-16:
In a subroutine for `REMAP=M6 modalgroup=6 ngc=...` I use M70/M72 to store and restore the modal state. When the feedrate changes between these in the subroutine I get the following G-Code error message which is displayed when loading a program: `M7x: restore_settings failed executing ' F123,0': Bad character ',' used.` The saved feedrate was F123 here. 
When the feedrate does not change this is not an issue, so a workaround is to separately store and restore the #<_feed> variable. 

My current idea where this originates: `snprinf` of a float changes behavior depending on the locale. I'm in Germany, where a decimal point is a comma. 
This is where it probably happens:  [linuxcnc/src/emc/rs274ngc/interp_convert.cc in line 2752](https://github.com/LinuxCNC/linuxcnc/blob/380bcd3b2582dfd9a6f936354a7d1faa7ded05bd/src/emc/rs274ngc/interp_convert.cc#L2752).

One solution would be to use `setlocale` prior to `snprintf` and then restore the locale as suggested here: https://stackoverflow.com/a/14753747

For completeness: 
## Here are the steps I follow to reproduce the issue:

 0. Use the german or french locale
 1. Put a subroutine for M6 in place in the ini file (`REMAP=M6 modalgroup=6 ngc=...`). Example here: https://forum.linuxcnc.org/10-advanced-configuration/5596-manual-tool-change-tool-lengh-touch-off?start=30#48235
 2. In the subroutine do an M70, change feed, M72
 3. load a gcode program

### This is what I expected to happen:
Uneventful restoration of the modal state.

### This is what happened instead:
G-Code error messae: `M7x: restore_settings failed executing ' F0,0': Bad character ',' used.`

### It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

### Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): debian. The linuxcnc iso
 * I am running ...
   * [ x] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 2.8.1
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): gmoccapy
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): mesa 7i96

andypugh, 2021-05-16:
Does F10,1 work in actual G-code in your Locale? I suspect that the answer is no, as G-code is rather old and was originally developed in the USA where they rarely consider the possibility of other countries doing things differently. 

It is interesting that the code only saves feedrate to one decimal place, and spindle speed to zero places. I think that is probably wrong too.

The answer might be to change:
`snprintf(buf,sizeof(buf)," F%.1f", float_saved[i]);`
to
snprintf(buf,sizeof(buf)," F%i.%04i", (int)floor(float_saved[i]), (int)(10000 * (float_saved - floor(float_saved)));
Which is a rather inelegant way to force a specific decimal radix, but feels better than changing LC_NUMERIC

andypugh, 2021-05-16:
The notes in this commit are interesting, in this regard. 

https://github.com/LinuxCNC/linuxcnc/commit/5772776053feaf8701e54c48ced6b932e8d3baee

Luddi1, 2021-05-17:
Same issue with F10,1. 

I just followed the Build LinuxCNC guide, checked out the 2.8.1 tag and applied this diff :
```
index b4a424a75..67c24536c 100644
--- a/src/emc/rs274ngc/interp_convert.cc
+++ b/src/emc/rs274ngc/interp_convert.cc
@@ -2712,7 +2712,7 @@ int Interp::gen_settings(double *current, double *saved, std::string &cmd)
            switch (i) {
            case 0: break; // sequence_number - no point in restoring
            case 1:
-               snprintf(buf,sizeof(buf)," F%.1f", saved[i]);
+               snprintf(buf,sizeof(buf)," F%i.%04i", (int)floor(saved[i]), (int)(10000 * (saved[i] - floor(saved[i]))));
                 cmd += buf;
                break;
            case 2:
```
That seems to resolve the issue. 
Elegant or not, this is robust to changes in the locale settings.

andypugh, 2021-05-17:
@Luddi1 Are you able to check this change from Jepler? 

To get the change as a patch file, use this URL:
https://github.com/jepler/linuxcnc/commit/d7666adb7d35d8fafbd9cbac0af3e1ef240d01df.patch

Luddi1, 2021-05-17:
I checked out jepler:locale-independent-rs274ngc and compiled and tested it. This resolves the issue. 

Thanks for the work guys :+1:

Luddi1, 2021-05-17:
Sorry. Maybe this should be closed when merged into master.

PR #1159 (2021-05-17): rs274: Temporarily force the LC_NUMERIC setting to C as needed
Code in the gcode interpreter assumes that number parsing/formatting
is done according to the C locale, as gcode part programs always use
"." as the decimal separator.

However, some of the user interfaces set the LC_NUMERIC to the user's
locale and then call into the interpreter to prepare previews of the
part program.

The last time I addressed this, it was by changing one routine to work via
an ostringstream, which is not affected by locales by default.
This time I took a different approach, which is to temporarily force the
LC_ [... cut]
commit d7666adb (2021-05-17): rs274: Temporarily force the LC_NUMERIC setting to C as needed

Code in the gcode interpreter assumes that number parsing/formatting
is done according to the C locale, as gcode part programs always use
"." as the decimal separator.

However, some of the user interfaces set the LC_NUMERIC to the user's
locale and then call into the interpreter to prepare previews of the
part program.

The last time I addressed this, it was by changing one routine to work via
an ostringstream, which is not affected by locales by default.
This time I took a different approach, which is to temporarily force the
LC [... cut]
closed 2021-05-17 by andypugh