LinuxCNC/linuxcnc #1168: Build with libeditreadline-dev where available (debian buster)

jepler, 2021-05-29:
Paraphrased from some bug filings on debian.org -- we _ACTUALLY_ try hard to depend on libreadline-gplv2-dev but maybe this is changing in Debian Buster?

> 
> This package depends on libreadline8 which is GPL-3+ licensed. According 
> to debian/copyright parts of your package are GPL-2-only licensed. If 
> that is also (transitively) the case for the binaries that link with 
> libreadline.so.8 it might be legally problematic (see 
> https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility).
> 
> There is an easy solution to the problem: Replacing the build dependency 
> libreadline-dev with libeditreadline-dev links with the BSD-licensed 
> libedit library which is a readline replacement.

jepler, 2021-05-29:
https://packages.debian.org/buster/libreadline-gplv2-dev  The package we use now is not available after buster

rene-dev, 2021-06-03:
fixed by https://github.com/LinuxCNC/linuxcnc/pull/1164/commits/bf688f182ece7e7c4488cbc49b2fbc5db9624265
maybe this package should be replaced by libeditreadline-dev?
https://github.com/LinuxCNC/linuxcnc/blob/master/scripts/travis-install-build-deps.sh#L2

closed 2021-06-03 by rene-dev