LinuxCNC/linuxcnc #1170: classicladder not working in current master

Sigma1912, 2021-06-01:
1. Built the current master branch as a rip install. All seems to be successful. 
 2. trying to run a hal file containing loadrt classicladder_rt .. fails with error "file not found"
 3. trying to open classicladder gui with the screen button in gmoccapy settings page fails with error ""Classicladder real-time component not detected"

andypugh, 2021-06-01:
Has this been fixed in the meantime? I just pulled the latest master. 
````make clean
./autogen.sh
./configure --with-realtime=uspace
make
sudo make setuid
halrun
loadrt classicladder_rt
````
And it worked

rene-dev, 2021-06-01:
whoops, I broke classicladder in gtk3 related buildsystem changes I did. I will fix this tomorrow.

commit efa2fe8d (2021-06-02): fix building classicladder with gtk2, closes #1170
closed 2021-06-02 by rene-dev