LinuxCNC/linuxcnc #1182: master-gtk3 - gmoccapy - settings

gmoccapy, 2021-06-06:
settings of Themes and sound does not work

andypugh, 2021-06-06:
I have been trying to get themes to work with Touchy today, and I think that there may be problems outside our control. 

Touchy now works with _some_ themes, but there are lots of warnings of changed property names.

gmoccapy, 2021-06-06:
HLo Andy, can you share your touch, stuff, so I can try to adapt that to gmoccapy

Final solution need to be to use css style for customizing the view of the gui

andypugh, 2021-06-06:
I was hoping that I could have a look to see if the same (naive) fixes would help, but at the moment I can't even start the gmoccapy sample configs in master-gtk3

rene-dev, 2021-06-07:
why does the ui need to set its theme? whats wrong with using the system theme?
@andypugh which error do you get when starting gmoccapy?

andypugh, 2021-06-07:
> @andypugh which error do you get when starting gmoccapy?

First a dialog box:
https://photos.app.goo.gl/Xys9JQeVkxkSsmdaA
Then a second dialog box (With Gmoccapy open behind it) 
https://photos.app.goo.gl/GboNxENjSonM5Kh88
Closing the dialog box closes Gmoccapy:

````
andypugh@buster:~/linuxcnc$ linuxcnc
LINUXCNC - 2.9.0~pre0
Machine configuration directory is '/home/andypugh/linuxcnc-dev/configs/sim/gmoccapy'
Machine configuration file is '4_axis.ini'
Starting LinuxCNC...
Found file(REL): ./core_sim4.hal
Note: Using POSIX realtime

trivkins: coordinates:xyzc
   Joint 0 ==> Axis X
   Joint 1 ==> Axis Y
   Joint 2 ==> Axis Z
   Joint 3 ==> Axis C
trivkins: Recommend: kinstype=both

Unexpected realtime delay on task 0 with period 100000
This Message will only display once per session.
Run the Latency Test and resolve before continuing.
Found file(REL): ./spindle_sim.hal
Found file(REL): ./simulated_home.hal
Unexpected realtime delay on task 0 with period 100000
This Message will only display once per session.
Run the Latency Test and resolve before continuing.

Found an error!
The following information may be useful in troubleshooting:

Traceback (most recent call last):
  File "/home/andypugh/linuxcnc-dev/lib/python/gladevcp/iconview.py", line 188, in __init__
    pixbuf = Gtk.IconTheme.get_default().load_icon("gtk-quit", 48, 0)
gi.repository.GLib.GError: gtk-icon-theme-error-quark: Icon 'gtk-quit' not present in theme Tango (0)

task: main loop took 0.538538 seconds
task: main loop took 0.556755 seconds
task: main loop took 0.555996 seconds
task: main loop took 0.578372 seconds
task: main loop took 0.013969 seconds
task: main loop took 0.034967 seconds
task: main loop took 0.030826 seconds
task: main loop took 0.032279 seconds
task: main loop took 0.011937 seconds
task: main loop took 0.021109 seconds
'IconFileSelection' object has no attribute 'store'
'IconFileSelection' object has no attribute 'iconView'
'IconFileSelection' object has no attribute 'store'
'IconFileSelection' object has no attribute 'iconView'
'IconFileSelection' object has no attribute 'on_btn_jump_to'
'IconFileSelection' object has no attribute 'store'
'IconFileSelection' object has no attribute 'iconView'
'IconFileSelection' object has no attribute 'store'
'IconFileSelection' object has no attribute 'iconView'
Icon from stock
0  =  /home/andypugh/linuxcnc-dev/bin/gmoccapy
1  =  -ini
2  =  /home/andypugh/linuxcnc-dev/configs/sim/gmoccapy/4_axis.ini
Entry = trivkins
Entry = coordinates=xyzc
found the following coordinates xyzc

**** GMOCCAPY GETINIINFO **** 
Number of joints = 4
[... 148 lines cut ...]
Jog Pin Changed
z-
Jog Button released = z-
Jog Pin Changed
c+
Jog Button released = c+
Jog Pin Changed
c-
Jog Button released = c-
on tool chnage

(gmoccapy:8138): Gtk-WARNING **: 19:15:00.178: Invalid icon size 48

(gmoccapy:8138): Gtk-WARNING **: 19:15:00.178: Invalid icon size 48

_kill_dynamic_childs
Shutting down and cleaning up LinuxCNC...
task: 237979 cycles, min=0.000005, max=0.578372, avg=0.001121, 16 latency excursions (> 10x expected cycle time of 0.001000s)
Note: Using POSIX realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/andypugh/linuxcnc_debug.txt
and
    /home/andypugh/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
````

gmoccapy, 2021-06-07:
No icon gtk quit on theme tango

Can you change the theme and try with clearlooks?
Even this need to be fixed by me🙄

andypugh, 2021-06-07:
"Clearlooks" isn't the same sort of "theme" as "Tango". 
Tango seems to be an "icon theme" and is also, I think, the default with xfce / the LinuxCNC ISO. 
I selected "Menta" instead and Gmoccapy started right up.

gmoccapy, 2023-10-07:
I do not know who has fixed that, but it is working as expected

PR #1411 (2021-12-12): Feature/update gtk icons
Since GTK 3.10 the usage of the so called "stock icons" is deprecated.
Instead named icons according to the [Freedesktop icon naming spec](https://specifications.freedesktop.org/icon-naming-spec/icon-naming-spec-latest.html) should be used.

I've changed them in gmoccapy related files.

This should fix the problems mentioned here: #1182 

Closes: #1182
closed 2023-10-07 by gmoccapy