LinuxCNC/linuxcnc #1188: Error with RIP build using --with-python=python3

gcanzalo, 2021-06-15:
Attempting RIP build for rtai using the following:

```
make clean
./autogen.sh
./configure --with-python=python3
make
```

Produces the following:

  ```
CC [M]  /home/jerry/src/linuxcnc/src/objects/hal/drivers/pcl720.o
  LD [M]  /home/jerry/src/linuxcnc/src/pcl720.o
  CC [M]  /home/jerry/src/linuxcnc/src/objects/hal/drivers/mesa_uart.o
  LD [M]  /home/jerry/src/linuxcnc/src/mesa_uart.o
  CC [M]  /home/jerry/src/linuxcnc/src/objects/hal/drivers/pluto_step.o
Traceback (most recent call last):
  File "modsilent.py", line 14, in <module>
    m = duplicate_warning.match(line)
TypeError: cannot use a string pattern on a bytes-like object
make: *** [Makefile:501: modules] Error 1
```

gcanzalo, 2021-06-15:
Deleted my local repository and cloned it again. Now getting this error:
```
Preprocessing pluto_servo.comp
/usr/bin/python3 hal/drivers/rbf2h.py -g \
    -c "This is a component of pluto_servo, a PWM servo driver and" \
    -c "quadrature counter for emc2" \
    -c "Copyright (C) 2006, 2007 Jeff Epler" \
    hal/drivers/pluto_servo_firmware/pluto_servo.rbf > hal/drivers/pluto_servo_rbf.h
Traceback (most recent call last):
  File "hal/drivers/rbf2h.py", line 60, in <module>
    h = open(args[0]).read();
  File "/usr/lib/python3.7/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
make: *** [hal/drivers/Submakefile:2: hal/drivers/pluto_servo_rbf.h] Error 1
```

Am I doing something wrong compiling with python3 support?

rene-dev, 2021-06-15:
No, that's the way to do it. Which distro do you use?

gcanzalo, 2021-06-15:
Debian 10

```
uname -a
Linux proto 4.14.174-rtai-amd64 #3 SMP PREEMPT Wed Apr 22 12:38:32 BST 2020 x86_64 GNU/Linux

```

rene-dev, 2021-06-15:
I created a branch rtai-python3 which might fix the problems, can you please try? I do not have any system that can run rtai.

andypugh, 2021-06-15:
I can reproduce the problem here. Which is a start.

rene-dev, 2021-06-15:
@gcanzalo  try this branch: https://github.com/LinuxCNC/linuxcnc/pull/1189

gcanzalo, 2021-06-15:
Thanks for the quick reply. I'm not a coder nor am I terribly familiar with git, but I think I have the correct branch checked out. Inital build produced the following:

```
Traceback (most recent call last):
  File "hal/drivers/rbf2h.py", line 64, in <module>
    print("%3d," % ord(c), end=' ')
TypeError: ord() expected string of length 1, but int found
make: *** [hal/drivers/Submakefile:2: hal/drivers/pluto_servo_rbf.h] Error 1

```

I removed the ord() and built again, which produced this:

  Building modules, stage 2.
  MODPOST 177 modules
Traceback (most recent call last):
  File "modsilent.py", line 14, in <module>
    m = duplicate_warning.match(line)
TypeError: cannot use a string pattern on a bytes-like object
make: *** [Makefile:501: modules] Error 1

andypugh, 2021-06-16:
It looks like you did not have the latest update from jepler.  Which isn't really a surprise as it hadn't been merged yet...

I have just merged it, can you try again? 

git reset --hard origin/master-gtk3
git pull
make clean
./autogen.sh
./configure --with-python=python3
make

gcanzalo, 2021-06-16:
@andypugh:

Same error:

Traceback (most recent call last):
  File "modsilent.py", line 14, in <module>
    m = duplicate_warning.match(line)
TypeError: cannot use a string pattern on a bytes-like object
make: *** [Makefile:501: modules] Error 1

andypugh, 2021-06-16:
The "ord()" issue is fixed, this is the next one to look at, I suppose.

andypugh, 2021-06-16:
It looks like 
````
 m = duplicate_warning.match(line.decode('utf-8'))
````
Might fix it.

gcanzalo, 2021-06-16:
That got it! Thanks!

Anything to worry about here?

Runtest: 236 tests run, 233 successful, 3 failed + 0 expected
Failed: 
	tests/pyhal
	tests/realtime-math
	tests/remap/remap-io

rene-dev, 2021-06-16:
Can you cd into the test, and run test.sh it will give a detailed error

gcanzalo, 2021-06-16:
pyhal:
```
./test
insmod: ERROR: could not insert module /usr/realtime-4.14.174-rtai-amd64/modules/rtai_hal.ko: File exists
HAL: ERROR: duplicate signal 'testSig-S'
Traceback (most recent call last):
  File "./test", line 33, in <module>
    c.sigNew("testSig-S", halType.SIGNED)
  File "/home/jerry/src/linuxcnc/lib/python/pyhal.py", line 230, in sigNew
    raise HalException('failed to create signal "{0}" with code {1}'.format(name, result))
pyhal.HalException: failed to create signal "testSig-S" with code -22

```
remap/remap-io:
```
bash test.sh
**********  Testing python remaps
----Testing M62/M63 digital output w/motion----
cmd: "M62 P1"
    M62 remapping pre: motion dout0=0; hal d_out=0
    M62 remapping post: motion dout0=1; hal d_out=1
cmd: "M63 P1"
    M63 remapping pre: motion dout0=1; hal d_out=1
    M63 remapping post: motion dout0=0; hal d_out=0
cmd: "M62 P1"
    M62 remapping pre: motion dout0=0; hal d_out=0
    M62 remapping post: motion dout0=1; hal d_out=1
cmd: "M63 P1"
    M63 remapping pre: motion dout0=1; hal d_out=1
    M63 remapping post: motion dout0=0; hal d_out=0
----Testing M64/M65 digital output, immediate----
cmd: "M64 P1"
    M64 remapping pre: motion dout0=0; hal d_out=0
    M64 remapping post: motion dout0=1; hal d_out=1
cmd: "M65 P1"
    M65 remapping pre: motion dout0=1; hal d_out=1
    M65 remapping post: motion dout0=0; hal d_out=0
cmd: "M64 P1"
    M64 remapping pre: motion dout0=0; hal d_out=0
    M64 remapping post: motion dout0=1; hal d_out=1
cmd: "M65 P1"
    M65 remapping pre: motion dout0=1; hal d_out=1
    M65 remapping post: motion dout0=0; hal d_out=0
----Testing M66 digital input----
cmd: "M66 P1"; input: 1.0000
    M66 remapping pre:  5399=0.000000; 100=0.000000; ain0=0.00; din0=0
    M66 remapping post:  5399=1.000000; 100=1.000000; ain0=0.00; din0=1
cmd: "M66 P1"; input: 0.0000
    M66 remapping pre:  5399=1.000000; 100=1.000000; ain0=0.00; din0=1
    M66 remapping post:  5399=0.000000; 100=0.000000; ain0=0.00; din0=0
----Testing M66 analog input----
cmd: "M66 E1 L0"; input: 42.1300
    M66 remapping pre:  5399=0.000000; 100=0.000000; ain0=0.00; din0=0
    M66 remapping post:  5399=42.130000; 100=42.130000; ain0=42.13; din0=0
cmd: "M66 E1 L0"; input: -13.4200
    M66 remapping pre:  5399=42.130000; 100=42.130000; ain0=42.13; din0=0
    M66 remapping post:  5399=-13.420000; 100=-13.420000; ain0=-13.42; din0=0
----Testing M67 analog output w/motion----
cmd: "M67 E1 Q42.13"
    M67 remapping pre: motion aout0=0.00; hal a_out=0.00
    M67 remapping post: motion aout0=42.13; hal a_out=42.13
cmd: "M67 E1 Q-13.42"
    M67 remapping pre: motion aout0=42.13; hal a_out=42.13
    M67 remapping post: motion aout0=-13.42; hal a_out=-13.42
cmd: "M67 E1 Q0.00"
    M67 remapping pre: motion aout0=-13.42; hal a_out=-13.42
    M67 remapping post: motion aout0=0.00; hal a_out=0.00
----Testing M68 analog output, immediate----
cmd: "M68 E1 Q42.13"
    M68 remapping pre: motion aout0=0.00; hal a_out=0.00
[... 61 lines cut ...]
    M68 remapping pre: motion aout0=0.00; hal a_out=0.00
    M68 remapping post: motion aout0=42.13; hal a_out=42.13
cmd: "M68 E1 Q-13.42"
    M68 remapping pre: motion aout0=42.13; hal a_out=42.13
    M68 remapping post: motion aout0=-13.42; hal a_out=-13.42
cmd: "M68 E1 Q0.00"
    M68 remapping pre: motion aout0=-13.42; hal a_out=-13.42
    M68 remapping post: motion aout0=0.00; hal a_out=0.00

```
realtime-math:
```
bash test.sh
+ halcompile --install rtmath.comp
make KBUILD_EXTRA_SYMBOLS=/home/jerry/src/linuxcnc/rtlib/Module.symvers -C /usr/src/linux-headers-4.14.174-rtai-amd64 SUBDIRS=`pwd` CC=gcc V=0 modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.174-rtai-amd64'
  CC [M]  /tmp/tmpdy06ivl6/rtmath.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /tmp/tmpdy06ivl6/rtmath.mod.o
  LD [M]  /tmp/tmpdy06ivl6/rtmath.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.14.174-rtai-amd64'
cp rtmath.ko /home/jerry/src/linuxcnc/rtlib/
+ halrun dotest.hal
```

andypugh, 2021-06-16:
> Anything to worry about here?
> Runtest: 236 tests run, 233 successful, 3 failed + 0 expected

I think it is something for us to worry about, but you should be good to go :-)

gcanzalo, 2021-06-16:
Got it hooked up to my machine and ran into more errors with lib/python/python_event. I edited the following lines in that file:

line 42: a = fcntl.ioctl(f, SZ(EVIOCGBIT+EV[o], sz), '\0' * **int(**((sz+7)/8**)**))
line 45: [deleted]
line 793: if name.find(pattern**.encode('utf-8')**) != -1 or fnmatch.fnmatch(name, pattern**.encode('utf-8')**):

Now LinuxCNC throws error:

Can't find module 'hm2_eth' in /home/jerry/src/linuxcnc/rtlib

Which indeed isn't there but is needed for my Mesa 7i96...

andypugh, 2021-06-16:
Ah. You can't use hm2_eth with kernel realtime (default RTAI)  You need to configure --with-realtime=uspace
(This _might_ actually work with RTAI as there is a userspace mode for RTAI. It's worth a try before you go looking for the preempt-RT kernel)

gcanzalo, 2021-06-16:
Oh. I've been running preempt and having issues with excursions, hence the interest in RTAI. That's a serious bummer. I'll try a preempt build in RTAI...

andypugh, 2021-06-17:
You mean uspace build in RTAI?

jepler, 2021-06-17:
There is a variant of "uspace" that can use rtai realtime (sometimes called LXRT), but it is not suitable for hm2_eth either.  Only the one which uses the PREEMPT RT kernel and the POSIX/pthreads API is suitable for hm2_eth.

gcanzalo, 2021-06-17:
@andypugh : Yes, sorry I wasn't clear.

@jepler: So there is no sense in trying the uspace version in RTAI?

Just curious why RTAI and ethernet don't get along? Is it insurmountable?

rene-dev, 2021-06-17:
short answer is rtai runs "below" linux kernel, and cant use any drivers that are present in Linux. there are Ethernet extensions for rtai, but they require special drivers, and linuxcnc does not make use of that.
rt-preempt runs just like any normal thread, and can use ethernet just like anything else. it should give good performance, if you have problems its likely to be unsuitable hardware, or some bios settings.

unfortunately there is currently no good documentation anywhere how to get good performance with rt-preempt.
most things on the wiki are very outdated.

can this issue be closed?

gcanzalo, 2021-06-17:
Yes. Thank you all for the help.

closed 2021-06-17 by gcanzalo