LinuxCNC/linuxcnc #1220: Raspberry Pi QtVCP Crashing

satiowadahc, 2021-08-06:
I'm making an issue here as I investigate it this weekend if anyone has some more insights. 

Using both the premade distribution image and Raspbian it appears I can't seem to run a PyQt5 based interface. All the GTK interfaces seems to be ok.

Most of the time it is a singular `/usr/bin/linuxcnc: line 930 Segmentation Fault` with little other information.

I made a display that removes gcode graphics so the OpenGL issue is not the root cause.

Twice it has been a mutex error but I didn't think to copy the exact error.

Running a bare bones PyQt5 "App". Everything runs up until `app.exec()`
 
Haven't tried any python2 code. 
Tried Rasbian and the prebuilt image
Tried my own displays and QtVCP
Pi4 2Gb and 4Gb.
Everything tried does work on my laptop and desktop. (Ubuntu 20.04)

c-morley, 2021-08-07:
Are you saying a plain pyqt5 app crashes or a plain qtvcp app crashes?

satiowadahc, 2021-08-07:
both?

I took it out of a class and have a script version working

```python3
#! /usr/bin/python3
from PyQt5 import QtWidgets
import linuxcnc
import hal

s = linuxcnc.stat()
c = hal.component("c")
c.ready()

app = QtWidgets.QApplication([])
label = QtWidgets.QLabel(f"s.actual_position[0]")
label.show()
app.exec()
```

satiowadahc, 2021-08-08:
Appears to a pi specific threading library. 
https://code.woboq.org/userspace/glibc/nptl/pthread_mutex_lock.c.html#431

```
python3: pthread_mutex_lock.c:433: __pthread_mutex_lock_full: Assertion `INTERNAL_SYSCALL_ERRNO (e, __err) != ESRCH || !robust' failed.
```

satiowadahc, 2021-08-08:
Hmm looks like an extra window I forgot to remove from the ini file loaded OriginOffsetDialog. Removing that reference now allows it to load but as soon as I load OriginOffsetDialog, app.exec() causes a seg fault. The last thing I can find that executes right before the seg fault:

```bash
    rtn = self.app.exec()
  File "/usr/lib/python3/dist-packages/hal_glib.py", line 703, in update
    self.emit('periodic')
  File "/usr/lib/python3/dist-packages/qtvcp/qt_tstat.py", line 367, in periodic_check
    traceback.print_stack()
```
From Wolf-Fencing
```python
    def periodic_check(self, w):
        print("periodic")
        if self._delay < 9:
            traceback.print_stack()
            self._delay += 1
            return
```

satiowadahc, 2021-08-08:
Wondering if its a conflict between PyQt5 and GLib event loops. That would explain the occasional Mutex error

c-morley, 2021-08-08:
that is what I was going to suggest you test if it was a qtvcp only problem.
The qt/glib is a sensitive problem on x86.
You can see in core.py, class STATUS, function set_timer.

I would comment out the timer and see if that has an effect.

satiowadahc, 2021-08-08:
Commenting out those two lines and replacing with pass doesn't seem to have much effect. Commenting out emit('periodic') in two spots in hal_glib does seems to allow it "load" but everything's very frozen. 

Yard work is calling so I'll have to drop it for today.

cakesalob, 2021-08-08:
As a baseline, try loading the sim/qtvcp_screens/tester  and sim/axis/qtvcp/qtvcp_tab , those 2 should load without error, without doing anything fancy. those dont use the qt5_graphics, so if those dont load something is wrong.  I just tested both on py3 and old py2 master versions with no issue. I use a custom offset tab with axis using the originoffset widget and that still works in py2 also. I only have every gotten segmentation errors when Ive been actively screwing with the qt5 install flags/rpi4 graphics drivers in attempt to resolved the qt5_graphics issue
The graphics not loading didnt used to be a process killing critical error , but it is again for some reason. Periodically I remove the graphics in creator and most of the guis run fine without them.

satiowadahc, 2021-08-08:
Current Status: OriginOffsetDialog and graphics are the only two things not loading and causing seg faults.

For me the graphics aren't super critical, most of our mills I make a plane jane "push this to home" and "push this to run the program" interface. Our operators appreciate a basic thing with out extra features. Graphics I normally only have in simulations.

I'm thinking If its a known issue of glib and qt conflicting I might just remove the problem code of Originoffsetdialog for now, and close this issue.

cakesalob, 2021-08-08:
Is it both Originoffset widgets or just the Originoffsetdialog ? I just did my originoffset tab for py3 and it doesnt have any issues, but im just using originoffset, not originoffsetdialog

c-morley, 2021-08-08:
If the problem is 'periodic' being called, not the actual Gobject timer, then that is a new problem.
And when I say glib/qt is a problem in x86, I mean they must be sequenced correctly.
The sequence was different with pyqt4

In the future I may need to switch to qt signals rather then glib.

But I'm still not positive that is your actual fail case here.

rene-dev, 2021-08-19:
old, but could this be related? https://www.qt.io/blog/2006/02/24/qt-and-glib

satiowadahc, 2021-09-13:
40 some hours later of compiling QT from scratch and gcodegraphics seems to work now with QT5.15. 

Origin Offset Dialogue still crashes it, I don't know if any of our machinists actually use it vs just entering gcode's so I'm not super worried about it for this project. If I have time I'll come back and do some more investigating.