LinuxCNC/linuxcnc #1221: python 2 >> python 3 conversion issue __init__.py

robertspark, 2021-08-06:
master (current 2.9)

linuxcnc\lib\python\gladevcp\ ___init___.py

line 31:             for fname in glob.glob(extradir + "/*.py"):

&

line 57:         for fname in glob.glob(auxdir + "/*.py"):

**_type error must be str not bytes_** 

tried to have a go at sorting / fixing the error myself..... but I've not got a clue {tried a few things, quite a few.... no dice}

dngarrett, 2021-08-07:
bda8be8b563e7a  addresses #1220
if RIP: update and rebuild
if deb install: get current branch from buildbot

robertspark, 2021-08-07:
hi Dewey, I did get the buildbot yesterday, and it still doesn't work, I
will update again today and report back.

rene-dev, 2021-08-19:
what exactly are you doing to produce this error?

robertspark, 2021-08-25:
sorted now via clean instal of MX Linux and installing required dependencies, and a run in place install.

sorry I don't have the time and patience to detail the issue experienced at the time..... presume it's probably sorted.... who knows...the best I can suggest is please try to install a clean fresh 2.8.2 linuxcnc install from iso..... upgrade to master via buildbot..... install your config files...... and see if Gmoccapy and QTPlasmaC runs fine as it did not for me at the time and kept throwing errors.

closed 2021-08-25 by robertspark