LinuxCNC/linuxcnc #1222: Python 2>> 3 + GTK

robertspark, 2021-08-06:
linuxcnc\lib\python\gladevcp\**iconview.py**

line 188 in __init__
pixbuf = GtkIconTheme.get_default().load_icon("gtk-quit", 48.0)
gi.repository.GLib.GError:gtk: gtk-icon-theme-error-quark:Icon 'gtk-quit' 
not present in theme Tango(0)
[image]

___________
Clean VM install from 2.8.2 iso, upgrade to 2.9 Master via buildbot (06 Aug 2021)
load of gmoccapy lathe sim.

rene-dev, 2021-08-19:
this seems to be a theme related issue. Probably a package is missing.

robertspark, 2021-08-25:
sorted now via clean instal of MX Linux and installing required dependencies, and a run in place install.

sorry I don't have the time and patience to detail the issue experienced at the time..... presume it's probably sorted.... who knows...the best I can suggest is please try to install a clean fresh 2.8.2 linuxcnc install from iso..... upgrade to master via buildbot..... install your config files...... and see if Gmoccapy and QTPlasmaC runs fine as it did not for me at the time and kept throwing errors.

closed 2021-08-25 by robertspark