LinuxCNC/linuxcnc #1223: v2.9 - Python 2 > 3 Gtk - inconview.py {suspect syntax change / error}

robertspark, 2021-08-06:
linuxcnc\lib\python\gladevcp*iconview.py*

line 404
in show_filelabel
self.file_label.hide()
AttributeError: 'IconSelection' object has not attribute 'file_label'

[image]

_________________________________
Clean VM install from 2.8.2 iso, upgrade to 2.9 Master via buildbot (06 Aug 2021)
load of gmoccapy lathe sim.

rene-dev, 2021-08-19:
I cannot reproduce this. file_label is set here: https://github.com/LinuxCNC/linuxcnc/blob/master/lib/python/gladevcp/iconview.py#L201

robertspark, 2021-08-25:
sorted now via clean instal of MX Linux and installing required dependencies, and a run in place install.

sorry I don't have the time and patience to detail the issue experienced at the time..... presume it's probably sorted.... who knows...the best I can suggest is please try to install a clean fresh 2.8.2 linuxcnc install from iso..... upgrade to master via buildbot..... install your config files...... and see if Gmoccapy and QTPlasmaC runs fine as it did not for me at the time and kept throwing errors.

closed 2021-08-25 by robertspark