LinuxCNC/linuxcnc #1267: /tmp/.tool.mmap is a "temporary" permanent file

satiowadahc, 2021-09-13:
A mill has multiple user accounts as each operator has some strong opinions on what they want the desktop/environment to look like. This resulted in an interesting issue where we got an error `tool_mmap_creator() lseek fail` and iocontrol would crash but everything else would run "fine".

To Reproduce, have two accounts on the computer. User A logs in and runs linuxcnc, closes it properly. User B logs in and runs linuxcnc, and gets the error. If B deletes the tmp file they can run linuxcnc normally.

The issue comes from /tmp/.tool.mmap being created with 600 permissions. So I thought logically I'd just change the define in src/emc/tooldata/tooldata_mmap.cc to `#define TOOL_MMAP_MODE     0666`. This resulted in the file being created with 0660 (which I'm still a little confused about and definitely missing something). 

My next thought is to switch the O_CREATE flag to an O_TMPFILE which requires the name to be dropped. Or Delete the file at shutdown. 

Looking for any other suggestions on this.

OS: Ubuntu 20.04, Mint 20.04, Kubuntu 20.04
Kernel: 4.19.199 (Varying RT's)
Latest and greatest master branch.

saxa, 2021-09-13:
IMHO deleting the /tmp/ file at shutdown should be the right thing.

satiowadahc, 2021-09-13:
I'd agree with you, however this comment makes it look very intentional that it was not deleted at shutdown.
```c
void tool_mmap_close()
{
    if (!tool_mmap_base) { return; }
    // mapped file is not deleted
    // flush mmapped file to filesystem
    if (msync(tool_mmap_base, TOOL_MMAP_SIZE, MS_SYNC) == -1) {
        perror("tool_mmap_close(): msync fail");
    }
    if (munmap(tool_mmap_base, TOOL_MMAP_SIZE) < 0) {
        close(creator_fd);
        perror("tool_mmap_close(): munmapfail");
        exit(EXIT_FAILURE);
    }
    close(creator_fd);
} //tool_mmap_close()
```

closed 2021-09-17 by dngarrett

TurBoss, 2021-09-22:
Hello!

This fix made me found an issue I was having with editvcp (qtpyvcp designer)
I found that editvcp doesn't run until i start linuxcnc so the tool.mmap gets initialized
I was able to run editvcp after close linuxcnc

but now I can only run the editvcp while linuxcnc is running

how could I made a dummy tool.mmap without runing linuxcnc if posible at all?
any other ideas are welcome 

Thanks

satiowadahc, 2021-09-22:
@TurBoss Try running designer from command line to see where the error is coming from? 

`designer qt=5`

TurBoss, 2021-09-22:
hello
got this errors
```
[EditVCP][INFO]  Loading UI file: /home/turboss/Projects/tnc/tnc/ui/window.ui (editvcp.py:161)
[EditVCP][INFO]  Loading QSS file: /home/turboss/Projects/tnc/tnc/style.qss (editvcp.py:170)
[EditVCP][INFO]  Starting QtDesigner ... (editvcp.py:191)
[EditVCP.Designer][INFO]  No [KINS] JOINTS entry in self.ini file, using 3 (editvcp.py:93)
[EditVCP.Designer][INFO]  No default jog increments entry found in [DISPLAY] of self.ini file (editvcp.py:93)
[EditVCP.Designer][INFO]  tool_mmap_user(): file open fail: No existe el fichero o el directorio (editvcp.py:93)
[EditVCP.Designer][INFO]  tool_mmap_user(): no mmap file,continuing (editvcp.py:93)
[EditVCP.Designer][INFO]  mmap tool data not available, continuing emc/usr_intf/axis/extensions/emcmodule.cc (editvcp.py:93)
[EditVCP][INFO]  EditVCP finished (editvcp.py:209)
```

thanks

dngarrett, 2021-09-22:
I suspect you need to review all calls to  stat poll() and handle errors that will
occur if LinuxCNC is not running (commented earlier on commit be53acf):

In the master branch, tooldata is stored in a memory mapped file which is now (be53acf) unlinked at normal LinuxCNC terminations to better support use of more than one user account.

If an application uses the linuxcnc python module when LinuxCNC is not running it must handle errors from poll().

Example:
$ cat tst.py
import linuxcnc as l
s=l.stat()
try: s.poll(); print("Status poll success")
except Exception as e: print(e); print("Status poll fail: Is Linuxcnc Running?")
# ... handle exception here to prevent future calls to s.poll()

$ python tst.py
tool_mmap_user(): file open fail: No such file or directory
tool_mmap_user(): no mmap file,continuing
mmap tool data not available, continuing emc/usr_intf/axis/extensions/emcmodule.cc
emcStatusBuffer invalid err=3
Status poll fail: Is Linuxcnc Running?

commit be53acf1 (2021-09-17): tooldata_mmap:unlink TOOL_MMAP_FILENAME at close

closes #1267

Notes:
1) Abnormal LinuxCNC termination (power outage for example)
   may result in the file remaining regardless.
   Manual removal of the the file may be required if LinuxCNC is
   to be started by a user other than the user running
   LinuxCNC at the abnormal termination event if system is
   configured to retain files in /tmp.
2) Related: LinuxCNC uses setuid for critical executables:
     uspace: bin/rtapi_app
     rtai:   bin/pci_write,bin/pci_read
     any:    bin/linuxcnc_module_helper
   The Makefile for the setuid target [... cut]