LinuxCNC/linuxcnc #1268: libboost fault _ZTIN5boost6python17error_already_setE

arabel1a, 2021-09-16:
I've builded latest linuxcnc version from github, resolved dependencies as written in developer manual/building run-in-place. Howewer, it crashes and does not pass 2 tests with similar *errors: ImportError: /home/user/linuxcnc-dev/lib/libpyplugin.so.0: undefined symbol: _ZTIN5boost6python17error_already_setE*
As far as i understood, there is a problem with libboost installation, probably some required packages does not appear in the dependency list. 
First unpassed test: interp/python/error in module gcode
Using debian wheezy with PREEMPT rt kernel 4.19.152-1~deb9u1.
What should i install to make it work correctly?

rene-dev, 2021-09-16:
Wheezy does not work with master anymore, please update your system.

arabel1a, 2021-09-17:
That was a mistake, i'm using debian 10 buster, not wheezy.

rene-dev, 2021-09-17:
try this:
> sudo apt install build-essential git dpkg-dev devscripts
git clone git://github.com/linuxcnc/linuxcnc.git
cd linuxcnc-dev/debian
./configure uspace
cd ..
sudo mk-build-deps --install
cd src
./autogen.sh
./configure
make -j8
. ../scripts/rip-environment
linuxcnc

rene-dev, 2021-09-17:
Try a clean git checkout please.