LinuxCNC/linuxcnc #1271: OpenGL buffer error when selecting line in a large program

c-morley, 2021-09-24:
reported on forum:
https://forum.linuxcnc.org/qtvcp/40905-qtvcp-bugs?start=10

same error in QtVcp  AXIS and GMoccapy (master branch)

similar to ( GMoccapy):

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rs274/glcanon.py", line 347, in inner
    return f(self, *args, **kw)
  File "/usr/lib/python3/dist-packages/gremlin.py", line 417, in select_fire
    self.select(x, y)
  File "/usr/lib/python3/dist-packages/rs274/glcanon.py", line 531, in select
    buffer = glRenderMode(GL_RENDER)
  File "/usr/lib/python3/dist-packages/OpenGL/GL/pointers.py", line 262, in glRenderMode
    "glSelectBuffer too small to hold selection results",
OpenGL.error.GLError: GLError(
	err = GL_STACK_OVERFLOW,
	description = b'stack overflow',
	result = 'glSelectBuffer too small to hold selection results'
)

c-morley, 2021-09-24:
The tested file was 5.4mb and 280000 lines of code

rene-dev, 2021-10-01:
there is code that increases the selection size if its too small, but it does not handle an overflow.