LinuxCNC/linuxcnc #1279: wrong unit handling of interpreter

DjangoReinhard, 2021-09-27:
# wrong unit handling of (standalone-)interpreter

## Here are the steps I follow to reproduce the issue:

 1. rip installation of github master
 2. with this tooltable:
     * T2 P1 Z20 D2 ;2.00mm 4 schneiden fräser 
     * T3 P2 Z53 D3 ;3.00mm 2 flute endmill 
     * T6 P3 Z30 D6 ;6.00mm 4 schneiden fräser 
     * T4 P5 Z40 D8 ;8,00mm 4 schneiden fräser 
     * T23 P7 Z50 D10 ;10,00mm 4 schneiden fräser 
     * T7 P8 Z50 D12 ;12,00mm 4 schneiden fräser 
     * T1 P9 Z123 D4.568 ;45 deg v cutter 
 3. and this commandline:
```
cat gmoccapy_2_tools_with_cutter_radius_compensation.ngc  | rs274 -g -v linuxcnc.var -i some.ini -t tool.tbl
```
4. I got these results:
```
executing
    1 N..... USE_LENGTH_UNITS(CANON_UNITS_MM)
    2 N..... SET_G5X_OFFSET(1, 8165.0024, 12700.0000, -5080.0000, 0.0000, 0.0000, 0.0000)
    3 N..... SET_G92_OFFSET(3004.1677, 209.2678, 78.7400, 0.0000, 0.0000, 0.0000)
    4 N..... SET_XY_ROTATION(0.0000)
    5 N..... SET_FEED_REFERENCE(CANON_XYZ)
READ => ; Gmoccapy_2_tools_with_compensation.ngc
READ => ; testprogram to mill two circles with cutter radius compensation
READ => ; and two different tools
READ =>
READ => ; set the basic settings, this should be
READ => ; on every program the beginning
READ => G17
    6 N..... SELECT_PLANE(CANON_PLANE_XY)
READ => G21
    7 N..... USE_LENGTH_UNITS(CANON_UNITS_MM)
READ => G54
    8 N..... COMMENT("interpreter: continuing to use same coordinate system")
READ => G61
    9 N..... SET_MOTION_CONTROL_MODE(CANON_EXACT_PATH)
READ => G40
   10 N..... COMMENT("interpreter: cutter radius compensation off")
READ => G49
   11 N..... USE_TOOL_LENGTH_OFFSET(0.0000 0.0000 0.0000, 0.0000 0.0000 0.0000, 0.0000 0.0000 0.0000)
READ => G80
   12 N..... COMMENT("interpreter: motion mode set to none")
READ => G90
READ =>
READ => ; get the first tool
READ => T3 M6
   13 N..... SELECT_TOOL(3)
   14 N..... START_CHANGE()
   15 N..... STOP_SPINDLE_TURNING(0)
   16 N..... COMMENT("AXIS,hide")
   17 N..... STRAIGHT_TRAVERSE(-11169.1702, -12909.2678, 5001.2600, 0.0000, 0.0000, 0.0000)
   18 N..... COMMENT("AXIS,show")
   19 N..... CHANGE_TOOL(2)
READ => G43
   20 N..... USE_TOOL_LENGTH_OFFSET(0.0000 0.0000 1346.2000, 0.0000 0.0000 0.0000, 0.0000 0.0000 0.0000)
READ => T1
   21 N..... SELECT_TOOL(1)
READ =>
READ => ; go to center of the workpiece
READ => G0 X50 Y50
   22 N..... STRAIGHT_TRAVERSE(50.0000, 50.0000, 3655.0600, 0.0000, 0.0000, 0.0000)
READ => G0 Z30
   23 N..... STRAIGHT_TRAVERSE(50.0000, 50.0000, 30.0000, 0.0000, 0.0000, 0.0000)
READ =>
READ => ; turn on spindle clockwise
READ => S3000
   24 N..... SET_SPINDLE_SPEED(0, 3000.0000)
READ => M3
   25 N..... START_SPINDLE_CLOCKWISE(0)
READ =>
READ => ; coolant on
READ => M8
   26 N..... FLOOD_ON()
READ =>
READ => G0 Z10
   27 N..... STRAIGHT_TRAVERSE(50.0000, 50.0000, 10.0000, 0.0000, 0.0000, 0.0000)
READ => F250
   28 N..... SET_FEED_RATE(250.0000)
READ => G1 Z0
   29 N..... STRAIGHT_FEED(50.0000, 50.0000, 0.0000, 0.0000, 0.0000, 0.0000)
READ =>
READ => ; tool radius compensation
READ => ; go in circle with lower feed rate
READ => G41
   30 N..... COMMENT("interpreter: cutter radius compensation on left")
READ => G3 X70 Y50 Z-1 I10 J0
Radius of cutter compensation entry arc is not greater than the tool radius
G3 X70 Y50 Z-1 I10 J0
READ =>
READ => ; make first circle with normal feed
READ => F2000
   31 N..... SET_FEED_RATE(2000.0000)
READ => G3 I -20
Radius of cutter compensation entry arc is not greater than the tool radius
G3 I -20
READ =>
READ => ; out of the workpiece
READ => G3 X50 Y50 Z1 I-10 J0
Radius of cutter compensation entry arc is not greater than the tool radius
G3 X50 Y50 Z1 I-10 J0
READ => G0 Z30
Length of cutter compensation entry move is not greater than the tool radius
G0 Z30
READ =>
READ => ; coolant off
READ => M9
   32 N..... MIST_OFF()
   33 N..... FLOOD_OFF()
READ =>
READ => ; turn off cutter radius compensation, otherwise no tool change is possible
READ => G40
   34 N..... COMMENT("interpreter: cutter radius compensation off")
READ => M6
   35 N..... START_CHANGE()
   36 N..... STOP_SPINDLE_TURNING(0)
   37 N..... COMMENT("AXIS,hide")
   38 N..... STRAIGHT_TRAVERSE(50.0000, 50.0000, 3655.0600, 0.0000, 0.0000, 0.0000)
   39 N..... COMMENT("AXIS,show")
   40 N..... CHANGE_TOOL(7)
READ => G43
   41 N..... USE_TOOL_LENGTH_OFFSET(0.0000 0.0000 3124.2000, 0.0000 0.0000 0.0000, 0.0000 0.0000 0.0000)
READ =>
READ => ; optional stop for testing
READ => (MSG, optional stop, i.e. for cleaning the workpiece, or other thinks)
   42 N..... MESSAGE(" optional stop, i.e. for cleaning the workpiece, or other thinks")
READ => M1
   43 N..... OPTIONAL_PROGRAM_STOP()
READ =>
READ => ; go again to the center of the workpiece
READ => G0 X50 Y50
   44 N..... STRAIGHT_TRAVERSE(50.0000, 50.0000, 1877.0600, 0.0000, 0.0000, 0.0000)
READ => G0 Z10
   45 N..... STRAIGHT_TRAVERSE(50.0000, 50.0000, 10.0000, 0.0000, 0.0000, 0.0000)
READ =>
READ => ; turn on spindle clockwise
READ => S6000
   46 N..... SET_SPINDLE_SPEED(0, 6000.0000)
READ => M3
   47 N..... START_SPINDLE_CLOCKWISE(0)
READ =>
READ => ; mist on
READ => M7
   48 N..... MIST_ON()
READ =>
READ => G0 Z10
   49 N..... STRAIGHT_TRAVERSE(50.0000, 50.0000, 10.0000, 0.0000, 0.0000, 0.0000)
READ => F400
   50 N..... SET_FEED_RATE(400.0000)
READ => G1 Z1
   51 N..... STRAIGHT_FEED(50.0000, 50.0000, 1.0000, 0.0000, 0.0000, 0.0000)
READ =>
READ => ; Werkzeugradienkompensation ein
READ => ; Anfahrradius mit Eintauchgeschwindigkeit
READ => G41
   52 N..... COMMENT("interpreter: cutter radius compensation on left")
READ => G3 X65 Y50 Z-1 I7.5 J0
Radius of cutter compensation entry arc is not greater than the tool radius
G3 X65 Y50 Z-1 I7.5 J0
READ =>
READ => ; 2. Kreis mit Vorschubgeschwindigkeit fräsen
READ => F1000
   53 N..... SET_FEED_RATE(1000.0000)
READ => G3 I-15
Radius of cutter compensation entry arc is not greater than the tool radius
G3 I-15
READ =>
READ => ; Aus dem Werkstück
READ => G3 X50 Y50 Z1 I-7.5 J0
Radius of cutter compensation entry arc is not greater than the tool radius
G3 X50 Y50 Z1 I-7.5 J0
READ => G0 Z30
Length of cutter compensation entry move is not greater than the tool radius
G0 Z30
READ =>
READ => G0 Z30
Length of cutter compensation entry move is not greater than the tool radius
G0 Z30
READ => G0 X50 Y50
Length of cutter compensation entry move is not greater than the tool radius
G0 X50 Y50
READ =>
READ => M9 ; coolant off
   54 N..... MIST_OFF()
   55 N..... FLOOD_OFF()
READ => M5 ; spindle off
   56 N..... STOP_SPINDLE_TURNING(0)
READ => M2 ; program end
   57 N..... SET_G5X_OFFSET(1, 8165.0024, 12700.0000, -5080.0000, 0.0000, 0.0000, 0.0000)
   58 N..... SET_XY_ROTATION(0.0000)
   59 N..... SET_FEED_MODE(0, 0)
   60 N..... SET_FEED_RATE(0.0000)
   61 N..... STOP_SPINDLE_TURNING(0)
   62 N..... SET_SPINDLE_MODE(0 0.0000)
   63 N..... PROGRAM_END()
READ =>    64 N..... ON_RESET()
```

## This is what I expected to happen:
on processing G43 with tool #3 I expect the toollength to be 53mm, but interpreter calculates 1346.2mm - which is the length value from tooltable treaten as imperial unit and converting it to mm. All arcs/circles failed with error about to big tool radius

Initialization of backend uses mm settings, and G21 has been used explicitly - so I see no reason, why tooltable entries of a mm-machine should be treaten as imperial values.

## Information about my hardware and software:

 * I am using debian stable (bullseye)
 * I am using this kernel version 5.10.0-8-amd64
 * I am running ...
   * [ ] rip installation from linuxcnc github master self compiled

andypugh, 2021-09-27:
I believe that the tool table is assumed to be in the base units of the configuration. 

Were you using a metric config?

andypugh, 2021-09-28:
There goes that theory. 
I run metric machines, and have not seen this problem. 
Is this also a problem if you use the Axis interface? I seem to recall that Gmoccapy does a lot more internal tool handling that the other GUIs do.

rene-dev, 2021-09-28:
he is not using any UI, running rs274 directly. This is looks like a bug. the tooltable should always be interpreted in machine units.

andypugh, 2021-09-28:
In that case I suspect that the sai doesn't actually look for units in the .ini. I don't see any evidence of a search for TRAJ in saicannon.cc and I do see a hard-coded function:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/sai/saicanon.cc#L720

DjangoReinhard, 2021-09-29:
Andy you're right!
The hardcoded function is the evil.
I tried to change hardcoded number with "_sai._length_unit_factor" and it seems to work.

andypugh, 2021-09-29:
> Andy you're right! The hardcoded function is the evil. I tried to change hardcoded number with "_sai._length_unit_factor" and it seems to work.

But I suspect that variable mirrors the G20 / G21 status, not the machine base units. 
That function should probably look at the INI, after checking that an INI is specified.

closed 2023-10-04 by andypugh