LinuxCNC/linuxcnc #1286: Frame drawing delaying execution of commands

fwillo, 2021-10-03:
I switched to a new computer for controlling my CNC machine. Here I wanted to use the integrated Intel HD 2000 graphics card. Prior to this, I had a dedicated GPU because the internal one was causing massive real time problems. But now it seems that the internal GPU is causing massive lags. This means, that when LinuxCNC is started, and I issue a manual move command, the command is not executed as long as I press the button, but as long as the window needs to draw the new frame in the "Preview" section. I cannot open a menu or anything else at all as well until the new frame is drawn. In my case, the CNC was moving in one direction for 5-6 seconds, even if g-codes were executed.

The GPU acceleration of my system is described at the [arch wiki](https://wiki.archlinux.org/title/Intel_graphics#Installation) and I tried different stuff in order to solve this (glxgears works fine everywhere, so the drawing must be running on some different backend). I can solve this problem as well by getting an external GPU again, but now I think this problem became secondary. 

**It makes me wonder, even if the drawing process is fast enough, if the drawing of the preview frame (or the frame of the window in general) is introducing delays of some order to the execution of movement in the CNC machine?** And if so, shouldn't it be decoupled such that the frame drawing is not causing delays of commands and shutting down the whole program until finished (respecting change of input independently to the rest of the program)?

If I'm having a wrong idea of how linuxcnc in the whole works, please correct me.  I'm not too familiar with the code of linuxcnc and was observing this behaviour causing this question. Looking forward to some contribution.

## Information about my hardware and software:
 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Arch Linux
 * I am using this kernel version (shown by `uname -a`): 5.14
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [x] A binary I built myself (described [here](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=linuxcnc))
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): 2.8.0
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): parallel port

## Here are the steps I follow to reproduce the issue:
 1. A machine running with insufficient graphics capabilities (e.g. Intel HD 2000 graphics on Arch Linux)
 2. Open LinuxCNC
 3. Try some manual movements and observe behaviour

## This is what I expected to happen:
Immediate response to keys even though the frame is not finished drawing

## This is what happened instead:
The command of the respetive manual movement is executed as long as the frame is being drawn

## It worked properly before this:
It "worked" when no graphics acceleration were causing perceptible lags.

pcw-mesa, 2021-10-04:
I guess this depends on how the GUI/user interface is written.
AFAIK Graphic delays do not affect programmed motion but can 
likely affect keyboard/mouse input processing.  

It may be that LinuxCNC  always waits for the preview to complete 
before processing commands.

Do you have a very complex backplot? If so you can disable
the preview in the gcode file:  https://linuxcnc.org/docs/html/gui/axis.html#axis:preview-control

fwillo, 2021-10-05:
Indeed I do not have a complex backplot. A solution, of course, would be to disable this. I could also switch from Preview to DRO, which solves the problem with the crappy driver circumstances. Nevertheless it bothered me, because for me it seems that the lag is caused by the frame rendering. So in principle, from my point of view, even in DRO (or Preview disabled), the rendering of the whole window with text etc. delays action from the program. The extreme lag when viewing the preview with Intel HD graphics is not reduced to manual input, but affects executed g-code as well.

andypugh, 2021-10-05:
It _shouldn't_ affect G-code, nor should it affect HAL-jogs (eg with hardware buttons or jogwheels). 
The point of the realtime subsystem is to endure that user-space delays do not affect motion.