LinuxCNC/linuxcnc #1299: name 'QGLWidget' is not defined in qt5_graphics.py

annhan, 2021-10-12:
Hi,

I imported the NURB module to my screen.
The V2.8, my screen work ok.

I updated my Linux to master brand and can't build with the name ```'QGLWidget' is not defined``` in qt5_graphics.py error.

my error:

```
Traceback (most recent call last):
  File "/home/mwork/mworklinuxcnc/bin/qtvcp", line 430, in <module>
    APP = QTVCP()
  File "/home/mwork/mworklinuxcnc/bin/qtvcp", line 231, in __init__
    window.load_extension(opts.usermod)
  File "/home/mwork/mworklinuxcnc/lib/python/qtvcp/qt_makegui.py", line 259, in load_extension
    methods, self.handler_module, self.handler_instance = self._load_handlers([handlerpath], self.halcomp, self)
  File "/home/mwork/mworklinuxcnc/lib/python/qtvcp/qt_makegui.py", line 287, in _load_handlers
    mod = __import__(basename)
  File "/home/mwork/mdragon/mdragon_handler.py", line 20, in <module>
    from qtvcp.widgets.nurbs_editor import NurbsEditor
  File "/home/mwork/mworklinuxcnc/lib/python/qtvcp/widgets/nurbs_editor.py", line 52, in <module>
    from qt5_graphics import Lcnc_3dGraphics
  File "/home/mwork/mworklinuxcnc/lib/python/qt5_graphics.py", line 183, in <module>
    class Lcnc_3dGraphics(QGLWidget,  glcanon.GlCanonDraw, glnav.GlNavBase):
NameError: name 'QGLWidget' is not defined

````
I'm using master brand

thanks

c-morley, 2021-10-12:
Are you compiling master ? If so, did you make clean before re-compiling?

closed 2021-10-13 by annhan

c-morley, 2021-10-13:
Excellent. also just added Nurbs editor to the designer editor so it's easier to add to a screen.
Please me me know if the nurbs editor was useful or could be improved.
I made it just to see if it would work to have a display to show the shape in realtime.