LinuxCNC/linuxcnc #1348: gcode position variables dont obey radius/diameter mode

rene-dev, 2021-11-01:
- run sim/axis-lathe, enable and home
- go to mdi, type
- g7
- g0x1
- (DEBUG,#<_x>)
- g8
- (DEBUG,#<_x>)

in both cases x is 0.5. the docs say: "Return current relative X coordinate including all offsets."
Shoud we consider this a bug? should #<_x> reflect diameter mode?

andypugh, 2021-11-01:
I think the current behaviour is correct. It returns the axis position. 
I would consider it the job of the code that is being run to interpret it as diameter or radius.

rene-dev, 2021-11-01:
But the documentation doesn't say axis position, it says coordinate. Maybe the best way is to update the documentation. I came across this while doing calculations for a threading cycle in a post processor for a cam.

commit 8a966531 (2021-11-21): add note about diameter mode for variables. closes #1348
closed 2021-11-21 by rene-dev