LinuxCNC/linuxcnc #1349: Add array of strings to label digital IO and analog IO to Motion

satiowadahc, 2021-11-02:
Writing thoughts down as this would be a nice to have, I might get to it later this week

Add three strings of arrays to motion:
- One for labeling digital pins 
- One for labeling analog pins
- One for labeling misc errors.

It should be an error if number of IO and the arrays are provided. 

EG:
```
din_names=endpoint_prox
dout_names=airvalve,vacuumservo,warningled
misc_errors=safety_door,motor_temp,chips_full
```
these would make hal pins:
motion.din-endpoint_prox
motion.dout-airvalve,
motion.dout-vacuumservo
motion.dout-warningled

motion.err-safety_door (Or keep the number but change the resulting error message)
motion.err-motor_temp
motion.err-chips_full

Since C variable names and halpins don't need to line up this should be pretty minimal changes

andypugh, 2021-11-02:
The more I think about this the more I like it.

SebKuzminsky, 2021-11-02:
Don't net names and pin aliases cover some of this use case?

satiowadahc, 2021-11-03:
Some. but sometimes I'm unlinking from halcmd so I can debug a mill and manually toggle a pin, and what pin was that again? let me just pull up the hal file... or did I change it there already.., ?

andypugh, 2021-11-03:
It feels like a natural extension of the "names=" option that many of the other HAL components have. It should only touch the code fairly lightly too, nothing changes after the point that the pin names are defined.

dngarrett, 2021-11-03:
It's hard to improve on the existing halcmd alias provisions which work on any hal pin or parameter

SebKuzminsky, 2021-11-03:
The `names` parameters that some/many modules have operates on instance names, not pin names.

I agree with Dewey we already have this requested feature, in the form of hal pin aliases: http://linuxcnc.org/docs/2.8/html/man/man1/halcmd.1.html#COMMANDS

c-morley, 2021-11-03:
This seems cleaner then alias. What is the complaint of having this option, other then there is another way to do it (which then using alias, shows two pins that do the same thing)?

satiowadahc, 2021-11-03:
Its another convenience function in my opinion. Defining parameters in the INI also makes it nicer to explicitly access in python.

 MR is up, I'm gonna keep using it, hope the community finds some use in it.

"If a programmer built a car, there would be 7 ways to turn the radio up. Including the using the steering wheel - Useful feature I must say"

c-morley, 2021-12-11:
Merged for now at least. I see you didn't add error pins. I kinda liked the idea - linuxcnc's error message handling is pretty restrictive.

satiowadahc, 2021-12-11:
I use the misc-error pins already with no issues. Quite handy I find. 

I have a week left of my busy period then maybe I can finish the documentation on this.

c-morley, 2021-12-11:
Oh I see I missed the error pins - they were included.
Docs would be great thanks.

rene-dev, 2023-05-06:
probably closed by https://github.com/LinuxCNC/linuxcnc/pull/1352/

closed 2023-05-06 by rene-dev