LinuxCNC/linuxcnc #1350: qt dragon tool table tool add/delete buttons not calling widget methods

cunninghamks, 2021-11-03:
## Reproduce
 1. Launch QT Dragon
 2. Enable then Home Machine
 3. Switch to Tool tab
 4. Attempt to Add or Delete tool(s)

## This is what I expected to happen:
Expect tool(s) to either be added or deleted.

## This is what happened instead:
Appears that the python commands tied to the IndicatedPushButtons are not actually calling the add/delete methods on the loaded tool_offsetview widget.  This was confirmed with the addition of logging lines within the widget.

[https://github.com/LinuxCNC/linuxcnc/blob/62c48d8c6e5d0661cd642003ece33e11d6c8d840/share/qtvcp/screens/qtdragon_hd/qtdragon_hd.ui#L2245](https://github.com/LinuxCNC/linuxcnc/blob/62c48d8c6e5d0661cd642003ece33e11d6c8d840/share/qtvcp/screens/qtdragon_hd/qtdragon_hd.ui#L2245)

## Potential Resolution
Issue is fixed when editing the QT UI to switch add/delete buttons to QTPushButtons, relinking slots to handler methods, and subsequently calling the loaded widget instance methods from the handler.

## Information about my hardware and software:

Debian Buster 10
LinuxCNC 2.9.0-pre0-4905-g62c48d8c6 installed using buildbot
QT Dragon HD Interface
Cheap parallel port breakout board.

c-morley, 2021-11-03:
Thanks for report and detective work. I fixed it differently but that helped a lot. Pushed to master.

cunninghamks, 2021-11-03:
@c-morley works great.  Thank you.