LinuxCNC/linuxcnc #1363: Building debs in master fails with permissions issue after yesterday's update

rodw-au, 2021-11-09:
## Here are the steps I follow to reproduce the issue:

 1. cd ./src git pull
 2. cd ..
 3. dpkg-buildpackage -b -uc

## This is what I expected to happen:
debs would be built just like they did yesterday
**## This is what happened instead:**

# executing regular installation with regular prefix but unterneath the debian folder
DESTDIR=/home/plasma2/linuxcnc-dev/debian/linuxcnc-uspace /usr/bin/make -C src install
make[2]: Entering directory '/home/plasma2/linuxcnc-dev/src'
emc/Submakefile:4: warning: overriding recipe for target '../include/linuxcnc.h'
emc/Submakefile:4: warning: ignoring old recipe for target '../include/linuxcnc.h'
install -o root -d -m 0755	/home/plasma2/linuxcnc-dev/debian/linuxcnc-uspace/usr/lib/linuxcnc/modules/linuxcnc \
	/home/plasma2/linuxcnc-dev/debian/linuxcnc-uspace/usr/bin \
	/home/plasma2/linuxcnc-dev/debian/linuxcnc-uspace/etc/linuxcnc
install: cannot change owner and permissions of ‘/home/plasma2/linuxcnc-dev/debian/linuxcnc-uspace/usr/lib/linuxcnc/modules/linuxcnc’: Operation not permitted
install: cannot change owner and permissions of ‘/home/plasma2/linuxcnc-dev/debian/linuxcnc-uspace/usr/bin’: Operation not permitted
install: cannot change owner and permissions of ‘/home/plasma2/linuxcnc-dev/debian/linuxcnc-uspace/etc/linuxcnc’: Operation not permitted
make[2]: *** [Makefile:798: install-kernel-dep] Error 1
make[2]: Leaving directory '/home/plasma2/linuxcnc-dev/src'
make[1]: *** [debian/rules:71: override_dh_auto_install] Error 2
make[1]: Leaving directory '/home/plasma2/linuxcnc-dev'
make: *** [debian/rules:39: binary] Error 2
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version Debian Bullseye 11
 * I am using this kernel version (shown by `uname -a`): Preempt_rt
 * I am running ... the build process

rodw-au, 2021-11-09:
Sorry, elevating privileges with su solved this issue.
I'll leave it open becasue these docs should be updated to reflect this new requirement
http://linuxcnc.org/docs/devel/html/code/building-linuxcnc.html#_building_debian_packages

phillc54, 2021-11-09:
I built packages today, both with and without documentation. It built fine as a normal user.

rodw-au, 2021-11-09:
I did see some changes after I pulled it the first time, so I updated before using SU.
Maybe they fixed it.
All very exciting to get this far though!

I'll build them again now and see what happens when I come back in the morning.

rodw-au, 2021-11-09:
Actually that failed right at the beginning:

plasma2@plasma2:~/linuxcnc-dev$ dpkg-buildpackage -b -uc
dpkg-buildpackage: info: source package linuxcnc
dpkg-buildpackage: info: source version 1:2.9.0~pre0
dpkg-buildpackage: info: source distribution UNRELEASED
dpkg-buildpackage: info: source changed by Sebastian Kuzminsky <seb@highlab.com>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source --before-build .
 fakeroot debian/rules clean
# With debhelper 10 the --parallel becomes the default
dh clean --parallel
	rm -f debian/linuxcnc-doc-cn.debhelper.log debian/linuxcnc-doc-en.debhelper.log debian/linuxcnc-doc-es.debhelper.log debian/linuxcnc-doc-fr.debhelper.log debian/linuxcnc-uspace-dev.debhelper.log debian/linuxcnc-uspace.debhelper.log
   debian/rules override_dh_auto_clean
make[1]: Entering directory '/home/plasma2/linuxcnc-dev'
dh_auto_clean
	install -d /home/plasma2/linuxcnc-dev/debian/.debhelper/generated/_source/home
dh_auto_clean: error: mkdir /home/plasma2/linuxcnc-dev/debian/.debhelper/generated/_source: Permission denied
make[1]: *** [debian/rules:59: override_dh_auto_clean] Error 255
make[1]: Leaving directory '/home/plasma2/linuxcnc-dev'
make: *** [debian/rules:39: clean] Error 2
dpkg-buildpackage: error: fakeroot debian/rules clean subprocess returned exit status 2
plasma2@plasma2:~/linuxcnc-dev$ ^C
plasma2@plasma2:~/linuxcnc-dev$

rodw-au, 2021-11-09:
I tried this
cd linuxcnc-dev/debian
> ./configure uspace

and the problem persists
Maybe root owns some folders now

rodw-au, 2021-11-09:
So I pulled a fresh copy onto my Chromebook tonight and it built the debs without a drama.
I'll run chown over the home folder on my machine and try again in the morning. Very strange. 
I don't know how permissions changed in the first place.

smoe, 2021-11-09:
I once run into this, too. The permissions are about setting the "suid" bit that needs to be set for two binaries. If I understand this right, then we are helped and warned by a package taking that action if the package is not built with the -rfakeroot argument to dpkg-buildpackage. The debian/control file also got the "Rules-Requires-Root: binary-targets" flag because of that. In the past, all the packages were run with the -rfakeroot flag, but since most packages do not need the suid, this has mostly been forgotten about.

TL;DR: please try again with `dpkg-buildpackage -rfakeroot`, the "su" should not be required.

rodw-au, 2021-11-10:
Steffen, thank you. Your suggestion worked but only after I used chown to take back the permissions from the root user.
I also found the no-docs parameter to ~/linuxcnc-dev/debian/configure so hopefully that will save a lot of time.

It would be nice if the docs could be updated to cover the use of --rfakeroot and the use of the configure no-docs parameter to save the next poor soul who ventures into this.

phillc54, 2021-11-10:
If you do **./configure --help** in the /debian directory it will let you know about no-docs

rodw-au, 2021-11-10:
Thanks Phil. Thats not really helpful to have the documentation so hidden. Users (myself included) will usually be following the scanty documentation  here.
http://linuxcnc.org/docs/devel/html/code/building-linuxcnc.html#_building_debian_packages
and have no idea about what they are doing

phillc54, 2021-11-10:
That is a standard Linux way of getting help.

You can edit the page directly from Github and make a PR.
[https://github.com/LinuxCNC/linuxcnc/blob/master/docs/src/code/building-linuxcnc.adoc](url)

smoe, 2021-11-10:
I completely agree, this documentation needs to be updated. Thank you for pointing that out. I am not sure if I now take the opportunity to contribute something from you, @rodw-au, and you definitely serve this, but I would now prepare that pull request that @phillc54 suggested. If either of you wants to contribute something to the source tree as an exercise then I will find something for you and guide you - ping me.

rodw-au, 2021-11-10:
Steffen, I would prefer that somebody familiar with the dpkg build process and the recent changes updated the documentation. I really have no clue about the process and just blindly followed the documents and had been doing so until the recent changes broke the process. I don't feel qualified enough to describe the changes. I think the main issue is the need for --rfakeroot to be covered.

I also wondered if .configure could take an argument to build docs for just one language eg. lang=en. The only time I ever met a Spaniard was in Shanghai. There is no call for languages other than en here in Australia.

rodw-au, 2021-11-10:
Looks like this can be closed now.

closed 2021-11-10 by rodw-au