LinuxCNC/linuxcnc #1379: Segfaults when closing Linuxcnc using QTvcp on Debian 12

rodw-au, 2021-11-14:
## Here are the steps I follow to reproduce the issue:

 1. Install Debian 12 and real time kernel
 2. Install Linuxcnc, 
 3. run from the menu
 4. Close linuxcnc and observe segfault

This forum thread refers
https://forum.linuxcnc.org/38-general-linuxcnc-questions/44225-debian-11-bullseye-segfaults-qtplasmac?start=0

## This is what I expected to happen:

Linuxcnc to close gracefully without a segfault or error window

We wondered if this could be Python 3.9 related.

rene-dev, 2021-11-19:
I can confirm this issue with python3.9 on Ubuntu 21.10. with the configs/sim/qtvcp_screens/qtdragon

linuxcnc: line 930: 172401 Segmentation fault      (core dumped) $EMCDISPLAY -ini "$INIFILE" $EMCDISPLAYARGS $EXTRA_ARGS
The python interpreter crashes, But I currently do not have a debug build of python.

> Thread 1 "qtvcp" received signal SIGSEGV, Segmentation fault.
0x00000000008eaa20 in PyCFunction_Type ()
(gdb) bt
#0  0x00000000008eaa20 in PyCFunction_Type ()
#1  0x00007f3668fe015b in  () at /usr/lib/python3/dist-packages/PyQt5/QtCore.abi3.so
#2  0x00007f36690ecf00 in  () at /usr/lib/python3/dist-packages/PyQt5/sip.cpython-39-x86_64-linux-gnu.so
#3  0x00007f3668fe00bd in  () at /usr/lib/python3/dist-packages/PyQt5/QtCore.abi3.so
#4  0x000000000052444f in  ()
#5  0x00000000006203d4 in  ()
#6  0x0000000000609a51 in  ()
#7  0x0000000000609641 in Py_FinalizeEx ()
#8  0x000000000061ae9c in Py_Exit ()
#9  0x000000000060aa4f in  ()
#10 0x000000000060a996 in PyErr_PrintEx ()
#11 0x0000000000607e86 in PyRun_SimpleFileExFlags ()
#12 0x00000000005fcb66 in Py_RunMain ()
#13 0x00000000005d906d in Py_BytesMain ()
#14 0x00007f366de98fd0 in __libc_start_call_main (main=main@entry=0x5d9030, argc=argc@entry=5, argv=argv@entry=0x7ffc26cbdb48) at ../sysdeps/nptl/libc_start_call_main.h:58
#15 0x00007f366de9907d in __libc_start_main_impl (main=0x5d9030, argc=5, argv=0x7ffc26cbdb48, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffc26cbdb38) at ../csu/libc-start.c:409
#16 0x00000000005d8f65 in _start ()
(gdb)

rene-dev, 2021-11-19:
https://bugzilla.redhat.com/show_bug.cgi?id=1844781

c-morley, 2021-11-23:
Thank you for this Rene - it's beyound my skill to debug.

rodw-au, 2021-12-06:
Just fixed a typo in the title. This was observed on Debian 11 Bullseye

Prismatoid, 2021-12-07:
In my case, the segmentation fault was being caused by python-espeak library (more info at https://forum.linuxcnc.org/plasmac/43880-qtplasmac-1-0-98-error-on-shutdown-python-segmentation-fault#228489).
A workaround is to drop the use of espeak in "audio_player.py", but a more stable solution would be to port audio_player.py to use the new python-espeak version 0.6.3, which is not backwards compatible. A first version of a ported file is attached. It might be useful as a first step, but it needs thorough testing (and possibly tweaking) by expert users. The workaround and the preliminary solution have worked in a Raspberry Pi 4B-8GB setup with version 2.9.0~pre0.

Prismatoid, 2021-12-07:
In my case, the segmentation fault was being caused by python-espeak library (more info at https://forum.linuxcnc.org/plasmac/43880-qtplasmac-1-0-98-error-on-shutdown-python-segmentation-fault#228489).
A workaround is to drop the use of espeak in "audio_player.py" from QtVCP, but perhaps a more stable solution would be to port audio_player.py to use the new python-espeak version 0.6.3, which is not backwards compatible. A first attempt at a ported file is attached; it might be useful but it needs thorough testing (and possibly fixing) by expert users. The workaround and the preliminary solution work well in a Raspberry Pi 4B-8GB setup with LinuxCNC 2.9.0~pre0 on Debian 11 Linux 5.10.82-rt58-v8+. Glad to provide more details if needed.
[audio_player.zip](https://github.com/LinuxCNC/linuxcnc/files/7670981/audio_player.zip)