LinuxCNC/linuxcnc #1383: PNCCONF BUGS. Backlash on YZAC + SSR inversion

3Dpassion, 2021-11-15:
## Here are the steps I follow to reproduce the issue:

 1. Backlash only sets for X axis, everything else is greyed out.
 2. Backlash value rounds up in some weird way, if you set 0.06 it forces 0.05 for example.
 3. Inverting SSR output caused generation of faulty code.

Latest LinuxCNC and MESA 7i96 board.

c-morley, 2021-11-16:
I see the backlash problem.
The backlash setting has a maximum setting of .05 - i bet you use metric -  what would a reasonable max setting for metric be?
can you tell me more about the faulty code for SSR and what pins are SSR?

3Dpassion, 2021-11-16:
> I see the backlash problem. The backlash setting has a maximum setting of .05 - i bet you use metric - what would a reasonable max setting for metric be? can you tell me more about the faulty code for SSR and what pins are SSR?

Well, depends on how bad the machine is. I already set all that manually anyways, but I do not see why limit that field in any way at all and not let user enter what he needs. 0.05in is 1.27mm, I'd definitely set it to allow at least 0.5mm, but again, why limit it. My Harbor Freight knee mill for example turned out to have around 0.05mm backlash on XY and a whole 0.1 on Z. Just for the record.

On mesa 7i96 there are dedicated SSR outputs. GPIO OUT they are. And they simply do not support the command the way this configurator generates it. That method only works for inputs and not for outputs. 

As a side note also, I don't think you can convert input into an output on this board, but configurator offers that fucntion for some reason. I would double check that.

c-morley, 2021-11-16:
We limit entries to keep settings reasonable for people who don't really know what to set it too - also the gui toolkit requires limits.
I set them at 1mm or .050 inch - IMHO machine worse then this should be fixed anyways.

I need much more detailed info on what works and what doesn't HAL code wise. I don't own the hardware so can't test my self.
A list of all HAL pins on a 7i96 would be helpful for instance.
Also what pins are SSR  i don't think they all are.

c-morley, 2021-11-16:
They backlash selection and range fixes have been push to master now.

3Dpassion, 2021-11-16:
> I set them at 1mm or .050 inch - IMHO machine worse then this should be fixed anyways.

Agree! I have 0.05ish mm and even that is substantial. 

> I need much more detailed info on what works and what doesn't HAL code wise. I don't own the hardware so can't test my self. A list of all HAL pins on a 7i96 would be helpful for instance. Also what pins are SSR I don't think they all are.

pin example: hm2_7i96.0.ssr.00.out-00

I figured whoever design the configurator would know this anyways. This pin basically doesn't take that syntax at all.

pcw-mesa, 2021-11-19:
Its currently awkward to invert SSR outputs (you would need to add a 'not' component for each inverted output)
So I have added  'invert' pins to the driver (in master)  so 'setp hm2_7i96.0.ssr.00.invert-02' would invert
the on/off state of SSR output 2

3Dpassion, 2021-11-19:
Amazing responsiveness guys! I can no longer use a configurator as I modified a lot of things manually already, what would be the invert command for the pin example that I mentioned above? I am not too good with HAL syntax, its my first experience with these things.

pcw-mesa, 2021-11-19:
You would need to fetch the latest master as I just added the "invert" SSR pins this morning

3Dpassion, 2021-11-19:
> You would need to fetch the latest master as I just added the "invert" SSR pins this morning

Oh wait, are we talking about Mesa board firmware now? Or PNCCONF? I am confused. Should I reflash the firmware then?

pcw-mesa, 2021-11-19:
I am talking about LinuxCNC 

You need the latest master branch, as the 
SSR invert feature was added to the hostmot2 
driver.

3Dpassion, 2021-11-19:
> I am talking about LinuxCNC
> 
> You need the latest master branch, as the SSR invert feature was added to the hostmot2 driver.

Gotcha, thank you very much! I'll just reprogram my driver to invert that inside of it instead. Simpler this way. But glad we now have that taken care of for future use!

closed 2021-11-19 by 3Dpassion