LinuxCNC/linuxcnc #1397: Possible missing information ini config for TRAJ

alkabal, 2021-11-22:
Hi

I'm probably wrong but i see nothing in documentation about this TRAJ config

https://github.com/LinuxCNC/linuxcnc/search?q=NO_PROBE_JOG_ERROR
https://github.com/LinuxCNC/linuxcnc/search?q=NO_PROBE_HOME_ERROR

http://linuxcnc.org/docs/2.8/html/config/ini-config.html#_traj_section

Br

alkabal, 2021-11-22:
code related to this
https://github.com/LinuxCNC/linuxcnc/blob/75b03131e4ec3abdad16b762a73f5716d09a6bc5/src/emc/motion/command.c#L2054

andypugh, 2021-11-23:
On Mon, 22 Nov 2021 at 01:27, alkabal ***@***.***> wrote:

> I'm probably wrong but i see nothing in documentation about this TRAJ
> config
>

On the contrary, I think that you are right, and the option has never been
documented.
(Also, I was unaware of it)

-- 
atp

alkabal, 2021-11-23:
I have find that searching other things.

I like to find a way for adding a motion.setter-input if this can be accepted. 
Imo this is actually too much trouble for managing a touch probe + a tool setter 

Can I ask what is your opinion about that ?

Best regards

c-morley, 2021-11-23:
Also using the Mcode to change those No Jog setting might be useful too.

grandixximo, 2026-10-05:
Documented in #1642 (merged 2022-03-06, in 2.8.3): both NO_PROBE_JOG_ERROR and NO_PROBE_HOME_ERROR are in the [TRAJ] section of the INI docs. Closing.

closed 2026-10-05 by grandixximo