LinuxCNC/linuxcnc #1398: Gmoccapy error - combi_dro.py

LJspice, 2021-11-22:
## Here are the steps I follow to reproduce the issue:

 1. Install linuxcnc buster preempt-RT 2.8 ISO
 2. Change repo to 2.9/master, update/upgrade
 3. Install missing dependency gnome-icon-theme
 4. Open sim gmoccapy config (Lathe_C.ini, though others do it)
 5. Click spindle on button. Error

## This is what I expected to happen:

No error message, spindle shows as on

## This is what happened instead:

Error message. After error appears, jog buttons stop working

```
Traceback (most recent call last):
  File "/usr/bin/gmoccapy", line 2279, in _periodic
    self._update_vc()
  File "/usr/bin/gmoccapy", line 3773, in _update_vc
    diameter = int(self.dro_dic["Combi_DRO_0"].get_position()[1] * 2)
  File "/usr/lib/python3/dist-packages/gladevcp/combi_dro.py", line 633, in get_position
    positions = self._position()
TypeError: _position() missing 5 required positional arguments: 'object', 'p', 'rel_p', 'dtg', and 'joint_actual_position'
```

<details>
  <summary>Click to expand complete terminal log</summary>

```
linuxcnc@lcnc-lathe:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
linuxcnc@lcnc-lathe:~$ uname -a
Linux lcnc-lathe 4.19.0-18-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.208-1 (2021-09-29) x86_64 GNU/Linux
linuxcnc@lcnc-lathe:~$ linuxcnc
LINUXCNC - 2.9.0-pre0-5081-g8a9665315
Machine configuration directory is '/home/linuxcnc/linuxcnc/configs/sim.gmoccapy.lathe_configs'
Machine configuration file is 'lathe_C.ini'
Starting LinuxCNC...
Found file(REL): ./core_sim_lathe_C.hal
Note: Using POSIX realtime
 
identityKinematicsSetup: coordinates:XZC
   Joint 0 ==> Axis X
   Joint 1 ==> Axis Z
   Joint 2 ==> Axis C
identityKinematicsSetup: Recommend: kinstype=both
 
Found file(REL): ./spindle_sim.hal
Found file(REL): ./simulated_home_lathe.hal
not enough values to unpack (expected 2, got 1)
'btn_sel_next'
'btn_sel_prev'
'btn_select'
'btn_home'
'btn_dir_up'
'btn_jump_to'
not enough values to unpack (expected 2, got 1)
not enough values to unpack (expected 2, got 1)
'IconFileSelection' object has no attribute 'on_btn_jump_to'
not enough values to unpack (expected 2, got 1)
not enough values to unpack (expected 2, got 1)
Icon from stock
0  =  /usr/bin/gmoccapy
1  =  -ini
[... 312 lines cut ...]
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version: `LINUXCNC - 2.9.0-pre0-5081-g8a9665315`
 * I am using this user interface: `gmoccapy`
 * I am using this interface hardware vendor and chipset: `SIM`

## Other

This exact same error presented itself in debian 11, building a binary myself from master. This error also presents in debian 10 (linuxCNC ISO) wth a buildbot master install. Switching from debian 11 back to 10 was an attempt at troubleshooting by getting back onto a supported version. Both fresh installs happened this weekend.

On debian 11, two similar errors appeared (though I forget steps to reproduce, I did play with an axis spindle orient sim, as well as gmoccapy):
```
Traceback (most recent call last):
File "/usr/lib/python3/dist-packages/gladevcp/combi_dro.py", line 271, in _on_eventbox_clicked
self.toggle_readout()
TypeError: 'bool' object is not callable
```
And
```
Traceback (most recent call last):
File "usr/lib/python3/dist-packages/gladevcp/offsetpage_widget.py", line 314, in set_editing
self.store[i][12]=self.convert_color(color)
File "usr/lib/python3/dist-packages/gladevcp/offset_page_widget.py", line 454, in convert_color
colortuple=((int(color.red*255.0),int(color.green*255.0),int(color.blue*255.0)))
AttributeError: 'NoneType' object has no attribute 'red'
```

hansu, 2021-12-27:
> Traceback (most recent call last):
File "/usr/lib/python3/dist-packages/gladevcp/combi_dro.py", line 271, in _on_eventbox_clicked
self.toggle_readout()
TypeError: 'bool' object is not callable

That I got fixed in https://github.com/hansu/linuxcnc/commit/6e218104a515078208b8b8b40aa8f0096218522e

hansu, 2021-12-28:
@LJspice Can you try to reproduce the third error? Thanks!

hansu, 2021-12-28:
And also clicking on D opens the entry for R. I don't know if this is intended...

[image]

LJspice, 2021-12-28:
@hansu I went through and clicked every button in the GUI to reproduce:

1. estop off, run on
2. Home machine
3. Go into offsets page
4. Edit offsets
5. Click return arrow, bottom right
6. Get error

[image]

hansu, 2021-12-28:
A possible fix for that I have done in https://github.com/hansu/linuxcnc/commit/3f28b7543159b400c56f5b1d95c3e67076e13ce8.
But I am not sure if that works as expected.
Because Norbert (@gmoccapy) has completely replaced the content and also the signature of the  `_position`-function  in https://github.com/LinuxCNC/linuxcnc/commit/d79a3db4c9351138cbd2dd54a3288be7dd6f3c93.
And I don't know if the calculations which are done in the old `_position`-function are really no longer needed.

hansu, 2021-12-28:
> @hansu I went through and clicked every button in the GUI to reproduce:
> 
>     1. estop off, run on
> 
>     2. Home machine
> 
>     3. Go into offsets page
> 
>     4. Edit offsets
> 
>     5. Click return arrow, bottom right
> 
>     6. Get error
> 

I think this is connected to the error, that one DRO elements has no color:

[image]

PR #1549 (2022-01-29): offsetpage_widget: Fix error on color conversion
Fixes https://github.com/LinuxCNC/linuxcnc/issues/1398#issuecomment-1002229456

Closes #1398
closed 2022-01-29 by c-morley