LinuxCNC/linuxcnc #1399: misleading or wrong documentation of feedrate in canon.hh

DerAndere1, 2021-11-24:
## Here are the steps I follow to reproduce the issue:

 1. Read https://github.com/LinuxCNC/linuxcnc/blob/8a9665315cd96e9da24e08a0885b9a3686f1e7aa/src/emc/nml_intf/canon.hh#L290-L294
 2. Compare this to the code at 
https://github.com/LinuxCNC/linuxcnc/blob/8a9665315cd96e9da24e08a0885b9a3686f1e7aa/src/emc/task/emccanon.cc#L675-L685
3. Observe that the  documentation in canon.hh is wrong. the code in emccannon is correct according to NIST RS274 interpreter version 3 documentation. 

## This is what I expected to happen:
If I understand correctly, canon.hh lines 295 ff  should paraphrase what is specified in the NIST RS274 Interpreter version 3 documentation, section "2.1.2.5 Feed Rate":

> For motion of two or three rotational axes with X, Y, and Z axes not moving, the rate is
> applied as follows. Let dA, dB, and dC be the angles in degrees through which the A, B,
> and C axes, respectively, must move. Let D = sqrt(dA*dA + dB*dB + dC*dC). Conceptually, D is a
> measure of total angular motion, using the usual Euclidean metric. Let T be the amount
> of time required to move through D degrees at the current feed rate in degrees per
> minute. The rotational axes should be moved in coordinated linear motion so that the
> elapsed time from the start to the end of the motion is T plus any time required for
> acceleration or deceleration.
> 

## This is what happened instead:

## It worked properly before this:
Was wrong at least since 2002

## Information about my hardware and software:

 * This is found in current master branch. I do not use LinuxCNC, I only try to understand it.

andypugh, 2021-11-24:
Do you want to propose a revised wording? 

It's possible that the comment was true at some point. And interesting that there is no mention of three rotary axes.

DerAndere1, 2021-11-25:
I take your comment as a confirmation of my finding. I will prepare a pull request.

closed 2022-03-14 by SebKuzminsky