LinuxCNC/linuxcnc #1404: MB2HAL errorcounter does not count high

andrewheeler82, 2021-11-25:
I've been using LinuxCNC for over 10 years now. Starting with the parallel port and now currently on the road with a Mesa 7i96. I run my VFD through MB2HAL and I noticed 2 strange things. I have created a patch for this, maybe the developer of MB2HAL could take a look at it. It's currently working so well for me.

--- mb2hal.c	2021-11-25 06:21:10.899817224 +0100
+++ mb2hal.c	2021-11-25 06:27:46.537111414 +0100
@@ -223,7 +223,7 @@ void *link_loop_and_logic(void *thrd_lin
             }
 
             if (ret != retOK && modbus_get_socket(this_mb_link->modbus) < 0) { //link failure
-                (*this_mb_tx->num_errors)++;
+                (**this_mb_tx->num_errors)++;
                 ERR(this_mb_tx->cfg_debug, "mb_tx_num[%d] mb_links[%d] thread[%d] fd[%d] link failure, going to close link",
                     this_mb_tx_num, this_mb_tx->mb_link_num, this_mb_link_num, modbus_get_socket(this_mb_link->modbus));
                 modbus_close(this_mb_link->modbus);
@@ -347,6 +347,7 @@ retCode get_tx_connection(const int this
         ret = modbus_connect(this_mb_link->modbus);
         if (ret != 0 || modbus_get_socket(this_mb_link->modbus) < 0) {
             modbus_set_socket(this_mb_link->modbus, -1); //some times ret was < 0 and fd > 0
+            (**this_mb_tx->num_errors)++;
             ERR(this_mb_tx->cfg_debug, "mb_tx_num[%d] mb_links[%d] cannot connect to link, ret[%d] fd[%d]",
                 this_mb_tx_num, this_mb_tx->mb_link_num, ret, modbus_get_socket(this_mb_link->modbus));
             return retOK; //not connected

hansu, 2021-11-25:
Yeah that makes sense.
I would also increase the error counter here:
```diff
    //set slave id according to each mb_tx
    ret = modbus_set_slave(this_mb_link->modbus, this_mb_tx->mb_tx_slave_id);
    if (ret != 0) {
+       (**this_mb_tx->num_errors)++;

        ERR(this_mb_tx->cfg_debug, "mb_tx_num[%d] mb_links[%d] cannot set slave [%d]",
            this_mb_tx_num, this_mb_tx->mb_link_num, this_mb_tx->mb_tx_slave_id);
        return retOK; //not connected
    }

```

andrewheeler82, 2021-11-25:
ok that sounds good. The second place where I added the counter. I use it to monitor for broken connections and trigger an emergency stop if the connection is broken. I would like to do more for the LinuxCNC project, but my programming skills are still at the beginning.

hansu, 2021-12-12:
@andrewheeler82 I propose to open a pull request for this. Then you have a better chance that this change comes into master ;-)

andrewheeler82, 2021-12-12:
Ok i will Check 😜

closed 2021-12-13 by andrewheeler82